MarlinFirmware/Marlin #9846: [PR] Add fan speed in M105 report

theodroid, 2018-02-27:
Hi, it would be nice if M105 could return the current print cooling fan speed, to allow the USB host to know the actual speed.
If using OctoPrint, some apps like Printoid (an OctoPrint app for Android) can take advantage of this feature, because for the moment there is no method for the host to know at which speed the fan is set.

Zer0t3ch, 2018-03-28:
To clarify: do you want it to report the speed (percentage) that it's currently set for, or do you want it to return the actual fan RPM? (The latter of which would be impossible, I think)

theodroid, 2018-03-28:
Just the fan speed in percentage (or the raw value from 0 to 255 if you prefer).

This feature will allow the host to display the actual speed of the fan:
For exemple, for hotends, if you turn on the heating via LCD or Gcode file, the host notice it and display the target temperature.
But for fans, if you turn it on via LCD or Gcode file, the host can't know it, so the user have no idea of the speed of his fan while running.

thinkyhead, 2018-03-28:
@MarlinFirmware/host-software-team — What do you think of this proposal?

Zer0t3ch, 2018-03-29:
@theodroid Oh yeah, the raw value makes more sense than a 0-100 percentage. 

If the "powers that be" are open to accepting this, I'd be happy to take it on. (Been trying to contribute, and this looks simple enough for me to take care of and save the time of someone more experienced.)

foosel, 2018-03-29:
Raw value is fine, but please ensure backwards compatibility of the output format. 

M105 responses are notoriously tricky to parse since a whole ton of variants exist (T included with multi extruder output, T not included, T included but is actually always T0, ...).

repetier, 2018-03-29:
Please don't.
M105 response is already long and that can easily lead to
a) blocking output when output buffer is full
b) increased communication errors. Don't ask why but experience has shown that increased fast output can create more communication errors. 
c) It is not volatile, so why block communication with static values.

Hosts can already know values if they set the fan speed. They just need to analyse the send values. In Repetier-Firmware we also communicate if user changes fan speed manually. Every time fan speed gets changed firmware sends
Fanspeed0:255
where 0 is the fan number and after : you have the raw speed.
This works 
a) For any number of fans
b) has lowest communication impact
c) Is easily parsable by hosts if they want to know fan speeds.

This is all is especially true for more elaborate printers having 4 extruder, a bed, 2 fans. Knowing the format you will see that M105 becomes very long for virtually nothing compared to sending same data when it changes. Temperatures changes all the time so reporting them repeatedly makes sense, fans don't.

foosel, 2018-03-29:
> In Repetier-Firmware we also communicate if user changes fan speed manually.

That would also be an option, making the need for a M105 extension non existent. (Btw, mirroring the same kind of verbose reponse on tool change would also be fairly helpful, I know Repetier does it)

And good point with the length, I'm already running into buffering issues with auto reporting on multi extruder setups. I hereby retract my earlier 👍 for this ;)

Zer0t3ch, 2018-03-29:
@repetier Adding automatic reporting when the fan speed changes sounds like a good idea. 

If you guys do end up wanting to add manual reporting, I would suggest an extension to M105 that reports ONLY fan speed with something like `M105 F-1` for all fans or `M105 F0` for only the first fan, and so on, but wouldn't report any fan information unless specified.

Zer0t3ch, 2018-03-30:
@repetier Now that I've had a chance to dig into the code, I've realized that it should be relatively easy to add reporting whenever M106/M107 is received, but it wouldn't be nearly as trivial to add reporting when fan speed is changed via an LCD. How important would you think that is? (reporting fan speed when altered via LCD menu) I'm not entirely sure how detrimental it would be to miss that because I don't fully understand the use-case for reporting it at all.

@theodroid

repetier, 2018-03-30:
Reporting on LCD change is the only one really important.
Think you have repetier-server running which shows fan speed in frontend. It shows off and you enable it on LCD. Server will still show off since it did not know user used a different interface.

On the other side if the user changes it in server interface it knows already the value so reporting is not really necessary here, but also no problem.

In repetier-firmware there is only one function to set fan speed, so we just have put the message there and do not really care who did change it.

theodroid, 2018-03-30:
I agree with @repetier 
Reporting on LCD change is the most important since the host doesn't know the fan speed have been changed.

Zer0t3ch, 2018-03-30:
@repetier Okay, thanks. It will be less simple and require some more restructuring, but it shouldn't be too difficult.

mamoit, 2018-05-16:
I was looking for a way to get the fan speed using a gcode, and there seems to be none.
Instead of opening a new feature request I thought it would be more productive to continue this discussion.
Even if does not come with `M105`, wouldn't it be interesting to add a gcode to get the current fan raw value?

VanessaE, 2020-06-26:
Could you please add a config option to disable this reporting?  In prints generated by Slic3r (and derivatives) with autocooling, this feature spams the host's console with speed changes.

It would have been better to include the speed in the M105 report instead.  Besides, some hosts like Pronterface already track the fan speed.

thinkyhead, 2020-07-01:
> Could you please add a config option to disable this reporting?

I assume you are referring to f3bf343c89a by @guruathwal ?

VanessaE, 2020-07-01:
Yep, that's the one.

thinkyhead, 2020-07-15:
#18654

thisiskeithb, 2021-07-05:
Added in #18013 / #18654. Enable `REPORT_FAN_CHANGE`.

closed 2021-07-05 by thisiskeithb