MarlinFirmware/Marlin #9847: UBL map display garbled with temperature auto-report

plehong, 2018-02-27:
I'm using Marlin on the bugfix-1.1.x branch with a RAMPS board connected to Repetier-Host running on Windows 10. `AUTO_REPORT_TEMPERATURES` and `EXTENDED_CAPABILITIES_REPORT` are enabled in `configuration_adv.h`

While trying to level the printer hotbed, I was surprised to see that parts of the display map printed with `G29 T` is very often missing. After spending a lot of time to track down the problem, I discovered that:
- When I connect the printer to Repetier-Host, it sees the `AUTO_REPORT_TEMPERATURES` capability and automatically make use of it. I found no way to disable it in the configuration panels.
- The line containing temperature auto reports is automatically suppressed in the log display if the `ACK` toggle is unchecked.
- Since the `unified_bed_leveling::display_map` method contains a call to the `idle` function, temperature auto reports are sometimes printed in the middle of a line in the bed level map, causing the beginning of that line to be wiped out by the ACK filtering feature of Repetier-Host.

I don't know the purpose of that call to the `idle` function. I suspect it's intended to refresh the LCD if it's the output target (ie if `map_type == 2`).

I think there are two ways to solve the problem:
1. Call `idle` only if `map_type == 2`.
2. Call `idle` only after EOL.

I've used solution 2 and it worked for me, but I might have broken something else. So, what would be the corrrect fix ?

thinkyhead, 2018-02-28:
The best solution is to temporarily disable the auto-report by setting its interval to 0 before map display and then restore it to its original value after map display.

TheSFReader, 2018-02-28:
Or we could add a flag in the auto-report(s) that, when raised from outside for the duration of a lengthy serial output, would mute the report (note that it should also be added for the auto-sd-status)

thinkyhead, 2018-02-28:
We're on the same page. I have updates to post tomorrow…

closed 2018-03-07 by thinkyhead