MarlinFirmware/Marlin #9848: [FR] Delay start after N hours N minutes from now

452, 2018-02-28:
### Feature Request
I want to start print my 3D Model e.g at 07:30 AM automatically. (because I live in apartment house and I do not have a garage to get rid of the noise of the 3d printer)

Be very nice to have menu options to start delay printing

**"Delay print"** -> **select model from sd card** e.g test.gcode -> **select delay e.g 05 min from now** or 24 H 09 min from now -> **start**
show and update time information "**printing will begin after e.g 09 h 03 min from now, model test.gcode**"
(in delay printing bed and extruder and fans off - standby mode, enabled by gcode (fans return to previous state when start print))

Studiodyne, 2018-02-28:
Francais : Si tu as un lecteur SD , tu mets le mode  AUTO START , et tu achetes un plug prise wifi 20€ , tu attends l'heure , et lorsque c'est l'heure , tu allumes le plug prise wifi

Et ca ira , et l'avantage , c'est que tu peux aussi mettre une camera ip , et tu peux surveiller que tout se passe bien et si ca merde , tu peux couper l'alimentation ,  ou si tout se passe bien , tu peux couper l'alim quand c'est fini. Economique aussi de courant , et de materiel allumé pour rien 

English; Buy a wifi wall plug , and use SD auto start file ,   power on the plug with your phone at the desired time ,  you can use cam to know if something wrong happen , and you can stop in emergency or when the print is finished . Economy of energy and material

Studiodyne, 2018-02-28:
Si t'as le wifi , un plug prise wifi , et le mode auto start sur carte sd
Si t'as pas le wifi , prise programmable ' si mecanique alors seulement 24h de print '
Si t'as le wifi , tu peux mettre le boitier octoprint , gerer par l'application telephone + boitier prise wifi , et une cam pour surveiller , avec le wifi l'avantage , c'est que  tu allumes le plug , ensuite tu demarres l'impression grace a octoprint ou par sd auto ,  l'avantage d'octoprint , c'est de pouvoir laisser l'imprimante prete a imprimer , pour pouvoir la lancer a l'improviste n'importe quand 
On est a l'epoque des energies durable , laisser l'alim de l'imprimante bouffer 50 w au repos pendant plusieurs heures juste parce qu'elle attend une instruction , c'est pas serieux  , et meme si tu pirates le compteur EDF lolll , c'est surtout pour pas abimer l'alim et laisser l'imprimante allumée surtout

Cette requete est excellente mais elle impose de laisser l'imprimante allumée  , il faut donc un truc externe qui reveille tout cela ,  plug wifi + octoprint , mais ca demande internet pour etre super top

Steeve

shitcreek, 2019-09-10:
Currently we support G4 to dwell/wait before continuing - http://marlinfw.org/docs/gcode/G004.html. The better solution though is probably to go with OctoPrint.
I wouldn't recommend having a print run on its own unattended - this could be disastrous.