MarlinFirmware/Marlin #9851: Z_SAFE_HOMING_X_POINT can't be reached by the nozzle

toenish, 2018-02-28:
Hi,

I have been using an older version (1.1.0-RC8) of Marlin a long time and till now everything had worked well. I had to do a small change and suddenly had problems with the T1 acceleration (similar to [#5967](https://github.com/MarlinFirmware/Marlin/issues/5967), but mine is way too slow). I downloaded the latest version of Marlin and two problems arise. First one will be in this thread and the second one will be in another one.

------------------------------------------------------------
### Compile Error
When I compile I get an error:
```
"Z_SAFE_HOMING_X_POINT can't be reached by the nozzle"
```
I'm using the same bed limits as before, but now it's giving me an error. I made a fast workaround and was able to ignore the problem, but I'm still curios to know how to avoid it.

I use Z_SAFE_HOMING and my parameters are the following:

#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// The size of the print bed
#define X_BED_SIZE 200
#define Y_BED_SIZE 200

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -42
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT (-11)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT (100)    // Y point for Z homing when homing all axes (G28).
#endif

Maybe somebody could help me with this issue. Thanks.

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1766791/Configuration_adv.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1766792/Configuration.zip)

thinkyhead, 2018-02-28:
Since your setup is unusual and you prefer to home Z off the bed (electrical contact?) you can just comment out the line in `SanityCheck.h` that generates that error.
```cpp
//      #error "Z_SAFE_HOMING_X_POINT can't be reached by the nozzle."
```

thinkyhead, 2018-02-28:
Ok, now patched. Thanks for the report!

commit 8c03ce4d (2018-02-28): Fix Z safe homing sanity check for no probe

Fix #9851
commit a18d6171 (2018-02-28): Fix Z safe homing sanity check for no probe

Fix #9851
closed 2018-02-28 by thinkyhead