MarlinFirmware/Marlin #9853: 20x4 LCD init issues 1.1.7/1.1.8

riverratdc, 2018-02-28:
I'm using a Prusa i3 clone with MKS GEN 1.2 electronics, inductive probe with bi-linear leveling & a 20x4 LCD defined as REPRAP_DISCOUNT_SMART_CONTROLLER in Configuration,h. Beginning with release 1.1.7 and persistent in release 1.1.8, three symbols on the LCD init incorrectly over half the time. The affected symbols are the degree symbol used with extruder and bed temperatures on LCD line 1, and the feedrate symbol in the first character position on line 3. The degree symbols are replaced with what appears to be a superscript "m", while I'm not quite sure what to call the symbol that replaces the feedrate symbol (the lower left quadrant of the character cell appears to be the original bitmapped "R", but I'm not sure what's in the upper left quadrant of of the cell). This symbol corruption can happen with either a cold, power-on reset or a warm, soft reset via USB-connected host. My current solution is to just keep resetting the printer until all three symbols display correctly (kinda' like playing a Prusa slot machine). 1.1.7/1.1.8 work great in all respects on my printer with the exception of the LCD init issues. Reversion to 1.1.6 or earlier eliminates the problem.

In looking at the changes implemented in version 1.1.7, I was drawn to issue #8526 [1.1.x] Fix up LCD init / charset as a possible culprit for my LCD issues.

[Marlin 1-1-8.zip](https://github.com/MarlinFirmware/Marlin/files/1767478/Marlin.1-1-8.zip)

gloomyandy, 2018-02-28:
I've also seen this problem with 1.1.8 and with bugfix-1.1.x I did not see it with 1.1.5. I can also see the problem after startup. Sometimes the initial display is fine, but on returning to the screen after being in one of the sub-menus I get the same "m" and other odd characters.

riverratdc, 2018-02-28:
Thanks, good to know. I rarely use the LCD sub-menus, mainly controlling the printer from host software. But your feedback indicates that the issue is not just confined to a reset init of the LCD. Points to code that basically runs anytime the LCD info screen is completely refreshed.

gloomyandy, 2018-02-28:
Thing is it seems somewhat random as to when it happens, so perhaps this could be some sort of noise issue? I know that the machine I have (Hictop 3dp-17) no longer uses the SD card reader in the LCD/control unit because the data would sometimes end up being corrupted. Instead they switched to a different main control PCB that has an SD card reader on it. I wonder if anything changed which means that the user defined characters are being defined either more or less frequently than before? Less often means that if they do get corrupted then they stay that way for longer, more often means bigger chance of a problem? I wonder if it is possible to read back the current contents (to allow it to be checked, or at least to help debug what is going on)? If I get chance I'll try and work out what has changed!

thinkyhead, 2018-02-28:
Duplicate of #9741

closed 2018-02-28 by thinkyhead

thinkyhead, 2018-03-02:
When an issue is marked as a duplicate it's not meant as a criticism. It's just for us maintainers to reduce the number of open issues for our own sanity and to help users locate interrelated issues.

cjsoong, 2018-03-06:
bugfix 2.0x also have the same problem
Quickly switch to the main menu and the info screen can duplicate this problem.
[image]

Bob-the-Kuhn, 2018-03-07:
The garbled characters are caused by an initialization problem,  See PR #9981 for details.

Here's the corrected 2.0.x file from the PR.
[ultralcd_impl_HD44780 2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/1788873/ultralcd_impl_HD44780.2.0.x.zip)

A 1.1.x PR will be done later Wednesday.

Bob-the-Kuhn, 2018-03-08:
Please try the latest bugfix.  The changes have been merged into 1.1.x and 2.0.x so the problem should be fixed.

riverratdc, 2018-03-08:
Last night's bugfix (3/7/2018) has yet to display any corrupted custom characters through roughly 50 printer resets. I'll keep giving it a workout, but the LCD issue appears resolved in this code.

Bob-the-Kuhn, 2018-03-08:
The issue, at least for me, only shows up when switching displays by clicking the encoder button.  I've never seen it on power up or reset.

riverratdc, 2018-03-08:
As I stated in the original issue above, a reset was the only place I did see the corruption (about half of all resets resulted in corrupted degree & feedrate custom chars.). I rarely use the encoder button for LCD sub-menus, relying instead on host software for printer control tasks.

Bob-the-Kuhn, 2018-03-08:
😉 What!!   You want me to read **AND** remember?!!!??  

I'm glad it works for you.  

I'm scratching my head as to why you were seeing it on reset.  I'm guessing that you have a custom screen enabled that comes up between the Marlin screen and the info screen.

riverratdc, 2018-03-08:
No custom screens, just the Marlin default boot screen with a customized printer name ("Sunhokey 2015 i3").

Thanks for the fix!