MarlinFirmware/Marlin #9856: Delta printer chaotic behavior on z probe error

ddenke8, 2018-02-28:
- Description: 
I have an flsun delta xl mk2 printer with marlin 1.1.8
When doing delta auto probing after the probe hit sometimes the probe does not reset to idle.
This causes chaotic behaviour in Marlin, instead of stopping or going up, it goes haywire, and crashes the head into the bed or collides with z pillar.

- Expected behaviour:
The head should keep rising while maintaining X, Y coordinates

- Actual behaviour:
Chaotic movement, head crash

- Steps to reproduce:
Start a delta auto calibration, then after a hit keep the z probe in active status

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1767608/config.zip)
[log.txt](https://github.com/MarlinFirmware/Marlin/files/1767632/log.txt)

thinkyhead, 2018-03-01:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

ddenke8, 2018-03-01:
I can confirm that bugfix-1.1.x is affected too. I used the configuration example which came with bugfix-1.1.x git clone ony changed a few unrelated options like board type, delta rod length, machine height, and steps/mm.

If the z probe functions correctly then the movements are smooth, precise and always makes slow contact with the bed. However if the endstop does not work then as if the steppers would get contradictory commands: their movement is non-linear / chaotic, and marlin loses the head position vevn without a missed step. This happenes even if the head does not crash anywhere.

The movement mechanics is otherwise good, produces good multi hour prints without a layer shift, the pulleys cannot be moved by hand if steppers are engaged etc.

thinkyhead, 2018-03-02:
I assume you're referring to `G33` delta auto-calibration. (I see you have no bed leveling enabled.)

When the `probe_pt` function fails to probe correctly it will say "Probing failed" and then return a "magic number" called NAN (not a number). My guess is that the `G33` code is not handling the failure gracefully. It should be checking for NAN and should abort the procedure or maybe re-try the probe.

@LVD-AC — Would you mind having a look at `G33` to see whether my hypothesis holds true?

LVD-AC, 2018-03-02:
This looks more a hardware problem to me of a probe switch that is stuck:

> the probe hit sometimes the probe does not reset to idle
> however if the endstop does not work then

G33 aborts just as it should at the 1st NAN received and spits out an error message on top of "probing failed" (since 1.1.5 if I remember it well). At the moment NAN is only returned if the probe does not trigger within set limits. But when the probe is stuck a real (but wrong) value is returned, not NAN. We could think of some extra testing during the probing routine e.g. test if the probe is still triggered after retracting and if so message "probe stuck" and `return NAN`.

thinkyhead, 2018-03-05:
Sounds like a good companion to the "abort if a probe point is way off" concept.

boelle, 2019-02-20:
@ddenke8 has this not been "fixed" in latest bugfix 2.0?

closed 2019-05-16 by thinkyhead