MarlinFirmware/Marlin #9862: Missing macro definition in Bed Skew Compensation

golobodo, 2018-02-28:
I tried that skew correction thing and I saw, that there is a missing macro definition in the (example) Configuration.h files: 
```cpp
#define XZ_SIDE_AD 200
```

Could you comment out the factors and logically rearrange the macros, too?

That would be what I mean:

```cpp
//#define SKEW_CORRECTION

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 282.8427124746
  #define XY_DIAG_BD 282.8427124746
  #define XY_SIDE_AD 200

  // Or, set the default skew factors directly here
  // to override the above measurements:
  //#define XY_SKEW_FACTOR 0.0

  //#define SKEW_CORRECTION_FOR_Z
  #if ENABLED(SKEW_CORRECTION_FOR_Z)
    #define XZ_DIAG_AC 282.8427124746
    #define XZ_DIAG_BD 282.8427124746
    #define XZ_SIDE_AD 200 
    //#define XZ_SKEW_FACTOR 0.0
    
    #define YZ_DIAG_AC 282.8427124746
    #define YZ_DIAG_BD 282.8427124746
    #define YZ_SIDE_AD 200
    //#define YZ_SKEW_FACTOR 0.0    
  #endif

  // Enable this option for M852 to set skew at runtime
  //#define SKEW_CORRECTION_GCODE
#endif
```

thinkyhead, 2018-03-01:
`XZ_SIDE_AD` is not included because it would always be the same as `XY_SIDE_AD`. It is simply the length of the front edge of the test object. It seemed messy to name it `XY_AND_XZ_SIDE_AD`.

closed 2018-03-01 by thinkyhead

golobodo, 2018-03-01:
But what, if I want to print skew towers in different sizes, because of a non-square bed like 200x300mm?
#define XZ_SIDE_AD could be 300 and #define YZ_SIDE_AD could be 200.

thinkyhead, 2018-03-01:
How would you fit a 300mm wide object on a 200mm wide bed?