MarlinFirmware/Marlin #9863: [2.0.x] MKS SBase SW SPI to TMC2130 not functioning. Certain pin headers don't work with SW SPI?

thepolodoc, 2018-02-28:
Is there a correct pin header for the MKS SBase to communicate over SW SPI to TMC2130 drivers as I have tried J8 and J7 pin headers with no effect so far. When I use these pins and have them defined in the SBASE Pins file as required by the compiler, I either get a driver error on the RRD display if that option is enabled or when it is disabled, the automatic current setting going into the tens of thousands! M122 seems to show that the drivers are not communcating properly because the driver registers are 0. 

Are there pins with the MKS Sase and TMC2130 drivers that other people are using to success and what is your pin setup for that? I do have the TMC2130 drivers setup as per issue thread #8971

yellobello, 2018-03-01:
Hi, I was just about to aks kinda the same question, namely which pins to use as SPI pins when I want to use TMC2130s whith my MKS SBASE.
My idea was to use a grbl board to put the 3 TMC2130s and then connect the step wires to the grbl board, plus then wire the spi pins... @thepolodoc how did you do it?

thepolodoc, 2018-03-01:
I used stepper breakout boards and made custom wires to connect the
required pins together on the tmc2130 and to the J8 pin header on the MKS
SBase. Then I connected the CS pin on each stepper (I'm using two) to it's
own dedicated pin on the SBase also on the J8 header. All the normal EN,
DIR, AND STEP wires are also connected to the stepper driver headers on the
SBase.

patternseek, 2018-09-10:
I'm pretty sure the pins in the current config file for 2.x bugfix don't work for SW SPI. At least I couldn't get them to. I got it working on HW SPI using the pin remapping in the file for the LCD SD card, but now I can't find enough pins to handle CS for 4 drivers. None of the pins that I've tried so far that aren't on J8 work.

gloomyandy, 2018-09-10:
What sort of display do you have?

None of the pins in J7 are available if you have an LCD display connected. They are basically all used for the LCD (via a level shifter). This leaves you with J8 and AUX-1. At the moment if you want to use the SD card reader on the LCD display then you need most of the pins in J8 for that (with a custom cable). The definitions in the pins file and in the spi_pins.h file in the HAL are basically setup for this. If you don't want an LCD based SD card and want to use the J8 pins you may need to make changes both to the pins files and to files in the HAL. 

 I'm working on a PR that will free up those pins. You may be able to get things working by using unused limit pins for CS, but remember those pins may have extra pull up resistors on them.  Basically the SBase board is pretty limited when it comes to adding expansion options (which I suppose given that it comes with built in drivers is not a big surprise). I opted to use TMC2208s with hardware settings to use stealth mode rather than using software control.

patternseek, 2018-09-10:
Oh also, I get compile errors if I try to use P0_02 or P0_03 ("Marlin/src/gcode/config/../../inc/../pins/pins_MKS_SBASE.h:219:28: error: 'P0_03' was not declared in this scope") apparently because "#define SERIAL_PORT_2 0" in Configuration.h is commented out which seems to stop the test "#if SERIAL_PORT != 0 && SERIAL_PORT_2 != 0" in pinmapping.h from passing. Removing the SERIAL_PORT_2 part of the test lets it compile but I don't know if this is a valid workaround or not.

patternseek, 2018-09-10:
Apologies for the spam. I did finally get SW SPI working by using these pins:

```
#define TMC_SW_MOSI    P0_03
#define TMC_SW_MISO    P0_02
#define TMC_SW_SCK     P1_22
```
The first two needed a hack to pinmapping.h as outlined above to make the pins available. The last one is on J8 now because it doesn't work on the TH4 header. Moving it to J8 got the 2130s working.

yellobello, 2018-09-11:
And did you also got the display working together with the TMC2130s?
Best regards

patternseek, 2018-09-11:
The screen yes but with SD card support disabled. I'll post a diff of the relevant parts  my config when I get a chance in case it's useful to anyone else

yellobello, 2018-09-11:
Yes, please!
I have TMC2130s laying around here waiting to be used with marlin on the
MKS.
Where did you mount them on to be used? I am thinking of just using the
tmcs for X and Y and
mount them on one of these cheap grbl shields and then maybe power it with
24v.

 Best Regards

patternseek, 2018-09-11:
OK I'm not sure a diff is the best in this case but here are the relevant changes as far as I'm aware. This is compared to the bugfix-2.0.x branch as of late last week:

If you don't have SERIAL_PORT_2 defined then ensure P0_03 and P0_02 are defined by:

In Marlin/src/HAL/HAL_LPC1768/include/pinmapping.h change:
```
116: "#if SERIAL_PORT != 0 && SERIAL_PORT_2 != 0" to "#if SERIAL_PORT != 0"
200: "#if SERIAL_PORT != 0 && SERIAL_PORT_2 != 0" to "#if SERIAL_PORT != 0"
254: "#if SERIAL_PORT != 0 && SERIAL_PORT_2 != 0" to "#if SERIAL_PORT != 0"
```

in Marlin/src/pins/pins_MKS_SBASE.h:

Comment out 
```
#define SCK_PIN            P1_22   // J8-2 (moved from EXP2 P0.7)
#define MISO_PIN           P1_23   // J8-3 (moved from EXP2 P0.8)
#define MOSI_PIN           P2_12   // J8-4 (moved from EXP2 P0.5)
#define SS_PIN             P0_28
#define SDSS               P0_06
```

Define the CS pins for the TMC2130s (edit the existing values):
```
  #define X_CS_PIN         P1_23
  #define Y_CS_PIN         P2_12
  #define Z_CS_PIN         P2_11
  #define E0_CS_PIN        P4_28
//  #define E1_CS_PIN
```

Uncomment:
```
#define TMC_USE_SW_SPI
```
Change:
```
#define TMC_SW_SCK     P0_26   // TH4
```
to
```
#define TMC_SW_SCK     P1_22  // J8
```

Hope that helps.

patternseek, 2018-09-11:
So essentially the main gotchas were that P0_02 and P0_03 weren't defined because I had no second serial port, and I couldn't get TH4 to work for TMC_SW_CLK.

patternseek, 2018-09-12:
Oh and yes I've got it all in a CNC shield

hectori4502, 2018-12-04:
Please any one with a working SBASE tmc2130 SPI
I need  your 
configuration.h
configuration_adv.h
pins_MKS_SBASE
I need to compare and test becouse I already try all and I still lost.
thanks.

patternseek, 2018-12-11:
@hectori4502 Have you tried my changes outlined above?

Here are the files. See my notes above.

Configuration.h:
https://pastebin.com/8GzxjV1d

Configuration_adv.h:
https://pastebin.com/p0fMDdMu

pins_MKS_SBASE.h:
https://pastebin.com/ce56z34y

formu25, 2018-12-13:
Hello
I want to do the same system with the TMC2130
Who goes through J8
Prove you make a clean schematic with the modification to be done in marlin 😊

[image]

gloomyandy, 2018-12-13:
It would really help if when proposing changes you based your configuration on the most recent pins.h file. It would also help if you explained why you have chosen to use a particular pin. So for instance I'm not sure why in the above configuration you have configured...
``` C
#define SCK_PIN P0_15 //P1_22 // J8-2 (moved from EXP2 P0.7)
#define MISO_PIN P1_31 //P1_23 // J8-3 (moved from EXP2 P0.8)
#define MOSI_PIN P0_18 //P2_12 // J8-4 (moved from EXP2 P0.9)
```
rather than simply using the default shared SPI lines at...
``` C
#define SCK_PIN           P0_07
#define MISO_PIN         P0_08
#define MOSI_PIN         P0_09
```

hectori4502, 2018-12-14:
with these pins the printer takes time to connect via usb (like 8 seconds)
and it's slow to receive commands
#define SCK_PIN            P0_07
#define MISO_PIN         P0_08
#define MOSI_PIN         P0_09
With these pins is fast.
#define SCK_PIN P0_15 //P1_22 // J8-2 (moved from EXP2 P0.7)
#define MISO_PIN P1_31 //P1_23 // J8-3 (moved from EXP2 P0.8)
#define MOSI_PIN P0_18 //P2_12 // J8-4 (moved from EXP2 P0.9)

gloomyandy, 2018-12-14:
Without seeing all of how you have your pins file configured it is not easy to work out what is going on. You really need to try the standard pins file as the various settings will interact with each other. In particular the SD card configuration will impact how the USB code operates as the SD card will be being shared with your PC. In particular we need to see what values you have defined for 
SS_PIN
ONBOARD_SD_CS
SD_DETECT_PIN
SHARED_SD_CARD

Have you actually tried an up to date pins file?

It is still not at all clear what you are trying to do by choosing those pin settings what do you actually have connected to those pins? Do you have an external SD card reader? Do you have SD card access defined?

hectori4502, 2018-12-14:
Hi gloomyandy,
I use a TFT35 thats came with a sd card. (EXP1 and EXP2 free)(My printer is a Rostock max V2 JK42HM-48 motors 0.9ºMKS SBASE 1.3)

I do the same config with  my MKS BASE GEN 1.4 and every things work fine.
Now my printer it´s working back with 200 steps by mm (I never knew why).

here my configuration files SBASE
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2681260/Marlin.zip)

I had to deactivate stealchop and // # define HYBRID THRESHOLD
to be able to control the stepers although I keep losing steps.

gloomyandy, 2018-12-14:
Your pins.h file is very different to the current versions. It would really make sense to redo your file and base it as much as you possibly can on the current configuration. If you don't then there is a good chance that updates may cause you problems as you will be using non-standard pins for things. I'm pretty sure that is what was causing your USB issues. Using a set of pin definitions that are so different from the standard ones will make it much harder for anyone to help you if you run into problems.

TommyLX, 2018-12-17:
Hi there,

I’m planning using MKS SBASE with touchscreen TFT35 and TMC2130 drivers over SPI control.

Is this even possible?

hectori4502, 2018-12-17:
Hi TommyLX,
For the TFT35.
set this in configutation.h
#define SERIAL_PORT -1  // USB
#define SERIAL_PORT_2  0 // MKS TFT35
connect to AUX-1 (I still don´t know where to connect the reset single pin from tft35)

hectori4502, 2018-12-17:
Hi gloomyandy,
as I have problems compiling. I use an old firmware that I have from October 4th that lets me configure M43 to check the pins. (in this firmware is when I have to multiply steps x16, from 200 to 3200)
with the new firmwares the steps are 200 but when compiling I receive many warnings (the same as the SBASE examples give me).

gloomyandy, 2018-12-17:
I've already given you what I consider the best advice I can. Start again with an up to date configuration and think carefully about all of the changes you are making. So for instance the TFT32 does not have an SD card that can be accessed by Marlin directly. Because of this you should almost certainly consider using the (new) default configuration of sharing the on board SD card with Marlin. If you do that (and you are using an up to date pins file), then you should have no need to mess around with any of the SPI pins they should all be set correctly for you. This default configuration should also leave as many of the other pins as possible available for use by the TMC devices.

formu25, 2018-12-18:
Hello 
Can you get some pictures of the connections plz :)

hectori4502, 2018-12-18:
https://cdn.thingiverse.com/assets/a6/03/31/95/98/MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf

formu25, 2018-12-18:
Sry "Can you get some pictures of the connections plz :)" for MKS sbase v1.3

hectori4502, 2018-12-20:
this original pins used in Marlin-Bugfix-2.0 December 19  works perfect .(using 24v I only use current 750 with 12v used to need current 1200 )

#define CHOPPER_TIMING CHOPPER_DEFAULT_24V
#define HYBRID_THRESHOLD

#if HAS_DRIVER(TMC2130)
  // J8
  #define X_CS_PIN         P1_22
  #define Y_CS_PIN         P1_23
  #define Z_CS_PIN         P2_12
  #define E0_CS_PIN        P2_11
  #define E1_CS_PIN        P4_28

// Hardware SPI is on EXP2. See if you can make it work:
// https://github.com/makerbase-mks/MKS-SBASE/issues/25
//#define TMC_USE_SW_SPI

I do not set LCD becouse a use TFT35
then the spi_pins.h set the sck mosi miso

#ifndef SCK_PIN
  #define SCK_PIN           P0_15
#endif
#ifndef MISO_PIN
  #define MISO_PIN          P0_17
#endif
#ifndef MOSI_PIN
  #define MOSI_PIN          P0_18
#endif

[image]

hectori4502, 2018-12-24:
Hi My Delta printer it´s working fine. XYZ tmc2130 E0 LV8729 . I already have the diag pins.
But I want to test:
#define SENSORLESS_HOMING // TMC2130 only
#define SENSORLESS_PROBING // TMC2130 only
If any one have a good delta configs sensorless please send it to me.

thanks 
Hectori4502

gloomyandy, 2018-12-24:
Those are not warnings they are compilation errors. They don't really have anything to do with this issue title and so you should probably open a new issue for them.

hectori4502, 2018-12-24:
when I uncomment:
#define TMC_Z_CALIBRATION
get this:
Marlin\src\gcode\feature\trinamic\M911-M915.cpp: In static member function 'static void GcodeSuite::M915()':
Marlin\src\gcode\feature\trinamic\M911-M915.cpp:415:29: error: assignment of read-only variable 'soft_endstops_enabled'
soft_endstops_enabled = false;
^~~~~
Marlin\src\gcode\feature\trinamic\M911-M915.cpp:430:29: error: assignment of read-only variable 'soft_endstops_enabled'
soft_endstops_enabled = true;
^~~~
*** [.pioenvs\LPC1768\src\src\gcode\feature\trinamic\M911-M915.cpp.o] Error 1

formu25, 2019-01-02:
[image]
[image]

boelle, 2019-02-20:
@thepolodoc 
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

pjotter1986, 2019-03-12:
hi dear friends,
I have a **MKS Sgen v1.0**
with tmc2130 i am looking for a working marlin config.
can someone with the same specs upload their 
Pins.h 
configuration.h
Configuration_adv.h
and platformio.
thanks and kind regarts.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

shitcreek, 2019-07-06:
> @thepolodoc
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

closed 2019-07-06 by shitcreek