MarlinFirmware/Marlin #9875: SOFTWARE_ENDSTOPS

toenish, 2018-03-01:
Hi,

I have set that my printer dimensions are the following (I have the version from [here](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x)):
```
// The size of the print bed
#define X_BED_SIZE 200
#define Y_BED_SIZE 215

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -42
#define Y_MIN_POS -25
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200
```

Now, when I upload it to the printer and ask M211 I get the following:
```
echo:Soft endstops: On   Min: X-42.00 Y0.00 Z0.00  Max: X200.00 Y214.00 Z200.00
```

X and Z are OK, but Y is very strange. What could be the reason for such interesting behavior? Where could I change manually the soft end-stop values?

thinkyhead, 2018-03-01:
Do you get the same results after `M502`?

toenish, 2018-03-01:
Yes, the same results.

thinkyhead, 2018-03-02:
What version of Marlin are you using?

`Conditionals_post.h` does this:
```cpp
// Get the linear boundaries of the bed
#define X_MIN_BED (X_CENTER - (X_BED_SIZE) / 2)
#define X_MAX_BED (X_CENTER + (X_BED_SIZE) / 2)
#define Y_MIN_BED (Y_CENTER - (Y_BED_SIZE) / 2)
#define Y_MAX_BED (Y_CENTER + (Y_BED_SIZE) / 2)
```
This is where the software endstops get their default values.

toenish, 2018-03-02:
I'm using the version I downloaded from Github ([here ](https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x)) on Wednesday. I assume it's the latest version.

thinkyhead, 2018-03-04:
For more helpful responses, try the [Marlin Help Forum](http://forums.reprap.org/list.php?415) and the [Marlin Facebook Group](https://www.facebook.com/groups/1049718498464482/).

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-08-17:
@toenish i see you use 1.1.x, try with 2.0.x instead as 1.1.x is not updated anymore

boelle, 2019-08-19:
as this is a question or support Q rather than a bug i will close this one, should it be a bug we can always reopen

https://github.com/MarlinFirmware/Marlin/issues/14694

closed 2019-08-19 by boelle