MarlinFirmware/Marlin #9879: Single Z-Probe

toenish, 2018-03-01:
Hi,

before one could use `G30 S-1` for probing in the current location. Now with the latest version and with the same command nothing happens. Looking the g-code table one should use something like this:
`G30 [E<|>] [X<pos>] [Y<pos>]`. 
I tried `G30 E X-11 Y110` and nothing happens.
How should this command look like to get the singe z-probe in the defined point?
What should be set in the config file to enable the singe z-probing?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1771614/Configuration.zip)

thinkyhead, 2018-03-02:
If the probe is outside of the bed when the command is executed, or if the position given is impossible for the probe to reach, the command currently exits with no output. In all other cases, the probe should take place.

There is no `S` parameter to the `G30` command.

- `G30` by itself will probe at the current position, if possible.
- `G30 X100 Y100` will probe at X100, Y100 (probably possible).

The `E` parameter additionally makes sure to engage/stow the probe.

toenish, 2018-03-02:
I have been using an older version (1.1.0-RC8) of Marlin a long time and till now everything had worked well. I had to update the firmware to the latest version and now so many things are not anymore working or giving me problems.

The probing points are very well reachable. I tried using just G30 and still nothing.

I use Z_SAFE_HOMING and my parameters are the following:

```
#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// The size of the print bed
#define X_BED_SIZE 200
#define Y_BED_SIZE 200

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -42
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT (-11)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT (100)    // Y point for Z homing when homing all axes (G28).
#endif
```

thinkyhead, 2018-03-04:
> so many things are not anymore working or giving me problems

Please file a separate issue for each problem you encounter. But test `bugfix-1.1.x` beforehand.

thinkyhead, 2018-03-04:
> I tried using just G30 and still nothing.

And how about `G30 X100 Y100`?

toenish, 2018-03-05:
I tried different G30 combinations and locations.
It seems that as long X>0 (I tried X=100) it works as expected, I assume it's connected with the other issue I had ([#9851](https://github.com/MarlinFirmware/Marlin/issues/9851)).
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1779753/Configuration_adv.zip)

thinkyhead, 2018-03-09:
It's interesting. The check for probe reachability seems to be ok, but we could check that anyway. Make this change to `gcode_G30` and we'll see what it says:
```diff
- if (!position_is_reachable_by_probe(xpos, ypos)) return;
+ if (!position_is_reachable_by_probe(xpos, ypos)) {
+   SERIAL_ECHOPAIR("Not reachable by probe? X", xpos);
+   SERIAL_ECHOLNPAIR(" Y", ypos);
+   return;
}
```

toenish, 2018-03-09:
ok
Not reachable by probe? X-11.00 Y110.00
ok

thinkyhead, 2018-03-09:
Here's the problem. The function `position_is_reachable_by_probe` should indicate whether the probe can reach the given point _at all_. But instead it's presumptuously asserting that the probe must be within the range from `MIN_PROBE_X, MIN_PROBE_Y` to `MAX_PROBE_X, MAX_PROBE_Y`.

I'll poke around the code and see how this function is being used, and see whether it's okay to change it, or if I need to add a new function to do the correct check for individual probing only.

Either way I will have a patch soon…

thinkyhead, 2018-03-09:
Ok, so with the PRs linked above the `position_is_reachable_by_probe` only requires that the probe be within the area you've specified for the bed, and additionally that the nozzle isn't outside of the physical bounds set by the `[XY]_(MIN|MAX)_POS` settings. Assuming your X-11 is considered a part of the bed area, then your `G30` should work. If it still barks after this change, you'll just need to make sure that Marlin knows that the physical bed is at least 11mm to the left of X0.

closed 2018-03-09 by thinkyhead

toenish, 2018-03-09:
I tried the bugfix and now the soft end-stop problem seems to be OK (#9875).
Still for probing when X negative it doesn't work. It doesn't give me the error message either that you suggested tried earlier.

thinkyhead, 2018-03-09:
You've specified that the bed is 200mm square with 100,100 as the center. However, you are trying to put the probe at X=-11, a position which Marlin now believes is not over the physical bed. If your bed actually extends to -11 in X then does it also extend to +211?

Whatever the case may be, for your special situation you'll need to edit `Conditionals_post.h` and modify this block so marlin doesn't expect the front-left corner of the physical bed to be at X=-11.
```cpp
  #define X_MIN_BED (X_CENTER - (X_BED_SIZE) / 2)
  #define X_MAX_BED (X_CENTER + (X_BED_SIZE) / 2)
  #define Y_MIN_BED (Y_CENTER - (Y_BED_SIZE) / 2)
  #define Y_MAX_BED (Y_CENTER + (Y_BED_SIZE) / 2)
```

toenish, 2018-03-09:
OK, now I have a new problem. I changed them to 
```
  #define X_MIN_BED X_MIN_POS //(X_CENTER - (X_BED_SIZE) / 2)
  #define X_MAX_BED X_MAX_POS //(X_CENTER + (X_BED_SIZE) / 2)
  #define Y_MIN_BED Y_MIN_POS //(Y_CENTER - (Y_BED_SIZE) / 2)
  #define Y_MAX_BED Y_MAX_POS //(Y_CENTER + (Y_BED_SIZE) / 2)
```
and that seems to make the trick.
BUT the new issue: the safe homing is defined at X-11, when I'm doing homing Marlin uses X-21. Should I open a new issue for that?

thinkyhead, 2018-03-09:
That would certainly do it. Note that those changes will also move the software endstops, and will also affect the behavior of Corner Leveling (if you need that).

thinkyhead, 2018-03-12:
I've made some additional changes that are hopefully going to settle all this probe boundary stuff more definitively.

There's a setting `MIN_PROBE_EDGE` that used to only apply to bed leveling. I've promoted it to a general option that can apply to all probing.

Note that there are also settings, `MIN_PROBE_X`, `MAX_PROBE_X`, `MIN_PROBE_Y`, `MAX_PROBE_Y` that can be used individually to override the probe-able region for rectangular beds. By default they are set to the bed boundaries, but inset by the `MIN_PROBE_EDGE`.

If you set `MIN_PROBE_EDGE` to 0 then the probe is allowed anywhere within the bed — so basically 0 to 200 in both dimensions. If you want to be able to probe past the bed bounds for any reason, the settings to alter are the aforementioned `MIN_PROBE_X`, `MAX_PROBE_X`, `MIN_PROBE_Y`, `MAX_PROBE_Y`.

This sort-of change in PRs #10068 and #10069 will be merged soon.

PR #10020 (2018-03-09): [2.0.x] Have position_is_reachable_by_probe use the whole bed
Fix #9879

Previously `position_is_reachable_by_probe` was limited to the area specified as the bounds for `G29` mesh leveling (even if leveling was disabled). This change modifies the function to properly consider the entire bed area so that `G30` and other non-leveling probing may take place.

Counterpart to #10019
PR #10019 (2018-03-09): [1.1.x] Have position_is_reachable_by_probe use the whole bed
Fix #9879

Previously `position_is_reachable_by_probe` was limited to the area specified as the bounds for `G29` mesh leveling (even if leveling was disabled). This change modifies the function to properly consider the entire bed area so that `G30` and other non-leveling probing may take place.

Counterpart to #10020