MarlinFirmware/Marlin #9880: Clarify documentation for M18, M84 - Disable steppers

bcsteeve, 2018-03-01:
Currently this is the wording:

>  If no steppers are specified, this command disables all steppers **immediately**.
> 
> If one or more axes are specified, this command disables the specified steppers **immediately**.

I added emphasis.  This isn't true, is it?  I wish it was.  But say I issue a G28 (or any G0) and then issue an M84... at least in my testing, it certainly doesn't stop the steppers "immediately".  Immediately after it completes whatever is in the queue, sure... but then that's not very immediate.

From the documentation, I had hoped that M84 could be used as a sort of "soft" M112.  M112, unfortunately, is a harsh overreaction to many little mistakes (forgetting to include F in a G0/1 and having to painfully wait for a minute as the stepper slowly makes its way).  So I really would like to use M84 as stated... to "immediately" disable the steppers.  But it doesn't appear to work that way.

So please either remove the terms "immediately" and replace with a more accurate description, or else clarify what is required for it to actually halt the steppers upon issuance of the command (ie. is there a "requires"?)

Cheers.

comps, 2018-03-02:
Just FTR, for stopping the steppers immediately (together with EMERGENCY_PARSER), there's M410.

bcsteeve, 2018-03-02:
Yes, I know that, thanks. I should have added that to my post to be helpful to future visitors.

thinkyhead, 2018-03-02:
The `M18` (`M84`) command waits for the current move in the queue to finish, then turns off power to the specified steppers. If there is another G-code move in the buffer, Marlin will faithfully power the steppers back on so that it can produce the move. The purpose of the command is not to cease movement or halt printing or cancel the current activity. It is to turn off steppers immediately (when the `M18` command comes up in the queue).

This will move X to 100 and then turn off the steppers:
```gcode
G1 X100
M18
```
This will move X to 100 and then turn off the steppers, and then turn them back on again and move X back to 0, leaving the steppers on:
```gcode
G1 X100
M18
G1 X0
```

Ultimately, the purpose of `M18` is to turn off the steppers when you're done with them. The command also allows you to set the period of time that the steppers should remain powered after finishing moves.

This will move X to 100, and then after 3 seconds the steppers will turn off:
```
M18 S3
G1 X100
```

bcsteeve, 2018-03-02:
Sure, that's fine and well explained.  It still isn't an appropriate use of the term, "immediately".  The goal of documentation is clarity, yes?  The word serves no purpose in your explanation.  Every command is executed "immediately... on its turn".  The terminology serves only to confuse... to hint that it *can* be used for the purposes you and I know it can't.

Adopt or don't. I stand by my suggestion.  It is simple enough to remove the two instances of the word "immediately".  At best, they offer nothing.  At worst, confusion.

thinkyhead, 2018-03-04:
Thanks for the suggestion. If you'd like to simply contribute to the documentation and cut out the middle-man, it is also open source, and we very much need contributors with a passion for good communication. https://github.com/MarlinFirmware/MarlinDocumentation

closed 2018-03-04 by thinkyhead

bcsteeve, 2018-03-04:
Oh, OK... I had no idea I could directly help.  There were a few simple typos I mentioned too, and there's no harm cleaning that up.