MarlinFirmware/Marlin #9882: [FR]Prevent filament runout false positives

crystalhand, 2018-03-01:
## Feature Request
I have run into some issues with the filament runout sensor falsely triggering using a mechanical endstop.  This seems to be relatively rare at about 1-2 false alarms every 20 hours of printing.  The sensor trips only for a second but that is enough for it to go into its filament change routine.  Would it be hard to set up a time/filament length amount to poll the runout sensor to ensure it is a genuine out of filament situation?  IE 
```
X= length between extruder and filament sensor /2
if filament sensor=tripped
   check again every X/5 extruded amount   **ie checks 5 times 
   if some threshold for filament sensor tripping=true 
          then genuine runout run M600
          else false positive runout and ignore
```
This would allow you to use the distance between the filament sensor and the extruder motor as the max distance for the false positive checking.  this would ensure you do not extrude without filament.

Studiodyne, 2018-03-02:
Add a capacitor on you sensor , it will ensure that all furtive peaks or spikes will be down

thinkyhead, 2018-03-05:
@crystalhand Your idea is sensible. We should at least do the typical "de-bounce" as we do with other switches. Out of curiosity, is the filament sensor a "normally closed" (NC) switch or a "normally open" (NO) switch. Most mechanical endstop switches are NC, while most optical ones are NO. NC switches are preferred because they are far less prone to errors (since they aren't susceptible to inductance).

thinkyhead, 2018-03-05:
Ok, I've incorporated a de-bounce into the filament runout sensor logic. Once it passes our build tests, the the changes will be merged into the bugfix branches. Thanks for bringing this to our attention!

thinkyhead, 2018-03-05:
Give the latest code a try and see if it works more reliably now. The runout sensor now has to be triggered for 5 consecutive checks before runout will be considered valid.

crystalhand, 2018-03-05:
Awesome thanks a ton for doing this.  I will flash this firmware to my machine and give it a try tonight.

thinkyhead, 2018-03-09:
> It seems that if the filament has a bend in just the right direction the mechanical switch has enough room to open.

Hopefully not too often or for too long. We can only do so much to mitigate hardware faults.

simonvez, 2019-02-15:
so I run 1.1.9 and I still have this issue? was it corrected ?

boelle, 2019-02-20:
@crystalhand still an issue

use latest bugfix 2.0

1.1.9 is outdated

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
This can be closed. Marlin 2.0 has several things to improve this including a runout distance which acts as a very wide debounce as well as the ability to disable the runout sensor if its acting up too much directly from the resume screen before continuing.

boelle, 2019-07-21:
closing this one as @InsanityAutomation says a lot has been added to improve this

closed 2019-07-21 by boelle