MarlinFirmware/Marlin #9884: Marlin-bugfix-1.1.x ~ TMC2130 Software SPI Not setting stepper current, not homing (sensorless endstops on Y)

thegrazingcow, 2018-03-01:
------------------------------------------------------------
### Bug Report
**- Description:** 
When using software SPI pins with a TMC2130 driver, the current is not set correctly. 

**- Expected behaviour:**
When a driver's current is set to 800mA, the set current should be roughly 800mA when using the M122 command.

**- Actual behaviour:** The "set current" becomes 1436mA. Whilst the set values seems to be scaling (non linearly) with the desired current, it's not even close. Similarly if the current is set to 700mA, the "set current" via M122 becomes 1215mA.

**- Steps to reproduce:**
Hardware: Ramps 1.4, RepRap Discount Full Graphic Smart Controller, TMC2130 Watterott drivers on X, Y, E0.
In order to use the LCD display, it consumes the AUX3 ports so I needed to change the pins the TMC2130 connects to using the recently implemented software SPI feature. I was using this feature to swap my X, Y, and E0 CS pins to 6, 5, and 4 respectively, and the MOSI, MISO, and SCK pins to 40, 44, and 42 respectively (therefore not using any AUX3 pins). Trying to set the current in both the EEPROM and firmware both failed to work. Additionally I was experiencing homing issues, where the Y carriage simply wouldn't move when sent a home command despite the end stops supposedly being open (using sensorless endstops on the Y carriage only). Same behavior before and after M502.

**Software SPI:**
[image]

**Hardware SPI:**
[image]

**Section of configuration_adv.h being changed (WORKING when highlighted line is commented out, NOT WORKING when it is left as is):**
[image]

Removing the display and **using the hardware SPI instead of software SPI solved all issues.**

------------------------------------------------------------

teemuatlut, 2018-03-01:
Your problem isn't incorrectly set driver current, but rather that your communication doesn't work. You can see this from the reported driver registers where the answer has been 0x00:00:00:00.
I'll need to take a look if SW SPI is broken on the library or if there is something else that's wrong.

teemuatlut, 2018-03-01:
I've released library version 2.1.5 that should fix this. It may take some time before it shows in the IDE but you can update from github already if you want.
Let me know if it solves your problem!

psavva, 2018-03-01:
Maybe it's better to show an error like SPI Communication failure when you detect 00 on the registers. I've seen this issue come up too many times.

teemuatlut, 2018-03-01:
My [TMC_debugging_v2](https://github.com/teemuatlut/Marlin/commit/75b17f108632b1acc4aa7d60b4792ede26ec7bb6) branch.

thegrazingcow, 2018-03-02:
It's probably worthy of a new bug thread, but it's too late so I'll mention it here for now. When I have set the microsteps to 256 in the TMC2130 config section, when I then go to home the axis Marlin restarts without giving any error codes afaik. Resolution is as simple as changing the msteps to something lower, like 64 (and ofc then changing the steps/mm)

teemuatlut, 2018-03-02:
It is way too demanding. On a Mega the suggested microstepping value is 16.
32 would be the absolute maximum and only for X and Y.

thinkyhead, 2018-03-04:
> Resolution is as simple as changing the msteps to something lower, like 64 (and ofc then changing the steps/mm)

There's little we can do to prevent the reboot except maybe throttle the maximum step rate even lower. Currently the `MAX_STEP_FREQUENCY` for AVR is "40000" (but this is further restricted to 20000, with 4 steps per ISR call). If this number of ISRs is likely to lead to reboots, it may help to reduce it and accept a lower ceiling on the max step rate.

Rtully, 2018-11-01:
Hi there, Hope you can Help. I'm running TMC2130s with a MKS GEN L. I can not get the TMC drivers to accept the changes from marlin configuration_adv.h. No matter what settings it just keeps reading the same configuration. Running Marlin bugfix 2.0.x and TMC2130 library version 2.5.
Pulling my hair out for days with this one.....
I have never set anything for the TMCs in the EEPROM! It just defaults to the following:
Send: M122
Recv: 		X	Y	Z	E
Recv: Enabled		false	false	false	false
Recv: Set current	700	700	700	750
Recv: RMS current	673	673	673	734
Recv: MAX current	949	949	949	1035
Recv: Run current	21/31	21/31	21/31	23/31
Recv: Hold current	10/31	10/31	10/31	11/31
Recv: CS actual		10/31	10/31	10/31	11/31
Recv: PWM scale	0	0	0	0
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		79	79	658	263
Recv: [mm/s]		100.08	100.08	3.00	30.06
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		3	3	3	3
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs	-20	2	0	0
Recv: DRVSTATUS	X	Y	Z	E
Recv: stallguard
Recv: sg_result		0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:
Recv: 	X	0x80:0A:00:00
Recv: 	Y	0x80:0A:00:00
Recv: 	Z	0x80:0A:00:00
Recv: 	E	0x80:0B:00:00
Recv: 
Recv: 
Recv: ok

boelle, 2019-02-20:
@thinkyhead given the lack of response i think we can close this one unless @thegrazingcow makes noise

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-16 by thegrazingcow