MarlinFirmware/Marlin #9891: Z homing problem

ruggb, 2018-03-02:
In the process of trying to set up my nozzle probe, I changed something that causes the head to retract to 10 after homing. Repetier shows the position as 0.00 but the log shows Z @ 10. If I click Z10 (Repetier) the position changes to 10 but the bed doesn't move. The 2nd click will move it to 20 and the position is @ 20.
I tried changing the offset with 206, and that changed the log value but z is still retracting about 10mm
If it is showing z20 and I enter m206 z-10 it will show z0. it changes by 2x what I enter.
Repetier shows the position as 0.00 but the log shows Z @ 10. If I click Z10 (Repetier) the position changes to 10 but the bed doesn't move. The 2nd click will move it to 20 and the position is @ 20.
I tried changing the offset with 206, and that changed the log value but z is still retracting about 10mm
I tried reseting the eeprom - no joy.
What did I do? 
how do i fix it?

thinkyhead, 2018-03-02:
Do `M502` and then `M500` and see if it helps.

ruggb, 2018-03-02:
BTW, I reloaded the previous config and it is definitely something that I changed.

Also, can someone explain this. I just can't wrap my brain around it.
////////////////
* Z_MIN_PROBE_ENDSTOP
 *
 * Enable this option for a probe connected to any pin except Z-Min.
 * (By default Marlin assumes the Z-Max endstop pin.)
 * To use a custom Z Probe pin, set Z_MIN_PROBE_PIN below.
 //////////////////
1st - Enable this option for a probe connected to any pin except Z-Min.
 * (By default Marlin assumes the Z-Max endstop pin.)

it sounds like you are wanting me to enable this if I have it connected to any pin other than the default Z_MIN, then you tell me the default is Z_MAX???

2nd - If I am not using the "default" (whatever it is) you want me to set the pin "below". There is no setting, that I recognize, for Z_MIN_PROBE_PIN, below in this file.

ruggb, 2018-03-02:
@AnHardt ?? what is an RH before V2.0 ?? - are you telling me the probe function does not work in 1.1.x bug fix releases??

ruggb, 2018-03-02:
It looks like 
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
and 
#define Z_MIN_PROBE_ENDSTOP
are redundant since they cannot both be enabled.
What is the difference?

ruggb, 2018-03-02:
the deeper I get into this the more confused I am. 
If this is defined
Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
the conditionals_LCD change the inverting state of the pin for BLTOUCH.
Where does it state that in Configuration.h?

It is almost appearing like I can't use a piezo hotend mounted endstop/probe combination. Is that correct?

Maybe I should use Z_MIN_PROBE_ENDSTOP for my probe - though it is not telling me the other is for BLTOUCH

ruggb, 2018-03-02:
so the Z_MIN_PROBE_PIN is not interrupt capable and I have endstop interrupts enabled
I think there is an issue here. I have no idea where it is, BUT I can't use one option because it sets the pin to one that is not interrupt capable and I can't use the other option because it sets it back to non-inverting to accommodate a BLTOUCH.
My piezo endstop is activated by the hotend. My piezo is also the bed level probe. My piezo operates with the same active low signal as the other endstops that are microswitches. It is connected to the Z_MIN input on my Ramps. It works perfectly as an endstop with my non-probe config. So far I have not found a combination of enables that will make it work and it appears to be centered around the above 2 items.

AnHardt, 2018-03-02:
I just tried to tell you that 'Repetier' (i assume Repetrier Host, not Firmware or Server) in the current version is not evaluating our position feedbacks. The displayed numbers on Repetier Host are currently meaningless. (#8500)

ruggb, 2018-03-02:
@AnHardt thanks - I am using V2.0.5 - it may not be meaningless, but the nozzle still ended up physically about 10mm off the deck.

I just loaded the latest Marlin build 63b1358 and configured it. The Z home now stays on the deck when homed, however, there are still some issues. G29 goes to all three locations (3 point leveling set) at each one it drops the bed maybe by the Z_HOMING_HEIGHT then moves to the next location, never homing. At the 3rd location it homes then retracts about the Z_HOMING_HEIGHT. If I do a Z home it drops it first so I assume that is what is coming into play. So why isn't it homing at the first 2 points?

I also have Nozzle park enabled that worked in the previous build, but does not in this one.
I'll try loading an RH 1.x

AnHardt, 2018-03-02:
As long as you are confusing 'homing' and 'probing' neither communication nor help is possible for me. Sorry.

ruggb, 2018-03-02:
How would you like me to describe my piezo hotend mounted endstop/probe so we can communicate?

ruggb, 2018-03-02:
I now have RH 1.6.2 loaded. Z_HOMING_HEIGHT = 4
If I do a home all, XYZ are at 0
I run this script, or any similar going to any XY location
g0 z10 f5000
g0 x300 y5
g28 z0

Z goes to Z+4, then moves to XY then Z homes to 0 and ends at 4,  RH says 0
if I do a Z home or run the SAME script again it homes to 0
If I run one script after another it gets farther away from Z home with each script.

THE other point is that it ran perfectly until I tried to implement the probe section. I am not confused about the difference between PROBE and HOME. So what about the probe implementation is screwing up the homing function and causing the probe function not to?????????

closed 2018-03-03 by ruggb

thinkyhead, 2018-03-04:
@ruggb — Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

You can use `M114 D` as you move around the bed to see lots of details about the current position. If you see strange things, also copy them into your text file.

Since you're using `3POINT` leveling, you can also use `G29 Q` at any time to see details about leveling at the current position.

We look forward to your next report. Hopefully we can narrow this down to a single cause and fix it.