MarlinFirmware/Marlin #9892: [BR] Command M5 does not switch Spindle & Laser (attached to PWM port) off

shabalex, 2018-03-02:
### Bug Report
- Description: 
Configured Marlin 1.1.8 to control CNC (former CNC 1610) with spindle motor and laser head.
Attached to PWM port D6
- Expected behaviour: 
Command M3 S255 should switch spindle/laser ON
Command M5 switches it OFF
- Actual behaviour: 
Command M5 does NOT switch PWM OFF
Command M3 S0 does

what to fix:
Index: Marlin_main.cpp
===================================================================
--- Marlin_main.cpp     (revision 19)
+++ Marlin_main.cpp     (working copy)
@@ -6410,6 +6410,7 @@
   inline void gcode_M5() {
     stepper.synchronize();
     WRITE(SPINDLE_LASER_ENABLE_PIN, !SPINDLE_LASER_ENABLE_INVERT);
+    analogWrite(SPINDLE_LASER_PWM_PIN, SPINDLE_LASER_PWM_INVERT ? 255 : 0);                           // only wr
ite low byte
     delay_for_power_down();
   }

thinkyhead, 2018-03-07:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

pratap3369, 2018-07-25:
I'm also having the same issue.Using 1.1.x

shabalex, 2018-07-25:
I haven't had a chance to put firmware from bugfix-1.1.x to my Ramps, but according to the code changes this should fix the problem I noticed.
Thanks!

closed 2018-11-05 by thinkyhead