MarlinFirmware/Marlin #9893: [FR] Add switch laser&spindle ON with G1 command

shabalex, 2018-03-02:
I'm using Marlin 1.1.8 in configuration for CNC 1610 to control Spindle motor or Laser head attached to PWM port.
Currently I have to use command M3 S255 to switch the laser beam ON before doing command G1 X.. Y ..
Then I need to insert command M5 in gcode to switch the laser OFF before moving head with command G0
Then I need switch it again and so on.

I would like to have a feature  to always switch laser ON automatically when doing G1 and switch it off on G0

AnHardt, 2018-03-02:
That isn't that easy as it looks at the first glance. It has to be synchronized with the planner buffer.

rszasz, 2018-03-03:
Using the spindle speed control word S along with the G0 or G1 might be the best solution as it seems to be used that way in other G code implementations. Then add an M code to determine if the spindle speed stays set or returns to its previous setting when the move ends.

I think it would require some significant recoding, but would be useful.

That or adding a new canned cycle for laser on, move, laser off with minimal delays. Another thing that would require a different G code, but might be useful for lasers in particular would be the ability to ensure the desired power per area, either by bringing the tool up to speed before the start position then triggering the laser, or setting the laser power based on tool velocity.

boelle, 2019-02-20:
@shabalex tried what @rszasz suggested?

remember to use latest bugfix 2.0

1.1.8 has been put in the grave long time ago

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
There are quite a few around laser timing and improvements. Maybe for 2.1 planning at some point we should consolidate them into a single thread and set a list of goals to work off of.

LinoBarreca, 2019-08-05:
I need this too. I am using a laser to smooth PLA's surface while printing. Actually I workarounded this using (and connecting) the laser as an extruder (neither marlin not the slicer know it's a laser).
This is the best approach to what I need because I also need to set the laser offset from the nozzle (since it's not at the same location on the head)...
Definitely handling the laser/spindle/extruder in a generic way (like "tool") in which you can set the tool's "power" (spindle speed/laser output/extruder speed/output) would be the best generic approach.

shitcreek, 2020-06-14:
The new laser codes now allows for inline firing of the laser using G1-G5 - see `LASER_MOVE_POWER`
Also, G0 and G28 can both turn off the laser: see `LASER_MOVE_G0_OFF` and `LASER_MOVE_G28_OFF`

closed 2020-06-14 by shitcreek