MarlinFirmware/Marlin #9894: Marlin wont unzip properly

beemergb, 2018-03-02:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```
I am trying to download and upload Marlin 1.1.8 ,it seems ok but when I unzip it the Marlin Icon does not appear, just a Attribution card sighn ! What am I doing wrong ? Hope you can help ,thank you  
------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

ikarisan, 2018-03-02:
I cannot confirm that. Are you using Windows, Linux or MacOS?
If your system does not show an Arduino icon for .ino files this file type is not registered within your system!

For windows:

1) Download Arduino IDE 
1a) Windows: https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip 
1b) MacOS: https://www.arduino.cc/download.php?f=/arduino-nightly-macosx.zip
1c) Linux: https://www.arduino.cc/download.php?f=/arduino-nightly-linux64.tar.xz

2) Unpack the archive and copy the content to a location within your user directory
3) Start the Arduino IDE once (to make sure it is known by the system)
4) Windows: Right click on the .ino file, select "open with" and then "Arduino IDE"
5) The icon should have been changed to the Arduino IDEs one.

thinkyhead, 2018-03-04:
Downloads from here seem ok: https://github.com/MarlinFirmware/Marlin/releases

boelle, 2019-02-20:
@beemergb problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead