MarlinFirmware/Marlin #9895: Z-Probing fails after turning on ATX PSU

uuzi, 2018-03-02:
------------------------------------------------------------
### Bug Report
- Description: After issuing M80 to turn on my ATX PSU, the first homing (and only the first, second try is perfectly fine) works ok for X and Y but for Z it just moves along the X axis to the middle and stays at the Y-min position. 
- Expected behaviour: Hotend should move to middle of bed for Z-probing

Just to emphasize, this ONLY happens on the first homing after I turn on the PSU. Every homing try after that works just as expected, until I turn off the PSU again. After that the first try will fail again.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1776123/config.zip)

AnHardt, 2018-03-02:
Does it make a difference if you switch the power supply or if you just reset the processor?

golobodo, 2018-03-03:
Here was my workaround. Put it in your start gcode before homing:
G0 Y1 ; TMC2130 Y-axis workaround

uuzi, 2018-03-03:
> Here was my workaround. Put it in your start gcode before homing:
G0 Y1 ; TMC2130 Y-axis workaround <

strangely enough, that works. so there is a bug in the tmc2130 implementation?

golobodo, 2018-03-03:
It is already fixed in the bugfix1.1.x branch.

thinkyhead, 2018-03-05:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

uuzi, 2018-03-20:
confirmed working correctly in `bugfix-1.1.x`

closed 2018-03-20 by uuzi