MarlinFirmware/Marlin #9896: Manual Probe Levelling

zafff, 2018-03-02:
### Bug Report
- Description: Without any endstop for the Z-axis, I uncommented #define PROBE_MANUALLY in marlin and compiled. However, when I try to level my bed, the bed moves down by 1mm (Z = Z + 1mm). Based on a previous post, I also commented Z_HOMING_HEIGHT and Z_CLEARANCE_BETWEEN_PROBES. The bed no longer moves down when X and Y are homed but only for Z. This behaviour is the same regardless of what position Z is. If I try G28 Z, the bed moves down by 1mm and sets the Z coordinates as zero.

The EEPROM was erased beforehand.

- Expected behaviour: Bed does not move before bed levelling of if G28 Z is issued.
- Actual behaviour: Bed moves down by 1mm before bed levelling or if G28 is issued.
- Steps to reproduce:
  - Uncomment  #define PROBE_MANUALLY and either G28 Z or G29 S1 to level the bed.

###

AnHardt, 2018-03-02:
Likely it's `Z_CLEARANCE_DEPLOY_PROBE` what you see.

zafff, 2018-03-02:
That is plausible. I have #define Z_CLEARANCE_DEPLOY_PROBE  uncommented but the value is 10 at the end. I would assume based on the code that it would move by 10mm. Is that not the case?

thinkyhead, 2018-03-10:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them into your next reply. We're interested in what kind of probe you are trying to use for your Z homing. All mechanical and electronic probes require some kind of clearance before their use.

zafff, 2018-03-10:
Thanks for you reply. I'll post my configs later as it is fairly late now.

I am not using any probe for the Z axis. I uncommented #define PROBE_MANUALLY and use a piece of paper until I got the correct height and then issued a G92 Z0.
However, if I try to do a bed levelling, then the bed lowers right before the first point by 1mm. This messes the overall levelling because each point has an offset of 1mm in all the 9 values (for a 3x3 grid).

thinkyhead, 2018-03-10:
I haven't seen this behavior with `PROBE_MANUALLY`.

How do you home Z without an endstop?

zafff, 2018-03-10:
I use a piece of paper and move the axis until the paper catches. Then I run G92 Z0 to reset the Z position to zero. However, if I run G28, the X and Y axes home respectively. But an interesting thing happen for Z; it moves by exactly 1mm away from the nozzle! The same thing happen when I try to level my bed. It homes all the axes and then move 1mm away from the axes which messes my Z.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1799830/Config.zip)

thinkyhead, 2018-03-11:
You can't use `G92` like that because it affects the workspace offset.
Well, not without disabling software endstops or setting your `Z_MIN_POS` to a negative value:
```cpp
#define Z_MIN_POS -4 // Give some room for tweaking
#define MANUAL_Z_HOME_POS 0
```

zafff, 2018-03-13:
I noticed that when I do relative motion because I have had a few nozzle crashes recently because of this. If the printer has not been homed in a particular axis, then relative motion does not work for that axis. I disabled software endstops for my printer's Z-axis to avoid running into these issues since I do not currently have an endstop for that axis.
However, the issue that the bed moves down still persists.

zafff, 2018-03-19:
Thanks for the pointers. Can you shine some light on the purpose of having a machine/logical space? Would offsets not be an easier solution?
I had some time today and got the chance to flash the latest 1.1.x bugfix(today) and I enabled NO_WORKSPACE_OFFSETS. I disabled the software endstops (min+max) for the Z axis. 

I found that G92 no longer works unless I force a G28. If I do not force the G28, then I get the dreaded "echo:endstops hit:  Z:0.00" although my bed is at a real location Z=200. When the printer boots, it thinks that it is at Z=0 and refuses to go at any negative values.

thinkyhead, 2018-03-19:
The main issue is that Marlin requires endstops and requires `G28` to establish the current position. There is no command to force Marlin to think it's at the home position without a `G28`.

> Would offsets not be an easier solution?

That's how it's implemented: as offsets. Internally Marlin (since 1.1.7) always operates in its native workspace. But when offsets are applied via `G92` or via `CNC_WORKSPACES`, all coordinates received from G-code have the workspace offset subtracted. When coordinates are reported on the LCD or via `M114` the offsets are added.

boelle, 2019-02-20:
@zafff problem solved?

zafff, 2019-02-20:
I pretty much resorted to auto-bed leveling now and marlin has excellent support for auto leveling because of the fact that marlin does not know what position it is at startup. I would urge you to get a cheap inductive probe or even a switch that can be mounted on the side.

boelle, 2019-02-20:
@zafff so since you sorted it maybe click the greeb close botton?

closed 2019-02-20 by thinkyhead