MarlinFirmware/Marlin #9897: [FR] Automatic Skew factor calculation for xz and yz

jonathanze, 2018-03-02:
From my understanding:

> 

 
Marlin automatically computes skew factors from these measurements.  
 * Skew factors may also be computed and set manually:
 *
 *  - Compute AB     : SQRT(2*AC*AC+2*BD*BD-4*AD*AD)/2
 *  - XY_SKEW_FACTOR : TAN(PI/2-ACOS((AC*AC-AB*AB-AD*AD)/(2*AB*AD)))
 *
 * If desired, follow the same procedure for XZ and YZ.
 * Use these diagrams for reference:
 *
 *    Y                     Z                     Z
 *    ^     B-------C       ^     B-------C       ^     B-------C
 *    |    /       /        |    /       /        |    /       /
 *    |   /       /         |   /       /         |   /       /
 *    |  A-------D          |  A-------D          |  A-------D
 *    +-------------->X     +-------------->X     +-------------->Y
 *     XY_SKEW_FACTOR        XZ_SKEW_FACTOR        YZ_SKEW_FACTOR
 */
#define SKEW_CORRECTION

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 141.22
  #define XY_DIAG_BD 141.45
  #define XY_SIDE_AD 100.06

  // Or, set the default skew factors directly here
  // to override the above measurements:
  //#define XY_SKEW_FACTOR 0.0

  #define SKEW_CORRECTION_FOR_Z
  #if ENABLED(SKEW_CORRECTION_FOR_Z)
    #define XZ_DIAG_AC 140.64
    #define XZ_DIAG_BD 140.32
    #define YZ_DIAG_AC 140.41
    #define YZ_DIAG_BD 140.44
    #define YZ_SIDE_AD 99.84
    #define XZ_SKEW_FACTOR 0.0
    #define YZ_SKEW_FACTOR 0.0
  #endif

  // Enable this option for M852 to set skew at runtime
  #define SKEW_CORRECTION_GCODE
#endif

> 

from the way I set it up and the way it's written it only computes it for the xy factor. Or am I wrong and everything is setup right? 

Unfortunately I don't understand th epurpose of this line  

> 

 Enable this option for M852 to set skew at runtime
  #define SKEW_CORRECTION_GCODE

>

AnHardt, 2018-03-02:
Uncomenting `SKEW_CORRECTION` activates XY-skew-correction.
Additionally uncommenting `SKEW_CORRECTION_FOR_Z` activates XZ and YZ correction.
Without uncommenting `SKEW_CORRECTION_GCODE` both option will only work on the configured values. With activated `SKEW_CORRECTION_GCODE` you will get G852 to change the factors on the fly.

jonathanze, 2018-03-02:
So that means I should uncomment skew_correction, sew_correction_for_z and skew_correction_gcode. This will automatically calculate all 3 factors and will give me the option to adjust over the lcd.

jonathanze, 2018-03-02:
Like this 
#define SKEW_CORRECTION

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 141.22
  #define XY_DIAG_BD 141.45
  #define XY_SIDE_AD 100.06

  // Or, set the default skew factors directly here
  // to override the above measurements:
  //#define XY_SKEW_FACTOR 0.0

  #define SKEW_CORRECTION_FOR_Z
  #if ENABLED(SKEW_CORRECTION_FOR_Z)
    #define XZ_DIAG_AC 140.64
    #define XZ_DIAG_BD 140.32
    #define YZ_DIAG_AC 140.41
    #define YZ_DIAG_BD 140.44
    #define YZ_SIDE_AD 99.84
    #define XZ_SKEW_FACTOR 0.0
    #define YZ_SKEW_FACTOR 0.0
  #endif

   Enable this option for M852 to set skew at runtime
  #define SKEW_CORRECTION_GCODE
#endif

thinkyhead, 2018-03-05:
You got it!

closed 2018-03-05 by thinkyhead