MarlinFirmware/Marlin #9899: [bugfix-2.0.x] Weird extruder motor behaviour since introduction of linear advance 1.5

Zwaubel, 2018-03-02:
### Bug Report
- Description: 
I'm struggling very badly to make my extruder motor behave properly since the introduction of linear advance 1.5. 
It is consistantly underextruding while printing, and moving less material when extruding than when retracting. 
For instance: I heated up the nozzle and than made the extruder move 10 mm in extrude direction via axis move menu. Now when I try to make a retract movement of the same length via the menu it is moving roughly 1.33 times more material back.
Actual values:
  - desired: 10 mm in extrude direction -> actual: 6 mm
  - desired: 10 mm in retract direction -> actual: 8 mm

Attempting to compensate this behaviour by adjusting the steps per mm setting fails since a retract movement still results in more material being retracted than what a extrude movement with the same commanded length will extrude.

- Expected behaviour: move exactly as much material as commanded
- Actual behaviour: moving less material as commanded and behaving inconsistantly on extrude and retract moves
- Steps to reproduce:
  - see Description

- Steps I tried to fix the problem:
Swapped the DRV8825 on the extruder and re-adjusted the motor current. 
Checked the steps per mm settings, tried adjusting them.

Printer details:
Custom cartesian bowden extruder printer with Arduino Mega 2560 + RAMPS 1.4
TMC2208 drivers for the X-, Y- and Z-Axis and a DRV8825 driver for the extruder

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1776612/Config.zip)

Studiodyne, 2018-03-03:
If linear advance set to 0 , same problem  ?

Zwaubel, 2018-03-03:
setting k to 0 does not change the behaviour, however disabling it in the configuration file fixes the problem

thinkyhead, 2018-03-05:
See if it helps to set `MINIMUM_STEPPER_PULSE`. Start with 8, and if that works, then keep reducing it until you find a point where it fails. If it doesn't work, well at least that was tried…

Zwaubel, 2018-03-06:
@thinkyhead that worked like a charm, thank you so much! I'm now at a value of 1. Is it known problem with DRVs needing a longer pulse width?

closed 2018-03-07 by Zwaubel

thinkyhead, 2018-03-08:
> Is it known problem with DRVs needing a longer pulse width?

Actually, it is. Apparently the DRV8825 needs a minimum of 1.9µs between stepper pulse start and stop. Since Linear Advance is only moving the E axis, it's more likely to need the extra delay. A value of 1 should leave the XYZ axes unaffected while adding just enough delay for E.

@Bob-the-Kuhn, @Sebastianv650, @tcm0116 — Check it out! Even a value of 1 on a Mega2560 helped get a DRV8825 working with Linear Advance, definitively proving the delay principle.

Some other issues may also be helped by the recent change setting `MINIMUM_STEPPER_PULSE` to 2, such as cases where LCD-based manual moves are shorter, or where the Z axis by itself doesn't move as it should. In future when people report these kinds of issues we'll have to remember to check if they're using DRV8825 drivers and recommend adjusting the minimum pulse length.

PCWink, 2018-05-11:
I just installed Marlin 2.0 on my MKS sbase 1.3 (was using smoothiware) and experienced something similar to this. With linear advance on, the extruder clicks (as if underpowered or hobb bolt skipping/grinding). With no load on the extruder it continues do so without any difference when adjusting voltage for stepper motor. With linear advance disabled it prints fine. In the config.adv MINIMUM_STEPPER_PULSE, the default is at 2, I also had changed it to 1. Would I need to increase this? I really enjoy the linear advance on my other 8bit printers, would't mind finding a way to get this fixed.

thinkyhead, 2018-05-13:
We have some hope that our current work overhauling the planner will improve the situation for Linear Advance. There's also a 2.0 Linear Advance work-in-progress, and we have hopes that it will improve things even more.

PCWink, 2018-05-14:
That is good news, LA is one of the big reasons I swapped from smoothieware to Marlin.

SJ-Innovation, 2018-05-21:
Can also confirm this issue, unaffected by MIN_STEPPER_PULSE & even when K=0. Only completely disabling LIN_ADVANCE stops this. Using TMC2130 drivers, RAMPS1.4

thinkyhead, 2018-05-22:
@SJ-Innovation — Still a problem in the current bugfix branches?

Actually, belay that question and hold off on testing the latest code, because we just merged some major changes and we know there are Linear Advance issues currently. We hope to have them solved by the end of the week.

PCWink, 2018-07-01:
Where can I check for updates on the LA issue?

Zwaubel, 2018-07-01:
@PCWink
It should be fixed. I think @ejtagle recently addressed it in a PR and at least for me, everything is working fine.

ModMike, 2018-07-20:
I just enabled Linear advance 1.5 and have a setting of .4.  Same issue as others, extruder stops extruding at approximately layer 25.

buchnoun, 2018-09-05:
Hi,
I'm having an issue with LA 1.5 enabled in marlin for anycubic i3 Mega. When I set k>0 extruder stop extruding at some points on some design but it would start again after a while... I don't know what to do except setting k to 0 which fix my problem (but does not enable LA if I understood well). I'm using 2208 and a Titan direct extruder. Funny thing is I can print the test pattern to determine K (0.2 seems good for my printer) but on most model it stops extruding at some point... Could anyone help me ?

ModMike, 2018-09-05:
I was having this problem on an Ender 3 and could not even print the test pattern.  Check you extruder jerk settings.  Mine stopped nearly immediately because my jerk setting was at 5.  Try 10 or even 20.

Too low a jerk setting will cause the extruder to ignore direction changes and consequently do nothing.

Sebastianv650, 2018-09-05:
This may be related with #11711 especialy if it's repeatable with one gcode file. But please keep the other issue clean, I will use that for questions and feedback and a long list of posts isn't helping 👍