MarlinFirmware/Marlin #9900: Stop Print issue

Kyzuuh, 2018-03-02:
### Bug Report
- Description: When stopping the print through the LCD menu, the motors do not disable.  This is using a slightly modified Kossel example config.  Additionally, if you stop the print through the menu while the printer is in the homing process at the beginning of the print, the homing process will continue, but will slam into the endstops until the printer is forcefully shut off.

thinkyhead, 2018-03-05:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@Kyzuuh Please test with the latest `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead