MarlinFirmware/Marlin #9902: LCD corruption with Marlin-bugfix-1.1 Anycubic Kossel

rszasz, 2018-03-03:
I've had a few instances of LCD corruption while changing speed during a print, and just now had it corrupt about 22 hours into a print.

Going to the menu screen and back to the info screen cleans most of it up but leaves some corruption. 

Is there a known fix for this, or is there an option for forcing a full refresh of the screen every few minutes?

[configfiles.zip](https://github.com/MarlinFirmware/Marlin/files/1776888/configfiles.zip)

gloomyandy, 2018-03-03:
Might be worth providing details (picture if you can) of what exactly the corruption looks like.

rszasz, 2018-03-03:
Will do if I can get it to happen again, but it looks like random garbage was written to the LCD. And switching to the menu and back clears up all but one or two characters of it.

hapklaar, 2018-03-03:
Same issue here. I suspect static electricity might play a role as it often happens when I touch the printer.

thinkyhead, 2018-03-04:
With character LCDs it's almost always poor contact between plug and pins. Try bending the LCD connector pins in a little to make a tighter fit, both on the LCD and on the board.

boelle, 2019-02-20:
@rszasz problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek