MarlinFirmware/Marlin #9903: [Bugfix 1.1.x] TMC2130 sensorless homing doesn't work after movement

golobodo, 2018-03-03:
- Description: My setup have watterott TMC2130 drivers for X and Y and LV8729 for Z and E0, installed on a RAMPS1.4. I'm using a 600W stepup converter to apply 24V to the motor drivers exclusivly. I've also soldered one 220uF capacitor near to the stepper drivers. The system is powered by a Seasonic SSP-300TGS TFX PSU.
- Expected behaviour: movement to physical min position during sensorless homing after movement of X and/or Y, as some commits of bugfix1.1.x before.
- Actual behaviour: after movement of X and/or Y, movement to its physical min position during sensorless homing doesn't work during the same power on session. So it doesen't move during homing and takes the actual position as min position. I also tried M502 + M500, but same behavior.
- Steps to reproduce:
  - Using latest bugfix_1.1.x HEAD cd4c35c543014ebc85947293e8e8063d4375e0a1
  - TMC2130stepper library v2.1.5
  - G28 X0 Y0
  - G28 X0 Y0

Configuration_adv.h TMC section:
```cpp
#define HAVE_TMC2130
#if ENABLED(HAVE_TMC2130) || ENABLED(HAVE_TMC2208)
  #define X_IS_TMC2130
  #define Y_IS_TMC2130
  #define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
  #define HOLD_MULTIPLIER    0.7  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256
  #define X_CURRENT          600  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256
  #define Y_CURRENT          650
  #define Y_MICROSTEPS        16

  //#define STEALTHCHOP

  #define MONITOR_DRIVER_STATUS

  #if ENABLED(MONITOR_DRIVER_STATUS)
    #define CURRENT_STEP_DOWN     50  // [mA]
    #define REPORT_CURRENT_CHANGE
    #define STOP_ON_ERROR
  #endif

  //#define HYBRID_THRESHOLD

  #define SENSORLESS_HOMING   // TMC2130 only

  #if ENABLED(SENSORLESS_HOMING)
    #define X_HOMING_SENSITIVITY  8
    #define Y_HOMING_SENSITIVITY  8
    #define Z_HOMING_SENSITIVITY  8
  #endif

  #define TMC_DEBUG

  //#define TMC_Z_CALIBRATION

  #define  TMC_ADV() {  }
```

pins_ramps.h, Steppers section:
//
// Steppers
//
#define X_STEP_PIN         54
#define X_DIR_PIN          55
#define X_ENABLE_PIN       38
#ifndef X_CS_PIN
  #define X_CS_PIN         66
#endif

#define Y_STEP_PIN         60
#define Y_DIR_PIN          61
#define Y_ENABLE_PIN       56
#ifndef Y_CS_PIN
  #define Y_CS_PIN         44
#endif

#define Z_STEP_PIN         46
#define Z_DIR_PIN          48
[... 325 lines cut ...]
Recv: pwm
Recv: threshold		0	0
Recv: [mm/s]		-	-
Recv: OT prewarn	false	false
Recv: OT prewarn has
Recv: been triggered	false	false
Recv: off time		5	5
Recv: blank time	24	24
Recv: hysterisis
Recv: -end		2	2
Recv: -start		3	3
Recv: Stallguard thrs	19	19
Recv: DRVSTATUS	X	Y
Recv: stallguard
Recv: sg_result		657	578
Recv: fsactive
Recv: stst		X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:0E:02:99
Recv: 	Y = 0x80:0F:02:5E

teemuatlut, 2018-03-03:
Is the problem that homing doesn't work after a G1 movement
or is the problem that homing doesn't work through the LCD?

golobodo, 2018-03-03:
- Power On
- G1 X10 moves 10mm
- G28 X0 makes no movement. Only the hissing sound changes in frequency.

Output:
Send: G28 X0
Recv: X:-35.00 Y:85.00 Z:2.00 E:0.00 Count X:-3500 Y:8500 Z:3200

thinkyhead, 2018-03-04:
@golobodo — Does `G28` by itself work after `G1`?

golobodo, 2018-03-04:
- Power On
- G1 X10 Y10 <-- this works
- G28 <-- only Z is correctly homing

I will flash an older bugfix1.1.x version and I'll try to identify the source that causes this behavior.

golobodo, 2018-03-04:
Looks like, my problem is already discussed in this issue: https://github.com/MarlinFirmware/Marlin/issues/8890

BenBlv, 2018-03-04:
I got almost same problem in here with spreadCycle on

- on stealth mode G28 works as expected without any issues.
- on spreadCycle mode G28 doing it all wrong.

**with spreadCycle On after a power cycle:**
X-axis homing correctly
Y-axis going to the opposite direction
Z-axis homing correctly

**After the first G28, another G28 will result:**
X-axis moving to the opposite direction until it hits the end of axis/motor stuttering.
Y-axis sometimes moving back and sometimes forward (depends with the position).
Z-homing correctly.

Motherboard: Gen-L 1.0, tmc2130
v1.1.8 bug fix (today) 4/3 + updated 2130 libraries.
SPI configured and communicates / correct address.
Z safe homing enabled
sensorless homing enabled
BLtouch enabled
Debug tmc enabled

closed 2018-03-04 by golobodo