MarlinFirmware/Marlin #9905: Heated Bed Pid Autotune TImeout

markolo25, 2018-03-03:
### Bug Report
- Description: Heated Bed PID Autotune times out
- Expected behaviour: Print Autotune values
- Actual behaviour: Autotune halts and indicates timeout
- Steps to reproduce:
  - M303 E-1 C8 S60

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- Bug has been fixed in 1.1.x-Bugfix, I just merged the Bugfix with the 1.1.x branch and it works got my values. 

```

markolo25, 2018-03-03:
It's been resolved.

closed 2018-03-03 by markolo25

madmax2, 2018-10-29:
still occurring for me in 2.x bugfix

AnHardt, 2018-10-29:
@madmax2 
Does your bed heat at all?
What temperature do you want to tune for?
How long does it take to reach that temperature without PIDTEMPBED?
Is the wanted temperature reachable at all? Even with the part cooling fan blowing on the bed?
Is  PIDTEMPBED on at all?

Don't append to an issue closed since half a year.
Please open a new issue. 
Please follow the instructions in the 'new issue template'.
A 'me to' has about no information what could help us to help you.

AnHardt, 2018-10-29:
Follow up #12265

madmax2, 2018-10-29:
@AnHardt yes, I just search for this problem and found this thread
so I thought to add comment here..

But I have also created my own post after posting this comment so didn't see your followup comment/question till now..
Please continue in the new thread I created

Thanks