MarlinFirmware/Marlin #9906: [bugfix-1.1.x] Misuse of kill() in emergency_parser()

ABH10, 2018-03-03:
When the emergency_parser() catches an M112 command it calls the kill() function.
The kill() function is supposed to send status messages to the host (MSG_ERR_KILLED and ACTION_ON_KILL), but as the emergency_parser() and the kill() function are invoked from inside an ISR, these status messages never reaches the host. The kill() function was not designed to be called from inside an ISR.
Instead of actually calling kill() inside the emergency_parser(), I propose to flag the detection of M112 and then postpone the call of kill() to the idle() "task". This is how I have solved it locally, as my host relies on receiving the MSG_ERR_KILLED message.

Roxy-3D, 2018-03-03:
I wonder if it would be better (and make the code cleaner) if we made kill() callable at interrupt time?   Setting flags and hoping to see them later just makes the code paths more difficult to follow.   And there are very real issues why the emergency parser needs to be at interrupt time.   

What if always had a non-interrupt driven serial transmit routine available...   And kill() used that to send the MSG_ERR_KILLED message?    It would end up being a few extra bytes of code space used...   But the code would be cleaner.

thinkyhead, 2018-03-04:
Well, the main purpose of catching `M112` in the emergency parser is so we don't have to wait for it to come up in the queue before it gets executed. So all we need to do is to add a `bool killed_flag` and set that in response to `M112`. Then the `idle()` function can call `kill()` in response to that flag.

Roxy-3D, 2018-03-04:
> Well, the main purpose of catching M112 in the emergency parser is so we don't have to wait for it to come up in the queue before it gets executed.

If I remember right...  There was also an issue where the serial buffer space could be full.   So, catching it in the ISR made sure the firmware would see it before it got lost.

thinkyhead, 2018-03-04:
Ah yes, I remember that being a consideration, and it applies to the other emergency commands too.

PR #9922 (2018-03-04): [1.1.x] For M112, don't call kill from the serial interrupt
Fix #9906

If `kill()` is called from the emergency parser there is no serial output. This PR modifies the handling of `M112` so it sets a flag which is checked in `manage_heaters`, so that `kill()` will be called from the main loop.

Counterpart to #9923
commit b6fa55aa (2018-03-04): Fix: M112 calling kill from interrupt (#9922)

Fix #9906
PR #9923 (2018-03-04): [2.0.x] For M112, don't call kill from the serial interrupt
Fix #9906

If `kill()` is called from the emergency parser there is no serial output. This PR modifies the handling of `M112` so it sets a flag which is checked in `manage_heaters`, so that `kill()` will be called from the main loop.

Counterpart to #9922
commit 649f3754 (2018-03-04): Fix: M112 calling kill from interrupt (#9923)

Fix #9906
closed 2018-07-31 by thinkyhead