MarlinFirmware/Marlin #9907: [2.0.x] MKS SBase I2C stepper current regulator

yellobello, 2018-03-03:
Hi Marlineers,

A quick question whether the issue with the I2C current regulator only going full-on on the MKS sbase has been fixed yet?

I could not confirm this behaviour, but I had issues with my rather large y-stage having issues with lost steps on very complex files.
I then bumped up the current up till 2.0 with no better results, so I assume the digipot code still has issues?
Best regards,
phil

thinkyhead, 2018-03-09:
It has been patched, so give it another test. There's been one report that steppers still get hot, so let us know if that is also the case for you.

Peca006, 2018-03-11:
I'm sorry. I've been quite busy this weekend so I still didn't reported an issue. I make a test with clear clone of bugfix-2.0.x branch from e27fd26 commit. Steppers goes hot rather quickly and after 30 minutes they are so hot that I cann't hold finger on them even if I set current to 0.1 A.
There is my configs:
[peca-config.zip](https://github.com/MarlinFirmware/Marlin/files/1800923/peca-config.zip)

Peca006, 2018-03-22:
Ok, finally I found my own mistake! I somehow didn't spot and forgot uncomment following line:
```
// Uncomment to enable an I2C based DIGIPOT like on the Azteeg X3 Pro
#define DIGIPOT_I2C
```
Take my apologize please.

closed 2018-04-02 by thinkyhead