MarlinFirmware/Marlin #9908: [2.0.x] STM32F! Maylan 200 / MPSM question

yellobello, 2018-03-03:
Hi,

I saw that there is a port for the Maylan M200 printer and I am just curious since there appears to also be some driver for the LCD screen.
OK, the LCD screen of this printer has its own STM32 dedicated processor, much like the DUET-LCD or the MKS LCD, so I wonder if the supported screen on the M200 displays marlin´s internal LCD menu or adapts to the M200´s LCD menu.

best regards,

xC0000005, 2018-03-04:
The malyan LCD driver doesn't run on the LCD, which is an ESP8266. There is some code that runs on the actual display, but this isn't that. The LCD driver simply interprets between the LCD's curly brace language and relatively standard gcode (there's some special support for how file listing and print commands are handled.)

thinkyhead, 2018-03-04:
Is this different from the one we currently support?
```cpp
//
// LCD for Malyan M200 printers.
// This requires SDSUPPORT to be enabled
//
//#define MALYAN_LCD
```

xC0000005, 2018-03-04:
Ok, re-reading the original question:
Marlin's LCD support adapts to the Malyan LCD menu's language - it does not display Marlin's LCD menu on the Malyan display controller.

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle