MarlinFirmware/Marlin #9909: Anet A8 - No Z movement while printing and no working start gcode

Korpsian, 2018-03-03:
- Description: While printing the Z axis are not moving up, for example: new layer begins, z axis is still the same, moving the z axis manually with LCD display input works fine. Also defined start gcode is not taking place. The printer is just homing and start printing.
- Expected behaviour: Z axis should move up while printing
- Actual behaviour: Z movent stays at the same height, also visible on the LCD display
- Steps to reproduce:
  - Slice with Cura 3.2.0 and put on a SD card 
  - Start printing.

About the printer: Bowden upgrade and autolevel sensor installed (working without problems)
I used the Anet A8 preconfig file.

Start gcode:
G28 ;Home
G29 ; level bed
G92 E0
G1 F200 E3
G92 E0

[configfile.zip](https://github.com/MarlinFirmware/Marlin/files/1777469/configfile.zip)

thinkyhead, 2018-03-09:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

Korpsian, 2018-03-09:
thinkyhead thx for the response! 👍 
I fixed the problem with the homing. The only thing left is that the start G-Code is not running. 
I have an auto leveling sensor installed, and edited the config for that but the mesh bed leveling is not starting when printing. 

As you can see above in the description, I use the following Start G-Code with Cura 

G28 ;Home
G29 ; level bed
G92 E0
G1 F200 E3
G92 E0

- The printer homes the nozzoe, stops for a little bit and then starts printing, nothing more is happening. Any idee how i can fix this?

thinkyhead, 2018-03-11:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. (This will also enable bed leveling.)
- Copy the log output into a **.TXT file** and attach it to your next reply.

Korpsian, 2018-03-17:
Sorry for not answerering a long time.

I googled around and ran in the issue that I can not send Gcode via Serial at all. 

Connecting to the Printer with PuTTY or via the Arduino Serial Monitor get's the following output:
```
؜H⸮a⸮A⸮⸮⸮
⸮⸮Ƹ⸮Ѫ⸮a`ޘ⸮y⸮Q⸮%⸮⸮⸮⸮⸮⸮z@⸮⸮⸮E⸮⸮⸮⸮⸮⸮⸮Ա⸮8⸮⸮⸮Y5⸮.⸮⸮⸮AL⸮-ِ⸮`
```

Sending Gcode's lead's to no response from the Printer, nothing is happening. I tried to reupload it and no changes happend. 
Checking the code by Arduino shows no errors.

I'm using Marlin Version 1.1.8

Korpsian, 2018-03-17:
Sorry for answering so slow. 

Connecting to the printer via Arduino Serial Monitor is working, the Printer is sending back debug information as you can see in the .txt file. 

What is not working is that I can not send Gcode via Serial to the printer. I manually started homing and bed leveling via the LCD. 

Starting the bed leveling lead's to homing the printer and after that stopping and doing nothing more.

I think `?Probe (R)ight position out of range.` is the problem? Any idea where i need to fix this?

[anet a8 output serial.txt](https://github.com/MarlinFirmware/Marlin/files/1821869/anet.a8.output.serial.txt)

thinkyhead, 2018-03-18:
> What is not working is that I can not send Gcode via Serial to the printer.

Did you enable Newlines in the serial monitor?
[image]

Korpsian, 2018-04-08:
[anet a8 output serial v2.txt](https://github.com/MarlinFirmware/Marlin/files/1887607/anet.a8.output.serial.v2.txt)

after a long time i found the time again to take care of the printer, in the appendix is the debug output

`?Probe (R)ight position out of range.` seems the problem, how can I fix this?

thinkyhead, 2018-04-10:
>  how can I fix this?

Try commenting out these lines and re-flash.
```cpp
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 220
  #define FRONT_PROBE_BED_POSITION 15
  #define BACK_PROBE_BED_POSITION 170
```

Korpsian, 2018-04-15:
I'm not so familiar with arduino but they seem to be commented out right? If yes I allready did that.

```
#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define GRID_MAX_POINTS_X 3
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 220
  #define FRONT_PROBE_BED_POSITION 15
  #define BACK_PROBE_BED_POSITION 170

  // The Z probe minimum outer margin (to validate G29 parameters).
  #define MIN_PROBE_EDGE 10

  // Probe along the Y axis, advancing X after each column
  //#define PROBE_Y_FIRST

  #if ENABLED(AUTO_BED_LEVELING_BILINEAR)

    // Beyond the probed grid, continue the implied tilt?
    // Default is to maintain the height of the nearest edge.
    //#define EXTRAPOLATE_BEYOND_GRID

    //
    // Experimental Subdivision of the grid by Catmull-Rom method.
    // Synthesizes intermediate points to produce a more detailed mesh.
    //
    //#define ABL_BILINEAR_SUBDIVISION
    #if ENABLED(ABL_BILINEAR_SUBDIVISION)
      // Number of subdivisions between probe points
      #define BILINEAR_SUBDIVISIONS 3
    #endif

  #endif
```

thinkyhead, 2018-04-16:
Not commented out:
```cpp
  #define LEFT_PROBE_BED_POSITION 15
  #define RIGHT_PROBE_BED_POSITION 220
  #define FRONT_PROBE_BED_POSITION 15
  #define BACK_PROBE_BED_POSITION 170
```
Commented out:
```cpp
  //#define LEFT_PROBE_BED_POSITION 15
  //#define RIGHT_PROBE_BED_POSITION 220
  //#define FRONT_PROBE_BED_POSITION 15
  //#define BACK_PROBE_BED_POSITION 170
```

Korpsian, 2018-04-29:
Well then it is commented out, as you can see in my reply.
Any other ideas?

thinkyhead, 2018-04-30:
Actually, your reply shows them as NOT commented out, therefore enabled. Add slashes in front, and of course be sure to download the latest bugfix branch for your testing.

Korpsian, 2018-04-30:
woops ok I got it sorry, 

commenting out these lines leads to the following error when checking the code and Arduino jump's to the file "SanityCheck.h":

```
exit status 1
'LEFT_PROBE_BED_POSITION' was not declared in this scope
```

Code where the error occurs:
```
    #ifdef DELTA_PROBEABLE_RADIUS
      static_assert(LEFT_PROBE_BED_POSITION >= -DELTA_PROBEABLE_RADIUS, "LEFT_PROBE_BED_POSITION must be within DELTA_PROBEABLE_RADIUS.");
      static_assert(RIGHT_PROBE_BED_POSITION <= DELTA_PROBEABLE_RADIUS, "RIGHT_PROBE_BED_POSITION must be within DELTA_PROBEABLE_RADIUS.");
      static_assert(FRONT_PROBE_BED_POSITION >= -DELTA_PROBEABLE_RADIUS, "FRONT_PROBE_BED_POSITION must be within DELTA_PROBEABLE_RADIUS.");
      static_assert(BACK_PROBE_BED_POSITION <= DELTA_PROBEABLE_RADIUS, "BACK_PROBE_BED_POSITION must be within DELTA_PROBEABLE_RADIUS.");
    #else
      static_assert(LEFT_PROBE_BED_POSITION < RIGHT_PROBE_BED_POSITION, "LEFT_PROBE_BED_POSITION must be less than RIGHT_PROBE_BED_POSITION.");
      static_assert(FRONT_PROBE_BED_POSITION < BACK_PROBE_BED_POSITION, "FRONT_PROBE_BED_POSITION must be less than BACK_PROBE_BED_POSITION.");
      static_assert(LEFT_PROBE_BED_POSITION >= MIN_PROBE_X, "LEFT_PROBE_BED_POSITION can't be reached by the Z probe.");
      static_assert(RIGHT_PROBE_BED_POSITION <= 245, "RIGHT_PROBE_BED_POSITION can't be reached by the Z probe.");
      static_assert(FRONT_PROBE_BED_POSITION >= MIN_PROBE_Y, "FRONT_PROBE_BED_POSITION can't be reached by the Z probe.");
      static_assert(BACK_PROBE_BED_POSITION <= MAX_PROBE_Y, "BACK_PROBE_BED_POSITION can't be reached by the Z probe.");
    #endif

```

The line
`      static_assert(LEFT_PROBE_BED_POSITION >= MIN_PROBE_X, "LEFT_PROBE_BED_POSITION can't be reached by the Z probe.");`
leads to the error

thinkyhead, 2018-05-01:
Please test with the latest `bugfix-1.1.x` branch. If the problem has been resolved then we can close this issue.

By using the current bugfix branches you get this in your `Conditionals_post.h`:
```cpp
  #if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)
    #ifndef LEFT_PROBE_BED_POSITION
      #define LEFT_PROBE_BED_POSITION MIN_PROBE_X
    #endif
    #ifndef RIGHT_PROBE_BED_POSITION
      #define RIGHT_PROBE_BED_POSITION MAX_PROBE_X
    #endif
    #ifndef FRONT_PROBE_BED_POSITION
      #define FRONT_PROBE_BED_POSITION MIN_PROBE_Y
    #endif
    #ifndef BACK_PROBE_BED_POSITION
      #define BACK_PROBE_BED_POSITION MAX_PROBE_Y
    #endif
  #endif
```

Korpsian, 2018-05-01:
Ok it's fixed now.

I used the newest bugfix and also redid the config from scratch (because I had after that a different error, I think I messed up some stuff in the old one) and now it's working again! Thanks for the help and also making this amazing project!

closed 2018-05-04 by thinkyhead