MarlinFirmware/Marlin #9911: Stowing Z Probe with servo M402 not working

huwmcgregor, 2018-03-03:
### Bug Report
- Description: ---Stowing Z Probe with servo M402 not working
- Expected behaviour: —Should Stow
- Actual behaviour: —Nothing Happens
- Steps to reproduce:
Set up Auto Bed Levelling as per configuration file. Z probe deploys without issue, but won't stow away with M402 or b y going through Auto Bed Procedure.

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/1777522/Archive.zip)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

-----------------------------------------------------------

thinkyhead, 2018-03-04:
Whenever there are probe issues, we ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Do a `G28` to prepare the machine for testing.
- Issue the command `M111 S247` to enable maximum logging.
- Send `M401` to deploy the probe, then `M402` to stow it.
- Copy the log output into a **.TXT file** and attach it to your next reply.

I look forward to your followup report. Hopefully we can narrow this down to a single cause and fix it.

huwmcgregor, 2018-03-04:
Thanks so much for your prompt reply and assistance on this matter.

I followed your instructions,

please find text file attached

Many Thanks

Huw

thinkyhead, 2018-03-10:
We can't receive attachments by email. Drop your file directly into a reply at https://github.com/MarlinFirmware/Marlin/issues/9911

huwmcgregor, 2018-03-11:
Oh sorry didn't realise

[Log File.zip](https://github.com/MarlinFirmware/Marlin/files/1800808/Log.File.zip)

thinkyhead, 2018-03-12:
> Oh sorry didn't realise

We just learned this ourselves. It would be convenient!

thinkyhead, 2018-03-12:
So I've whittled down the log to this:
```
G28
X:126.00 Y:83.00 Z:25.00 E:0.00 Count A:16720 B:3440 Z:100000
M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
M401
  current_position=(126.00, 83.00, 25.00) : set_probe_deployed
deploy: 1
do_probe_raise(5.00)
>>> do_blocking_move_to(126.00, 83.00, 25.00)
<<< do_blocking_move_to
M402
  current_position=(126.00, 83.00, 25.00) : set_probe_deployed
deploy: 0
do_probe_raise(5.00)
>>> do_blocking_move_to(126.00, 83.00, 25.00)
<<< do_blocking_move_to
```
And clearly the raise is being weird! It calls `do_probe_raise` with a value of 5, but the extra 5mm never gets passed on to `do_blocking_move_to`. Certainly looks like a bug to me! Chances are it will be easy to fix. Give me 10-20 minutes…

thinkyhead, 2018-03-12:
So, your configuration shows you have both a SERVO probe and a SOLENOID probe enabled…
```cpp
#define Z_ENDSTOP_SERVO_NR 0   // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES {70,0}  // Z Servo Deploy and Stow angles
. . .
// A probe that is deployed and stowed with a solenoid pin (SOL1_PIN)
#define SOLENOID_PROBE SOL1_PIN_0
```
Your `SOLENOID_PROBE` setting is wrong. It should just be:
```cpp
// A probe that is deployed and stowed with a solenoid pin (SOL1_PIN)
#define SOLENOID_PROBE
```
…and additionally, it needs to define SOL1_PIN. For example:
```cpp
#define SOL1_PIN 44
```
Do you intend to use a SOLENOID probe or a SERVO probe?

huwmcgregor, 2018-03-12:
Hi, No I’m only intending to use a servo to actuate a swing arm z stop. 

Checking My Configuration.h the Solenoid Probe is not activated.

Will Attach my configuration.h file

huwmcgregor, 2018-03-12:
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1802518/Configuration.h.zip)

thinkyhead, 2018-03-12:
Marlin should be throwing a SanityCheck error preventing compilation because you have `Z_ENDSTOP_SERVO_NR` set to 1 and `NUM_SERVOS` set to 1. You need to set `NUM_SERVOS` to 2 if you want to use servo index 1 for `Z_ENDSTOP_SERVO_NR`.

huwmcgregor, 2018-03-15:
I didn't have any luck with those settings. If I set as the following:

#define Z_ENDSTOP_SERVO_NR 0 

and

#define NUM_SERVOS 1

Then it deploy's with M401, but does not stow again with M402

thinkyhead, 2018-03-15:
Ok, good to have confirmation that the servo works when `Z_ENDSTOP_SERVO_NR` is set to 0. That shows that it's the correct value. So now the question is, why doesn't the servo move on stow?

So, does `M280` work reliably? For example, `M280 P0 S90`, `M280 P0 S45`, etc.

huwmcgregor, 2018-03-15:
Yes it does work but not for S0
I can work it the other way up to S160

Could this be a wiring issue therefore?

Many Thanks

Huw

thinkyhead, 2018-03-15:
It sounds like something weird is going on. Do you have a spare servo to test with?

AnHardt, 2018-03-15:
Does S1 work?

huwmcgregor, 2018-03-16:
Unfortunately I don't have another solenoid to test. Yes it works with S1.

I'm able to use it like this for bed levelling now, but would be good to find out what the issue is.

thinkyhead, 2018-03-18:
So, in other words… setting the Stow Angle to 1 works?
```cpp
#define Z_SERVO_ANGLES { 70, 1 }  // Z Servo Deploy and Stow angles
```

boelle, 2019-02-20:
@huwmcgregor problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-06 by shitcreek