MarlinFirmware/Marlin #9912: Z probe leveling issues in Linear or Bilinear mode

boboche, 2018-03-03:
Hello,

If I am not posting at the right place, kindly direct me to the proper space.

I've been fighting with configuring a Z probe (inductive) to my Tevo Tarantula and using two builds, 17 feb 2018 1.1x and 2 March 2018 1.1.x.

Probe works, configurations and margins, probe type, and probe position reporting works and is accurate.  If I issue a G29, it goes at the right places and works as intended.  If I check the data, the leveling matrix outputs correct values showing that the plate is indeed at an angle (tevo with only 1 Z stepper).

The problem is that once I print, it doesn't seem to take the bed level into consideration, it only takes the homing  (dead center) and Z offset.  Since the stock Tevo is basically at an angle because there's only one motor and rod to support the weight, there's a natural angle in X (i.e. X0 the head touches the bed, X200 the head is a few mm above the bed).

I have read everything about clearing the eeprom, storing data, doing M420 S1 before starting a print... I must still be doing something wrong.  

Saw there was issues with Bilinear, went to UBL but wouldn't compile due to memory space I guess, so went Linear, still the exact same issue.   

I am to the point where manual leveling was awesome... which, just by saying that, shows that I need meds, a fix or some help.

Am I missing something or its a known issue.   If its known, can I rollback to a stable version that work with leveling, and which leveling should I take into consideration when compiling my firmware?

Thanks.

Roxy-3D, 2018-03-03:
Probe and save a Bi-Linear matrix.    Please post the results here...   There are some very simple checks we can do to see if things are operating correctly, and if not, determine what is setup wrong.

boboche, 2018-03-03:
Just noticed the Linear now seems to be reacting in Z so I'll dig further.

Bilinear result of my last probing, if theres a specific command to run please advise (this is from M420 V) 

READ: Bilinear Leveling Grid:
READ: Bilinear Leveling Grid:
READ:       0      1      2
READ:  0 +1.830 +0.055 -1.641
READ:  1 +1.798 +0.006 -1.618
READ:  2 +1.796 +0.035 -1.553
READ: 
READ: echo:Bed Leveling On 
Bed Leveling On 
READ: echo:Fade Height Off
Fade Height Off

Fade height?

boboche, 2018-03-03:
Nothing like posting in a forum after weeks of messing around to not have the problem anymore... geez... sorry..

After a n-th flash, problem is gone...always had the right matrix but Z wasn't moving when printer was... now everything acts normal when I flashed back to Bilinear 2-march build.

Thanks.!

closed 2018-03-03 by boboche

Roxy-3D, 2018-03-04:
Perhaps you didn't have the bed leveling system activated?    Personally...   I think you will have better results with a higher resolution mesh.    At least 5 x 5 but if what you have works....   No reason to complicate things.