MarlinFirmware/Marlin #9915: Unable to do manual probing on Anycubic Kossel on fw 1.1.8

hapklaar, 2018-03-03:
I wanted to try manual bed leveling on my kossel, but run into some issues. Not sure if this is a problem with the fw or simply user error.

I uncommented `#define PROBE_MANUALLY ` and then issued g29. Auto homing commenced, the the nozzle moved to it's first position but then moved back up, 'through' the end stops (!!)

I also made a reddit post on it with a little vid showing the issue
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/1777619/Configuration.h.txt)
. Is something wrong with the firmware, or did I miss something?

thinkyhead, 2018-03-09:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

rszasz, 2018-03-14:
I have manual probing working and only made minimal tweaks to the bugfix-1.1.x branch but my fork is a few days behind. (not sure if my tweaks made any difference to leveling as they were mostly pin assignment fixes to try and get bltouch probing working)

https://github.com/rszasz/Marlin/tree/bugfix-1.1.x-trigorilla-manual-probe

Delta parameters have been tweaked for my build, so be careful not to crash the extruder if you build my  test fork directly

closed 2018-04-10 by thinkyhead