MarlinFirmware/Marlin #9916: Z Probe to Z_MAX pin - Auto Home problem

jamesbit, 2018-03-03:
Hi !

I need to connect my proximity inductive sensor to Z_MAX pin.

I use in Configuration.h :

#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_ENDSTOP   Z_MAX_PIN

Now I have a Working probe and correct Bed Leveling but for Auto Home the Firmware use the Z_MIN switch and not the Probe.

Is there a way to use The Z Probe for Auto Homing too ?

Thank you...

thinkyhead, 2018-03-07:
This is wrong:
```cpp
#define Z_MIN_PROBE_ENDSTOP   Z_MAX_PIN
```
This would be correct if you want to have both a Z probe and a z min endstop switch.
```cpp
#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN  32
```
Of course, you would change "32" to the `Z_MAX_PIN` value that applies to your board.

HOWEVER, if you want to use the Z probe for homing, you need to connect it to the Z Min endstop plug, then use the following…
```cpp
#define USE_ZMIN_PLUG
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define Z_MIN_PROBE_ENDSTOP
```
(And don't define `Z_MIN_PROBE_PIN`.)

jamesbit, 2018-03-07:
Excuse me ... maybe I did a bit 'of confusion.

In my configuration the Z probe is connected to z max endstop plug.
The z min switch is connected to zmin plug.

I used :

#define Z_MIN_PROBE_ENDSTOP
#define Z_MIN_PROBE_PIN  32

My board GT2560 rev. A uses the pin 32 for z max.

I can correctly perform the self-leveling of the bed but for the homing Marlin always uses the mechanical switch z min and not the Z probe.

I want use the z min switch only in an emergency to disable the engines.

My question is : 

In this configuration Is there a way to use The Z Probe for Auto Homing too ?

thinkyhead, 2018-03-09:
> I want use the z min switch only in an emergency to disable the engines.

Connect it to the Z max endstop plug on the board. Then connect your probe to the Z min enstop plug on the board. Then you can use:
```cpp
#define USE_ZMIN_PLUG
#define USE_ZMAX_PLUG
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define Z_MIN_PROBE_ENDSTOP
#define Z_HOME_DIR -1
```
Marlin will then use the Z probe for both homing and leveling.

To get the Z max endstop to abort an SD print in progress try:
```cpp
#define ABORT_ON_ENDSTOP_HIT_FEATURE_ENABLED
```
… and then insert `M120` near the start of your G-code to enable endstops.

jamesbit, 2018-03-12:
Ok...

Thank you.

I will try it...

boelle, 2019-02-20:
@jamesbit did you try it?

closed 2019-07-06 by shitcreek