MarlinFirmware/Marlin #9918: [Bugfix_2.0.x] Custom Bootscreen Inop Radds LCD

bwgeez02, 2018-03-03:
Custom bootcreen doesn't show anything on the Radds LCD. Marlin bootscreen works. Not a big deal, but I've been trying to get it to work for hours over the last few days.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1777876/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1777877/Configuration_adv.zip)
[_Bootscreen.zip](https://github.com/MarlinFirmware/Marlin/files/1777879/_Bootscreen.zip)

AnHardt, 2018-03-03:
As far i can remember the 'official' RADDS-display is a text display.
So no chance to display a custom boot screen, only defined for graphical displays.

bwgeez02, 2018-03-03:
So it will display the Marlin bootscreen, but won't display the binary a bootscreen from Marlins bitmap converter? Correct me if I'm wrong but isn't the Marlin logo that it is displaying from dogm_bitmap.h which is binary?

AnHardt, 2018-03-03:
What you see on the RADDS-display is
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/ultralcd_impl_HD44780.h#L521-L526

bwgeez02, 2018-03-03:
Gotcha. Just changed Marlin to my name and uploaded it to be sure. Thank you. I've spent hours researching this. Would've been nice to have an explanation in Configuration.h that a graphical display is needed for custom bootscreen, for those that don't know better.

closed 2018-03-03 by bwgeez02

AnHardt, 2018-03-04:
Please onore Marlin and don't change the bootscreen. Please don't force us to write a custom boot screen for text displays. And if you do so don't tell us about it. It makes us sade.

bwgeez02, 2018-03-04:
I only did it for testing purposes.

thinkyhead, 2018-03-04:
I'll be sure to add a sanity-check that throws an error if `CUSTOM_BOOTSCREEN` is enabled when no graphical display is enabled.

AnHardt, 2018-03-04:
Currently 'PanelDue' is not supported by Marlin.

Color displays are not supported.
Graphical displays other than 128x64 are not supported.
Touch controllers are not supported.

So, a bit larger task.

thinkyhead, 2018-03-04:
> Is there any reason the PanelDue isn't supported in 2.0.x?

Because no one has yet contributed code to support it.

bwgeez02, 2018-03-04:
I would if I could. The best I can do is testing the code. This morning I emailed Duet3d to apply to register with them. I'll see if I can get somebody to help.