MarlinFirmware/Marlin #9919: UBL issue with large probe offset

skyflyer, 2018-03-03:
### Bug Report / Question

- Description: I've been using Marlin with Tevo Tarantula printer. The branch I'm using is `bugfix-1.1.x` with @JimBrown's configuration enhancements from https://github.com/JimBrown/MarlinTarantula/ (and I have my own changes in the config as well -- bed size, probe offsets, bed leveling). The configuration file is here: [config.zip](https://github.com/MarlinFirmware/Marlin/files/1777948/config.zip)

  At first, I had the probe mounted on the heatsink and it was offset from nozzle by 20mm. After a while, I've added LPA fan duct (and also upgraded the firmware version) and the probe is now a long way from the nozzle. It's offset is now 39mm right of the nozzle and 44mm in front of the nozzle (so that's 58.8mm away from the nozzle diagonally). The latest commit that was merged to this firmware from MarlinFirmware/Marlin is 63b13588a.

  The problem is that the bed leveling does not seem to work -- it is worse than no bed leveling at all (if I just level the bed manually on four corners and print). I'm afraid that probing and offset calculation has something to do with how far the probe is from the nozzle and that it is not taking into account that offset (wild guess).

  The printer uses SN04 sensor. Probe Z offset is set to -0.45 (`M851 Z-0.45`).

- Expected behaviour: once the bed is probed and leveled, the Z spacing from the bed would be (more) consistent across different points on the bed than without bed leveling (measured with the paper)
- Actual behaviour: once the bed is leveled, the Z spacing from the bed is not consisent across different points on the bed. It is actually better if the bed leveling is not activated.
- Steps to reproduce:
  - `G28`
  - `G29 P1`
  - `G29 P3 T` - after that, mesh is valid
  - `G29 T`
  - `G29 S1`
  - `G29 A`
  - `M500`

Perhaps I'm doing something wrong? I've read through (some) related posts (#9601, #9823, #9622, #9529) and saw that I should perhaps increase the size of the grid (probe more points)?

I tested with the mesh intentionally higher on the right side, so that I could see how the Z offsets would be applied, and there were large gaps. When I tested with manually leveled bed (as best as I could), the Z offsets were far worse with leveling enabled than without it.
The whole [leveling.txt](https://github.com/MarlinFirmware/Marlin/files/1777959/leveling.txt) procedure.

I tried to print the mesh, but the results were awful (didn't even stick to bed on most places).

```
< 12:52:34: <<< probe_pt
< 12:52:34:   current_position=(151.00, 54.00, 6.60) : set_probe_deployed
< 12:52:34: deploy: 0
< 12:52:34: >>> do_blocking_move_to(151.00, 54.00, 6.60)
< 12:52:34: <<< do_blocking_move_to
< 12:52:34: >>> do_blocking_move_to(61.00, 144.00, 6.60)
< 12:52:35: <<< do_blocking_move_to
  12:53:15: Communication timeout - reset send buffer block
< 12:53:15: Bed Topography Report:
< 12:53:15: (0,4)                           (4,4)
< 12:53:15: (10,190)                        (190,190)
< 12:53:15:      .          .          .          .          .     
< 12:53:15:  
< 12:53:15:      .     [-0.819]  -0.084    0.521    0.999  
< 12:53:15:  
< 12:53:15:      .      -0.657   -0.006    0.541    1.054  
< 12:53:15:  
< 12:53:15:      .      -0.555    0.043    0.550    1.023  
< 12:53:15:  
< 12:53:16:      .      -0.548    0.076    0.630    1.154  
< 12:53:16: (10,10)                            (190,10)
< 12:53:16: (0,0)                             (4,0)
< 12:53:16: ubl.mesh_is_valid = 0
```
From what I understand, the coordinates of the mesh look right to me: `(10,10)` is one corner and `(190,190)` is the corner diagonal to this one). And the location of the probe when probing seemed right. What I don't understand, though, is why are the results worse with UBL than bed leveling at all? And I would really like to use bed leveling, since it is quite time consuming to level the bed by hand each time.

I suspect that the problem might be related to probe offset from the nozzle (not being taken into account correctly). The reason for this suspicion is this: when I tilted the bed even more, the distance between the nozzle and bed (when Z was set to 0) was even bigger.

1) Should I try to use more points? 

2) Is there anything else I could do to debug this issue?

3) If I'm reading the [docs](http://marlinfw.org/docs/gcode/G029-ubl.html) correctly, I should just `G28` and `G29 J` before new prints (once the bed leveling has been performed and saved)? The leveling info should be stored by `M500` and thus loaded from the EEPROM by default? Or do I need to `G29 L1`? So far, I've made G29 before each print that I've tested bed leveling with.

Thank you.

Roxy-3D, 2018-03-04:
First...   You are doing SAFE_HOMING pretty much at the center of the bed, right?   If so, that is good.   We want to keep doing that.    

Next...   Let's bump up your mesh size.   Let's go at least 8x8 but 10x10 is the most used and tested.

Then...   Use G29 P1 U T to collect points  (maybe 10 or 12 points) and use that to physically level your bed.   It probably will take 3 or 4 iterations to get the bed pretty level.   And depending on how much your bed is warped...   It will be a judgment call on how to handle things.       

When you get that done...     Do a normal G29 P1,  save the mesh...    And post it here...    I'll try to be responsive and guide you through things and comment if I see any errors creep in...

You should not be doing a G29 J until we have the mesh working correctly.   And most likely, at that point, you won't have any reason to use the J option.

>  The leveling info should be stored by M500 and thus loaded from the EEPROM by default?

Save your mesh with G29 S 1 (that will put it in mesh slot 1)    
You can automatically load the mesh and have bed leveling turned on if you also do a G29 L1,  G29 A followed by a M500.   You can verify that is happening by resetting the printer and doing a G29 W.  It should say bed leveling is active and that the mesh from slot 1 has been loaded.

 > Or do I need to G29 L1? So far, I've made G29 before each print that I've tested bed leveling with.

You do not want to build a mesh prior to each print.   Even with the auto probing...   The mesh is not perfect.   You want to build the mesh...   And then fine tune it, and save it so you can keep using it until your machine's mechanics change.   I typically get 2 or 3 months before a mesh needs to be adjusted (due to bearing wear or changing sheets of glass).

Even if you find part of your mesh defective (after a few months),  very often it is just one part of it and it is quicker to just edit that one section of the mesh rather than go through the whole mesh building process.

gloomyandy, 2018-03-04:
How up to date is your copy of Marlin? There have been recent changes to the G29 J code, before these I think doing a G29 J would often make things worse. But as above unless you need a G29 J I'd avoid it, certainly until you have a working mesh.

skyflyer, 2018-03-04:
2018-03-04

First of all, I think you've set me on the right path. The results are much better than before. Thank you! @Roxy-3D, thank you for your support.

@gloomyandy, Marlin is 3 days old. The latest commit is 63b1358. 

I'll keep a journal of what's going on and the steps I'm taking with hopes that we can all provide better documentation for other users in the future. With that said, I'd like to stay on this commit, in order to keep the variables to a minimum.

SAFE_HOMING: yes, I'm doing safe homing (Z axis) on the center of the bed. My probe is offset from the nozzle. When z-homing, the probe is positioned on the bed center, as one would expect.

1. I've increased the grid size to 10.
1. Compiled and uploaded the reconfigured firmware.
1. Heated the bed to 60 C
1. `M502` to reset the stored eeprom
1. `M851 Z-0.45` to set the probe z offset from nozzle
1. `G28` homing
1. `G1 X100 Y100 Z1 F5000` and checked that the z-probe offset is OK (with the paper method)
1. Manually "leveled" the bed (adjusted the screws on the bed, so that nozzle has apx. same pressure on the papre underneath it) on four points `[(10,10), (10,190), (190,190), (190,10)]`. This results in acceptable prints with bed leveling disabled.
1. `M111 S32` - to enable bed leveling debug output
1. Checked the status of UBL:
    ```
    < 16:16:49: Unified Bed Leveling System v1.01 inactive.
    < 16:16:49: No Mesh Loaded.
    < 16:16:49: UBL object count: 1
    < 16:16:49: planner.z_fade_height : 0.0000
    < 16:16:49: # of samples: 100
    < 16:16:49: Mean Mesh Height: 0.000000
    < 16:16:49: Standard Deviation: 0.000000
    < 16:16:49: zprobe_zoffset: -0.4500000
    < 16:16:49: MESH_MIN_X  ((((((200) / 2) - (200) / 2) + 10)>(0 - 0)?((((200) / 2) - (200) / 2) + 10):(0 - 0)))=10
    < 16:16:49: MESH_MIN_Y  ((((((200) / 2) - (200) / 2) + 10)>(0 - 0)?((((200) / 2) - (200) / 2) + 10):(0 - 0)))=10
    < 16:16:49: MESH_MAX_X  ((((((200) / 2) + (200) / 2) - (10))<(200 + 0)?((((200) / 2) + (200) / 2) - (10)):(200 + 0)))=190
    < 16:16:49: MESH_MAX_Y  ((((((200) / 2) + (200) / 2) - (10))<(200 + 0)?((((200) / 2) + (200) / 2) - (10)):(200 + 0)))=190
    < 16:16:49: GRID_MAX_POINTS_X  10
    < 16:16:49: GRID_MAX_POINTS_Y  10
    < 16:16:49: MESH_X_DIST  20.00
    < 16:16:49: MESH_Y_DIST  20.00
    < 16:16:49: X-Axis Mesh Points at: 10.000  30.000  50.000  70.000  90.000  110.000  130.000  150.000  170.000  190.000  
    < 16:16:50: Y-Axis Mesh Points at: 10.000  30.000  50.000  70.000  90.000  110.000  130.000  150.000  170.000  190.000  
    < 16:16:50: Kill pin on :41  state:1
    < 16:16:50: Unified Bed Leveling sanity checks passed.
    ```
1. `G29 P1 U T` - if I understand correctly, this does the detection at the perimeter. The result of the procedure is stored in the log file [2018-03-04-after-g29-p1-u-t-1.log](https://github.com/MarlinFirmware/Marlin/files/1778713/2018-03-04-after-g29-p1-u-t-1.log).
1. Since the bed is not that tilted (or am I wrong?), I lowered the front part of the bed by 1/4 turn.
1. I issued `G29 P1 U T` again and the results are much better (I don't think I can get it any better manually). The result of the procedure is stored in the log file [2018-03-04-after-g29-p1-u-t-2.log](https://github.com/MarlinFirmware/Marlin/files/1778714/2018-03-04-after-g29-p1-u-t-2.log).
1. `G29 P1` 
1. `G29 P3 T` - after this the `M420 V` report said the mesh wasn't valid?
1. `G29 S1` - after this, the `M420 V` report said the mesh is valid
1. `G29 A`
1. `M500` - saved settings
1. At this point, the log is stored in the file [2018-03-04-after-g29-S1.log](https://github.com/MarlinFirmware/Marlin/files/1778716/2018-03-04-after-g29-S1.log)
1. I then moved the nozzle to four corners (offset by 10mm from the sides) to check the situation with the paper. It was OK (not perfect around (190,10)).
1. The I printed a test print, which was only 1 layer high (0.2mm): 
1. And the print was almost OK. I took a slicer screenshot and noted the areas where the filament didn't stick to bed (or skirt): [image]

1. I also notice that the measure height of the print is 0.25 - 0.30, so I will adjust the sensor z offset on the next print by 0.05mm.

How do you suggest we proceed?

PS: I'd like to clarify one more thing:

> You should not be doing a G29 J until we have the mesh working correctly. And most likely, at that point, you won't have any reason to use the J option.

The bed on the printer is positioned on four screws with springs. So, when removing prints from the bed, the bed definitely moves and the bed is not tilted in the same way afterwards (I suppose, haven't measured, but since the carriage is not bullet-proof and the bed rests on simple springs with screws, I assume so). So, if my assumption is correct, I should "correct" the plane each time I start a new print, shouldn't I?

Roxy-3D, 2018-03-04:
If the bed tilts between prints...  G29 J will help with that.   But lets get the mesh fine tuned first.  I'm responding using my phone so a more complete response will have to wait until this evening.   But you should consider doing a G26 as a 'test' print.  That will let us see what has to be change to perfectly describe the bed's topology.  (and G29 P4 to edit those points)

skyflyer, 2018-03-06:
I did several G26, but none of them was OK. I'm guessing it has something to do with the way it moves around. Is there an STL file of G26 pattern so I can slice it and print it that way?

Also note that the probe accuracy might be an issue: 
* with 5 repetitions: `Mean: -0.061625 Min: -0.063 Max: -0.061 Range: 0.002, Standard Deviation: 0.000637`
* with 10 repetitions: `Mean: -0.039688 Min: -0.047 Max: -0.032 Range: 0.015, Standard Deviation: 0.004526`

What I've done:

1. I lowered the nozzle from z offset -0.45 to -0.50 (even though I think -0.45 is more appropriate)
1. G28 & the leveling info is stored and activated (verified with `M420 V`)
1. I checked different positions on the bed with the paper method and the tension varies between points
1. I figured it can't hurt, so I issued `G29 J` which resulted in the following output:
    ```
    +1.000000 +0.000000 -0.000186 
    +0.000000 +1.000000 +0.000084 
    +0.000186 -0.000084 +1.000000 
    LSF Results A=0.0001862  B=-0.0000836  D=-0.0145363
    bed plane normal = [0.0001862,-0.0000836,1.0000000]
    ```
1. I couldn't feel much of a difference with the paper method.
1. Saved the settings, rebooted
1. I printed my test STL file again (layer=0.2 speed=slow)
    * On some parts of the bed, the extruder is skipping, so I'm assuming it is a tad too low.
    * The center of the print (it is supposed to be 1mm wide) is so thin that it is barely visible - I guess to nozzle is too close to the bed)
    * When doing the infill, I stopped the print as it was obviously too low and extruder was constantly skipping.
    * See the images: [2018-03-06-tevo1](https://user-images.githubusercontent.com/415535/37028128-b11685ba-2133-11e8-8a79-3101f674f0ca.jpg) 
 [2018-03-06-tevo2](https://user-images.githubusercontent.com/415535/37028148-bf550624-2133-11e8-83a0-55fa918af05a.jpg)
1. I figured I should increase z probe sampling and set it to 4 and recompiled the firmware, `#define MULTIPLE_PROBING 4`
    * The G28 homing probes just two times (why is that? I figured it should probe 4 times?)
1. Re-calibrated Z-probe offset again, so it was spot on on X100 Y100.
1. Issued `G29 P1`, `G29 P3 T` (the output in the log file) -- I know the bed is not super-level, but you see how much it changes from print to print anyhow (G29 output from a few days ago). [2018-03-06-g29.log](https://github.com/MarlinFirmware/Marlin/files/1784603/2018-03-06-g29.log)
1. Tried to print the same 15x15x0.2 STL file as before and the nozzle is too low on the left side of the bed and too high on the right side of the bed. Here's the picture, anyhow: [2018-03-06-tevo3](https://user-images.githubusercontent.com/415535/37028230-03248d34-2134-11e8-9d92-44f1a9867614.jpg)

    * better than before, but still not good

Since the nozzle is expanding when heated, should we measure the distance to bed (the paper method) when the nozzle is hot?

From what I can gather, I see there's manual leveling (corrections) to be done? What do you suggest?

Thank you! I appreciate your help and the time you're putting into this.

PS: you said that I should `Use G29 P1 U T` and use that to manually level the bed. I skipped that part today, since every time I clean the bed from previous (failed) prints, the bed position changes (hence my questions about `G29 J`). As you can see from the images, the bed is mounted on four screws with springs and everytime I take off a print (or clean a bed), this changes on the micro level. 

PPS: The G29 P1 takes about 25 minutes, so it is not really practical to execute it before every print (and I know, you said this shouldn't be necessary in the future).

Roxy-3D, 2018-03-06:
Those spider prints (or what ever they are) look like they are getting good first layer adhesion.   

How much does the bed's position shift when you release the print from the bed?   (That is one nice thing about printing on glass.   You just unclamp the glass, and after the print is release, you clamp it back down at the same location).

skyflyer, 2018-03-06:
@Roxy-3D, the spider legs are actually part of bed design. :P 

I'm printing with black PLA so the print is poorly visible on the photo (sorry about that -- I don't have a quality PLA in a different color). I drew (red) lines around it, to illustrate it, but the print design is actually the same STL file I referenced earlier [image](https://user-images.githubusercontent.com/415535/36948192-41cec7ba-1fd7-11e8-859e-db79f9ff5053.png) and 
[zipped stl](https://github.com/MarlinFirmware/Marlin/files/1785910/15x15x0_2.stl.zip).
The annotated image is here:
[image]

The nozzle is either too high or too low, when traveling from one point to another.

> How much does the bed's position shift when you release the print from the bed?

Can't say exactly, but as you can expect it moves a lot when we're pulling off the print. 

Please help me with the following questions:

1. Should I take the nozzle-to-bed measurement with the nozzle heated?
2. Should I try to hand-tune the nozzle-to-bed distance on various points on the mesh after `G29 J` or `G29 P1`? Can you explain how?
3. Can you share an STL file of the G26 test print, so I can slice it myself and set the correct speed (etc) parameters?

As I said, currently, the print results are worse with UBL enabled than without UBL (and just manual leveling). I attribute this to two factors (but I could be wrong): inaccurate sensor (and thus inaccurate mesh) and my lack of  understanding what is going on.

Roxy-3D, 2018-03-07:
You are going to need some different colors anyway...   Get some yellow....

G26 is the right way to tune the mesh.    It doesn't matter if it doesn't print perfect the first time.   We can iterate to a very well tuned mesh in a few times.    

> Please help me with the following questions:
> Should I take the nozzle-to-bed measurement with the nozzle heated?

Yes.   But if that isn't how you print...   Do the same stuff to calibrate the machine as you are going to do when you print.

> Should I try to hand-tune the nozzle-to-bed distance on various points on the mesh after G29 J or G29 P1? Can you explain how?

Let's not mess around with 'J' until we have some reasonable prints with a well tuned mesh.

> Can you share an STL file of the G26 test print, so I can slice it myself and set the correct speed (etc) parameters?

Let's get G26 working...   The whole reason for G26 is so no matter what mesh dimensions or bed size you have....   The validation pattern prints and is correct.   Every mesh point's location is very clearly visible with the pattern in place.    

> As I said, currently, the print results are worse with UBL enabled than without UBL (and just manual leveling). I attribute this to two factors (but I could be wrong): inaccurate sensor (and thus inaccurate mesh) and my lack of understanding what is going on.

The thing that is important is you get a well tuned mesh defined for your bed.   You can start with a NULL mesh (G29 P0) and just edit (with no auto probing) if you want.

skyflyer, 2018-03-07:
@Roxy-3D, first of all, thank you for your support, quick answers and guidance. 

> You are going to need some different colors anyway... Get some yellow....

Bought a yellow filament before work today :)

1. Started from scratch (let's try to fine tune the mesh and get at least something printed by G26)
1. Heated the nozzle to 210, bed to 60
1. G28 - to home
1. Check nozzle height - it was too low. **N.B.**: Since homing is performed at the center, I'm moving the nozzle to (100,100) (`G1 X100 Y100 Z1 F5000`) before measuring the nozzle-to-bed distance.
1. Adjusted with `M851 Z` until it was OK.
1. Used `G29 P1 U T` to level it. This is the result
    ```
    Bed Topography Report:
    (0,9)                                                                   (9,9)
    (10,190)                                                                (190,190)
         .          .          .          .          .          .          .          .          .          .     
     
         .          .          .          .          .          .          .          .          .          .     
     
         .          .       0.070  [ 0.015]   0.013   -0.019   -0.011   -0.067   -0.111   -0.139  
     
         .          .       0.030   -0.005   -0.016    0.014   -0.030   -0.064   -0.062   -0.090  
     
         .          .       0.024   -0.021    0.006   -0.011   -0.017   -0.055   -0.070   -0.047  
     
         .          .       0.071    0.024   -0.000   -0.041   -0.067   -0.097   -0.142   -0.163  
     
         .          .       0.089    0.046    0.006   -0.007   -0.040   -0.081   -0.114   -0.147  
     
         .          .       0.065   -0.006    0.021    0.018    0.005   -0.015   -0.041   -0.061  
     
         .          .       0.044   -0.007    0.021   -0.005    0.034   -0.003   -0.034    0.006  
     
         .          .       0.087    0.065    0.072    0.051    0.035   -0.031   -0.048   -0.065  
    (10,10)                                                                    (190,10)
    (0,0)                                                                     (9,0)
    ```
    I believe the bed is level enough - am I mistaken?
1. Then issued `G29 P1` (the result is visible in the log file) and `G29 P3 T` and `G29 S1` and `G29 A`. The log file: [2018-03-07-ubl.log](https://github.com/MarlinFirmware/Marlin/files/1790597/2018-03-07-ubl.log)
    ```
    Bed Topography Report:
    (0,9)                                                                   (9,9)
    (10,190)                                                                (190,190)
      0.418    0.304    0.191    0.077    0.069    0.010   -0.030   -0.082   -0.103   -0.104  
     
      0.311    0.223    0.136    0.048    0.048    0.008   -0.030   -0.082   -0.103   -0.104  
     
      0.204    0.143    0.081    0.019    0.027    0.006   -0.030   -0.082   -0.103   -0.104  
     
      0.098    0.062    0.026   -0.010    0.006    0.003   -0.009   -0.044   -0.053   -0.034  
     
      0.100    0.056    0.012   -0.033   -0.008  [-0.008]  -0.023   -0.065   -0.056   -0.043  
     
      0.195    0.129    0.064   -0.002   -0.011   -0.028   -0.048   -0.111   -0.134   -0.132  
     
      0.214    0.148    0.082    0.016    0.023   -0.012   -0.040   -0.090   -0.113   -0.114  
     
      0.151    0.104    0.057    0.010    0.038    0.021    0.007   -0.027   -0.034   -0.034  
     
      0.130    0.093    0.056    0.019    0.040    0.045    0.014   -0.021   -0.038   -0.014  
     
      0.225    0.173    0.121    0.070    0.075    0.063    0.015   -0.039   -0.041   -0.033  
    (10,10)                                                                    (190,10)
    (0,0)                                                                     (9,0)
    ubl.mesh_is_
    ```
1. Time to try and test-print: `G26 B60 H210`. The results are not terrible, but the nozzle is too high most of the time (IMHO): [2018-03-07-g26-first-1](https://user-images.githubusercontent.com/415535/37113568-fe811ab4-2245-11e8-938c-9df7bc8fb049.jpg), [2018-03-07-g26-first-2](https://user-images.githubusercontent.com/415535/37113576-05715398-2246-11e8-8952-c3f31812b56c.jpg)
1. Lowered the nozzle using `M851` and printed another sample with `G26 B60 H210`. Even better. Here are two images of the same print, from different angles: [2018-03-07-g26-second-1](https://user-images.githubusercontent.com/415535/37113660-4b97ba92-2246-11e8-9e7c-9d251ee681db.jpg), [2018-03-08-g26-second-2](https://user-images.githubusercontent.com/415535/37113725-7f9c23e6-2246-11e8-8373-8721b417f166.jpg)

As you can see, the filament is "oozing" from the nozzle between moves (not sure how to correct this, but I suppose retraction multiplier could solve this? 

IMHO the front part of the bed is the worst. And then there are some points which need fixing. If I understand this procedure correctly, I should just position the nozzle in the vicinity of the correct X/Y coordinate and issue `G29 P4 T` and that will let me manually adjust the height? Or do I need to specify the `R` argument?

What are your thoughts?

Roxy-3D, 2018-03-08:
> Used G29 P1 U T to level it. This is the result

You have mesh values very close to 0.000 mm where the homing probe is done.  That is good!   Your Z_PROBE_OFFSET_FROM_EXTRUDER is set very accurately.

And the numbers along the edge look very level.   That is going to work out very well.

> As you can see, the filament is "oozing" from the nozzle between moves (not sure how to correct this, but I suppose retraction multiplier could solve this?

It looks like your mesh numbers are 'acceptable'.   Good job.  I wonder why the top right corner didn't print?   The extra lines because of retraction and ooozing can be ignored.   It would be nice if we could get rid of them easily.    Probably that is going to require extra options to the G26 command.

> IMHO the front part of the bed is the worst. And then there are some points which need fixing. If I understand this procedure correctly, I should just position the nozzle in the vicinity of the [...]

Yes...  Just position the nozzle in the middle of any area you want to tweak.   And then do a G29 P4 R
If the nozzle moves over a mesh point that is OK....   Just click the encoder wheel and it will move to the next spot without editing anything.

Be sure to save your original mesh and your edited mesh in different slots so you can recover if you decide you don't like what the edits did.

skyflyer, 2018-03-08:
@Roxy-3D, I'm having troubles editing the mesh. When I positioned the nozzle to the area I wanted to edit, issued `G29 P4 R`, the nozzle stayed very high over the bed, even though the display said `0.224`. I then aborted and started mesh editing with the nozzle Z at `0.0`. The nozzle lifted to `5.something` and the display was again `0.224`. I also tried `G29 P4 R2` to try just a couple of points and all I can say is that the R parameter seems to be working (in the sense than it stops after a couple of probes), but the Z height is still way over the bed, so can't really edit the Z distances.

Any ideas why that would be? It is not very useful to edit the the mesh 5mm above the bed, since I can measure it with the paper...

skyflyer, 2018-03-08:
Oh, and two more questions:
1. Is it possible to somehow manually enter the mesh coordinates?
1. Does the M502 clear the stored mesh?

Roxy-3D, 2018-03-09:
The height of the nozzle during a G29 P4 does not reflect any meaningful information.   The nozzle is moved over the mesh point you are editing.   But you should be using the LCD Display to determine how much you are changing the point.    

If you have any adhesion at all...   You should not be changing the mesh point by more than +/- .25mm.   That is very small amount.   Your eyes can not measure that accurately.    

You can manually enter the mesh coordinates two ways.   You can do a G29 P4 R X100 Y90 and it will start editing as close to (X,Y) = (100,90) as possible.     Alternatively, you can do a M421
```

  /**
   * M421: Set a single Mesh Bed Leveling Z coordinate
   *
   * Usage:
   *   M421 I<xindex> J<yindex> Z<linear>
   *   M421 I<xindex> J<yindex> Q<offset>
   *   M421 C Z<linear>
   *   M421 C Q<offset>
   */
``

M502 should not affect your mesh data.

skyflyer, 2018-03-09:
Thanks! I assumed (wrongly) that the nozzle would be lowered to 0.0 and then I can correct it with the paper method... I'll edit the points, do a test print and report back.

skyflyer, 2018-03-09:
I'm having issues with point editing. I edited the points, and then issued `G29 J` to correct the tilt of the bed. (is it still to early to use `G29 J`?)

Here are the results:

```
    rotation matrix:
    +0.999998 +0.000000 -0.001956 
    -0.000000 +1.000000 -0.000212 
    +0.001956 +0.000212 +0.999998 
    LSF Results A=0.0019559  B=0.0002117  D=-0.1668709
    bed plane normal = [0.0019559,0.0002117,0.9999981]
```

The problem is, on my second `G29 J`, nozzle crashed into the bed, since the SN04 didn't trigger on the front-right part of the bed (this sensor is not super reliable). Tried a couple of times, and it would crash every time. I've already ordered BL touch, but it is still in the mail, so ... 

I have two questions now:
1. Should I avoid `G29 J` for now? (at the moment, I can't do it anyhow)
1. Should I print `G26` with adjustments untill I get it right? I suppose the flow now should be:
    a. edit the mesh
    b. print `G26` validation pattern
    c. repeat

Is this the way to go?

Thanks!

gloomyandy, 2018-03-09:
If I was you I'd avoid G29 J until you have a working mesh. If you are then having issues with using G29 J because the probe does not work very well at some points on the bed (mine does not work well close to the edge), then remember that you can change the location of the three points used for G29 J (the co-ordinates are in the configuration.h file). 

Also worth remembering that you can use G29 S<n> to save different versions of your mesh to different slots in the eprom (and then use G29 L<n> to reload them). This makes it easy to try different edits if you wish.

Roxy-3D, 2018-03-10:
> I'm having issues with point editing. I edited the points, and then issued G29 J to correct the tilt of the bed. (is it still to early to use G29 J?)

The honest truth is most people have no need for the 'J' mesh tilt.   Mesh bed leveling is far superior to the original tilted plane version of bed leveling.  (Hat tip to @epatel  !!!)    If you have your mesh well tuned, you are going to get a good first layer.

It takes a bit of a contrived example to come with a place where you benefit from it.   One example would be you have the mesh well defined for the piece of glass you print on.  But you are careless and some times get a little piece of plastic under the glass when you put it back on the bed.   In that case...   Tilting the mesh is beneficial.

Unified Bed Leveling is supposed make all the previous bed leveling system features available to be used together.   That is why it is in UBL.   But the truth is...   You probably don't need it.

ManuelMcLure, 2018-03-10:
One reason J mesh tilt might be useful is for i3-style machines with two Z-motors - on those when the motors are disabled it's easy to get the X axis skewed a bit when pressing on the carriage such as when changing filament. Unless you remember to use your handy-dandy realignment jig to make the X-axis even before you start printing you might need a bit of compensation left-to-right to get a good print.

Roxy-3D, 2018-03-10:
OK!  I stand corrected!  I guess I never saw this reason because I always leave the steppers configured to never disengage.

But "Yes!"   That is exactly the type of scenario where you might want to do a 'J' mesh tilt.

thinkyhead, 2018-03-11:
> The honest truth is most people have no need for the 'J' mesh tilt.

Also, too much tilt tends to mess with Z fade behavior. So it's best to avoid the use of Z fade if the deviation between the lowest and highest point is large.

skyflyer, 2018-03-11:
It is quite challenging to fine tune the mesh. I'm tuning with `M421` and when I change the Z height, this is not reflected on the axis movement.

I.e.:
1. I have a 10x10 grid
1. I changed the (9,9) point by issuing `M421 I9 J9 Q2.0`
1. Verified with `G29 T` and `M420 V` that it is reflected in the mesh. But when I move the nozzle to position (190, 190) (which is the coordinate 9, 9), the nozzle does not change height as I would expect.

On the other hand, when I tested this in the middle of the mesh, it was reflected.

The reason I tried such a drastic value is that I was trying to lower the top right corner, but the nozzle did not go lower even after several `M421 I9 J9 Q-0.10`, even when I issued `G28` between them.

1. Are the border points special?
1. Should I be editing them differently? What should I do between edits? Reset the board?

thinkyhead, 2018-03-11:
> Are the border points special?

Nope! Perhaps you've discovered a bug.

thinkyhead, 2018-03-11:
> Are the border points special?

They shouldn't be. Perhaps you've discovered a bug.

Try changing all the points along the right edge. Then try the top edge. See if they all exhibit the same issue.

skyflyer, 2018-03-11:
For instance, position is `X170 Y90`

```
< 16:10:34: (10,190)                                                                (190,190)
< 16:10:35:   0.218    0.254    0.141   -0.023   -0.081   -0.140   -0.180   -0.232   -0.303   -0.704  
< 16:10:35:  
< 16:10:35:   0.211    0.173    0.086   -0.002   -0.052   -0.092   -0.130   -0.182   -0.203   -0.704  
< 16:10:35:  
< 16:10:35:   0.154    0.093    0.031   -0.031   -0.073   -0.094   -0.130   -0.182   -0.303   -0.504  
< 16:10:35:  
< 16:10:35:   0.048    0.062    0.026   -0.010   -0.044   -0.096   -0.109   -0.144   -0.203   -0.534  
< 16:10:35:  
< 16:10:35:   0.050    0.056    0.012   -0.033   -0.058   -0.108   -0.123   -0.165   -0.206   -0.493  
< 16:10:35:  
< 16:10:35:   0.145    0.129    0.064   -0.002   -0.061   -0.128   -0.147   -0.211  [-0.284]  -0.582  
< 16:10:35:  
< 16:10:36:   0.164    0.148    0.082    0.016   -0.027   -0.112   -0.140   -0.190   -0.263   -0.564  
< 16:10:36:  
< 16:10:36:   0.101    0.104    0.057    0.010   -0.012   -0.079   -0.093   -0.177   -0.184   -0.484  
< 16:10:36:  
< 16:10:36:   0.080    0.093    0.106    0.019   -0.060   -0.055   -0.136   -0.171   -0.188   -0.264  
< 16:10:36:  
< 16:10:36:   0.025    0.123    0.071    0.070    0.025    0.013   -0.085   -0.089   -0.141   -0.333  
< 16:10:36: (10,10)                                                                    (190,10)
< 16:10:36: (0,0)                                                                     (9,0)
```

When I move X 20mm to the right, to position `X190 Y90z`, I assume the nozzle should be lowered, but it is actually lifted! The lift happens on the transition betweeen `X189` and `X190`.

It is so strange, that I'm sharing a [recording](https://drop.v-m-a.net/files/2018-03-11_16.17.08.mp4) with you. I'm focusing on the Z rod, so you can clearly see what happens when X axis reaches the `X190`. The moovement is from `X170` and the command is `G1 X190 F300`.

Position `X190 Y110` exhibits the same movement. Same goes for `X190 Y130`, etc...

I then loaded the first grid that was done with auto leveling (`G29 P1`):

```
< 16:33:21: Bed Topography Report:
< 16:33:21: (0,9)                                                                   (9,9)
< 16:33:21: (10,190)                                                                (190,190)
< 16:33:22:   0.418    0.304    0.191    0.077    0.069    0.010   -0.030   -0.082   -0.103   -0.104  
< 16:33:22:  
< 16:33:22:   0.311    0.223    0.136    0.048    0.048    0.008   -0.030   -0.082   -0.103   -0.104  
< 16:33:22:  
< 16:33:22:   0.204    0.143    0.081    0.019    0.027    0.006   -0.030   -0.082   -0.103   -0.104  
< 16:33:22:  
< 16:33:22:   0.098    0.062    0.026   -0.010    0.006    0.003   -0.009   -0.044   -0.053   -0.034  
< 16:33:22:  
< 16:33:22:   0.100    0.056    0.012   -0.033   -0.008   -0.008   -0.023   -0.065   -0.056  [-0.043] 
< 16:33:22:  
< 16:33:22:   0.195    0.129    0.064   -0.002   -0.011   -0.028   -0.048   -0.111   -0.134   -0.132  
< 16:33:22:  
< 16:33:23:   0.214    0.148    0.082    0.016    0.023   -0.012   -0.040   -0.090   -0.113   -0.114  
< 16:33:23:  
< 16:33:23:   0.151    0.104    0.057    0.010    0.038    0.021    0.007   -0.027   -0.034   -0.034  
< 16:33:23:  
< 16:33:23:   0.130    0.093    0.056    0.019    0.040    0.045    0.014   -0.021   -0.038   -0.014  
< 16:33:23:  
< 16:33:23:   0.225    0.173    0.121    0.070    0.075    0.063    0.015   -0.039   -0.041   -0.033  
< 16:33:23: (10,10)                                                                    (190,10)
< 16:33:23: (0,0)                                                                     (9,0)
```

I edited the the point (9, 5) with `M421 I9 J5 Q0.5`, issued `G28` (just to be sure) and then tested that position. Same effect. Here is the mesh:

```
< 16:34:02: Bed Topography Report:
< 16:34:02: (0,9)                                                                   (9,9)
< 16:34:02: (10,190)                                                                (190,190)
< 16:34:03:   0.418    0.304    0.191    0.077    0.069    0.010   -0.030   -0.082   -0.103   -0.104  
< 16:34:03:  
< 16:34:03:   0.311    0.223    0.136    0.048    0.048    0.008   -0.030   -0.082   -0.103   -0.104  
< 16:34:03:  
< 16:34:03:   0.204    0.143    0.081    0.019    0.027    0.006   -0.030   -0.082   -0.103   -0.104  
< 16:34:03:  
< 16:34:03:   0.098    0.062    0.026   -0.010    0.006    0.003   -0.009   -0.044   -0.053   -0.034  
< 16:34:03:  
< 16:34:03:   0.100    0.056    0.012   -0.033   -0.008   -0.008   -0.023   -0.065   -0.056  [ 0.457] 
< 16:34:03:  
< 16:34:03:   0.195    0.129    0.064   -0.002   -0.011   -0.028   -0.048   -0.111   -0.134   -0.132  
< 16:34:03:  
< 16:34:04:   0.214    0.148    0.082    0.016    0.023   -0.012   -0.040   -0.090   -0.113   -0.114  
< 16:34:04:  
< 16:34:04:   0.151    0.104    0.057    0.010    0.038    0.021    0.007   -0.027   -0.034   -0.034  
< 16:34:04:  
< 16:34:04:   0.130    0.093    0.056    0.019    0.040    0.045    0.014   -0.021   -0.038   -0.014  
< 16:34:04:  
< 16:34:04:   0.225    0.173    0.121    0.070    0.075    0.063    0.015   -0.039   -0.041   -0.033  
< 16:34:04: (10,10)                                                                    (190,10)
< 16:34:04: (0,0)                                                                     (9,0)
```

I suppose the transition to `0.457` should be gradual from `-0.056` and not instant between `X189` and `X190`?

skyflyer, 2018-03-11:
BTW, the the Marlin version I'm using is now 10 days old (latest commit is 63b13588a and I'm using @JimBrown's branch for Tevo compatibility).

thinkyhead, 2018-03-11:
Well, there's an interesting difference in UBL versus ABL when it comes to points outside of the mesh. With ABL (bilinear) the height of the **nearest edge** is used beyond the mesh bounds, so you can think of it as "flattening out" when you move outside. If the `EXTRAPOLATE_BEYOND_GRID` option is enabled, ABL will instead continue to raise or lower by continuing the same slope. In both cases, it's designed to avoid any sudden transition when you move past the mesh bounds.

In contrast to that, the current UBL implementation simply ceases outside of the mesh area, suddenly jumping to a fixed Z offset value. Normally this offset is 0, but that can be changed using the `UBL_Z_RAISE_WHEN_OFF_MESH` setting. It's recommended to make this value larger than your highest positive mesh Z value, otherwise you can damage the bed when you get outside the mesh bounds with the nozzle at a low altitude.

Anyway, do you see that sudden transition early at X>=171 or Y>=171? Or do you see other strange behavior in that region 171<=X<=190 or 171<=Y<=190? If so, then this would indicate that the compensation is cutting out early — stopping one grid line too soon.

skyflyer, 2018-03-11:
@thinkyhead,

I can't test with `UBL_Z_RAISE_WHEN_OFF_MESH` since I don't have access to the printer every day. I can confirm though, that the "weird" behaviour was happening within grid bounds (when transitioning from X189 to X190, where X190 is the last grid line).

I'm really eager to get this UBL working, but I'm having so many issues and I can still print better without any leveling (but I'd like to avoid tuning the bed before every print). As I've mentioned, Tevo Tarantula's bed mount is not very sturdy, so the tilt definitely changes when I take of the print (or scrub the bed when cleaning G26 prints, etc.... 

Also, I didn't mention before (and I didn't record it either, since I was in a hurry), but the "weird z movements" are happening in some instances when traveling along X190 while moving Y (i.e. between Y140 and Y150). It would happen that when the nozzle would come to Y150, it would lift and lower on the same spot (it seems). Very strange.

thinkyhead, 2018-03-11:
My eyes are bleeding from scrutinizing the UBL motion code. I couldn't discern anything that would obviously lead to jumping while following close to an edge. But since the mesh bound is ~189.99999, and at ~190.00001 there's a hard transition, it's a bad idea in-general to go anywhere near the mesh boundary.

I did fix one typo that would cause problems for a non-square mesh, but it's not one that would have any effect on your situation.

skyflyer, 2018-03-11:
I'll try with a bigger mesh (one that covers the whole bed (200x200). I suppose I should set the `MESH_INSET` to `0`? Since my probe will not be able to measure all points, I'm assuming this is accounted for (i.e. probe/nozzle won't crash into bed?)

thinkyhead, 2018-03-11:
> Since my probe will not be able to measure all points

Correct. The UBL auto-probing will only move the probe where it will be within the bed, and it won't try to probe where the carriage can't reach at all.

Roxy-3D, 2018-03-12:
@skyflyer   You might want to try a G29 Q 2 T  to put a diagonal test line across the bed in the mesh data.
I think you will see the nozzle do the right thing as you move across the bed.

skyflyer, 2018-03-26:
Sorry for the radio silence: received BLTouch. Will mount it this week; Set the mesh inset to 0, so it covers all of the bed, and will re-calibrate the whole, bed, print the G26 test print, etc... Stay tuned :)

skyflyer, 2018-04-10:
@Roxy-3D, @thinkyhead and others,

I'm grateful for your help in this matter. The 3D touch sensor I received turned out to be faulty, so I can't really test UBL. I'm closing this one. Once I receive a replacement, I'll definitely give it another go and report back.

I appreciate all the help and pointers you've all provided. And of course, the work you all do on this amazing piece of software.

closed 2018-04-10 by skyflyer