MarlinFirmware/Marlin #9920: UBL bed level and bed size

vMeph, 2018-03-04:
using the latest Marlin-bugfix-2.0.x

im having a hard time to figuer out how UBL bed level works
dont matter what settings i use im always getting same results
my machine has a work area of 200 x 200 x 200
im using max endstops for Y and X and min endstops for Z but everytime i hit to go home always tells me im at  X90 Y90 X10  , shouldnt that be X190 Y 190 Z10, im not using //#define BED_CENTER_AT_0_0

so when i try use level conors on LCD the machine only goes to the midle of bed X0 Y0  
what im missing?

ManuelMcLure, 2018-03-04:
Are you using Z_SAFE_HOMING? That will home X and Y and then move the probe to the center of the bed before homing Z.

vMeph, 2018-03-04:
yes im using  Z_SAFE_HOMING

vMeph, 2018-03-04:
when i use unified bed leveling on lcd under step by step ubl if i try build a mesh it will only proble 4 point and i have #define GRID_MAX_POINTS_X 3
its only go up to x90 y90

`READ: ok T:0.00 /0.00 B:17.54 /0.00 @:0 B@:0
SENT: G29 P1
READ: echo:G29 P1
G29 P1
READ: Mesh invalidated. Probing mesh.
READ:   current_position=(195.00, 195.00, 10.00) : set_probe_deployed
READ: deploy: 1
READ: do_probe_raise(10.00)
READ: >>> do_blocking_move_to(195.00, 195.00, 10.00)
READ: <<< do_blocking_move_to
READ: >>> probe_pt(195.00, 195.00, no stow)
READ:   current_position=(195.00, 195.00, 10.00) : 
READ: >>> do_blocking_move_to(185.00, 185.00, 10.00)
READ: <<< do_blocking_move_to
READ:   current_position=(185.00, 185.00, 10.00) : set_probe_deployed
READ: deploy: 1
READ:   current_position=(185.00, 185.00, 10.00) : >>> run_z_probe
READ:   current_position=(185.00, 185.00, 10.00) : >>> do_probe_move
READ: set_bltouch_deployed(1)
READ: >>> do_blocking_move_to(185.00, 185.00, -10.00)
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: set_bltouch_deployed(0)
READ:   current_position=(185.00, 185.00, -0.10) : sync_plan_position
READ:   current_position=(185.00, 185.00, -0.10) : <<< do_probe_move
READ: 1st Probe Z:-0.10
READ: >>> do_blocking_move_to(185.00, 185.00, 4.90)
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ:   current_position=(185.00, 185.00, 4.90) : >>> do_probe_move
READ: set_bltouch_deployed(1)
READ: >>> do_blocking_move_to(185.00, 185.00, -10.00)
READ: echo:busy: processing
busy: processing
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: set_bltouch_deployed(0)
READ:   current_position=(185.00, 185.00, -0.11) : sync_plan_position
READ:   current_position=(185.00, 185.00, -0.11) : <<< do_probe_move
READ: 2nd Probe Z:-0.11 Discrepancy:0.01
READ: >>> do_blocking_move_to(185.00, 185.00, 4.89)
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: <<< probe_pt
READ: >>> probe_pt(100.00, 195.00, no stow)
READ:   current_position=(185.00, 185.00, 4.89) : 
READ: >>> do_blocking_move_to(90.00, 185.00, 4.89)
READ: <<< do_blocking_move_to
READ:   current_position=(90.00, 185.00, 4.89) : set_probe_deployed
READ: deploy: 1
READ:   current_position=(90.00, 185.00, 4.89) : >>> run_z_probe
READ:   current_position=(90.00, 185.00, 4.89) : >>> do_probe_move
READ: set_bltouch_deployed(1)
READ: >>> do_blocking_move_to(90.00, 185.00, -10.00)
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: set_bltouch_deployed(0)
READ:   current_position=(90.00, 185.00, -0.49) : sync_plan_position
READ:   current_position=(90.00, 185.00, -0.49) : <<< do_probe_move
READ: 1st Probe Z:-0.49
READ: >>> do_blocking_move_to(90.00, 185.00, 4.51)
READ: echo:busy: processing
[... 94 lines cut ...]
READ: deploy: 0
READ: do_probe_raise(10.00)
READ: >>> do_blocking_move_to(90.00, 90.00, 10.00)
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to(90.00, 90.00, 10.00)
READ: <<< do_blocking_move_to
READ: >>> do_blocking_move_to(195.00, 195.00, 10.00)
READ: echo:busy: processing
busy: processing
READ: <<< do_blocking_move_to
READ: ok`

i have on settings
#define MESH_INSET 5              // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 3      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_PROBE_PT_1_X 10      // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 10
  #define UBL_PROBE_PT_2_X 100
  #define UBL_PROBE_PT_2_Y 100
  #define UBL_PROBE_PT_3_X 200
  #define UBL_PROBE_PT_3_Y 200

Roxy-3D, 2018-03-04:
Seriously...   If you are using a mesh based bed leveling system, a 3 x 3 grid does not make sense.   You can do it...   But...

Crank up the mesh size to 10 x 10.   Do a G29 P1 and save it with a G29 S1.

Please post the mesh that was automatically generated by doing a cut and paste of the G29 T

With that...   I'll be able to guide you on getting your system set up.   (It wouldn't hurt to cut and paste a G29 W here also.)

vMeph, 2018-03-04:
@Roxy-3D  thanks for trying helping out
i have changed to
  #define MESH_EDIT_GFX_OVERLAY   // Display a graphics overlay while editing the mesh

  #define MESH_INSET 1              // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_PROBE_PT_1_X 39       // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 180
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 20
  #define UBL_PROBE_PT_3_X 180
  #define UBL_PROBE_PT_3_Y 20

but every time i try upload the lcd is just blank, it will dont show nothing on it
been trying to figuer out where is the problem but so far nothing
here its my configs files:

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1778446/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1778447/Configuration_adv.txt)

vMeph, 2018-03-04:
the proble was:
#define MANUAL_X_HOME_POS 200
#define MANUAL_Y_HOME_POS 200
i comment and lcd is back on

G29 T

READ: Bed Topography Report:
READ: 
READ: (0,9)                                                                   (9,9)
READ: (1,199)                                                                (199,199)
READ:      .      -8.667   -8.574   -8.492   -8.390   -8.283   -8.198   -8.103   -8.054   -7.985  
READ:  
READ:      .      -8.786   -8.706   -8.651   -8.515   -8.424   -8.290   -8.196   -8.134   -8.192  
READ:  
READ:      .      -8.947   -8.858   -8.719   -8.590   -8.508   -8.399   -8.317   -8.218   -8.234  
READ:  
READ:      .      -8.995   -8.927   -8.817   -8.701   -8.579   -8.505   -8.407   -8.304   -8.214  
READ:  
READ:      .      -9.064   -8.973   -8.894   -8.797   -8.681   -8.563   -8.499   -8.404   -8.307  
READ:  
READ:      .      -9.149   -9.090   -8.994   -8.883   -8.775   -8.714   -8.612   -8.497   -8.427  
READ:  
READ:      .      -9.277   -9.192   -9.126   -9.040   -8.940   -8.845   -8.762   -8.748   -8.669  
READ:  
READ:      .      -9.444   -9.348   -9.318   -9.251   -9.104   -9.046   -8.970   -8.946   -8.859  
READ:  
READ:      .      -9.621   -9.555   -9.498   -9.465   -9.332   -9.264   -9.210   -9.155   -9.048  
READ:  
READ: [    .   ]      .          .          .          .          .          .          .          .          .     
READ: (1,1)                                                                    (199,1)
READ: (0,0)                                                                     (9,0)
READ: ok

SENT: G29 W
READ: Unified Bed Leveling System v1.01 inactive.
READ: Mesh 1 Loaded.
READ: UBL object count: 1
READ: planner.z_fade_height : 0.0000
READ: # of samples: 81
READ: Mean Mesh Height: -8.755989
READ: Standard Deviation: 0.391400
READ: zprobe_zoffset: 0.0000000
READ: MESH_MIN_X  1=1
READ: MESH_MIN_Y  1=1
READ: MESH_MAX_X  200 - (1)=199
READ: MESH_MAX_Y  200 - (1)=199
READ: GRID_MAX_POINTS_X  10
READ: GRID_MAX_POINTS_Y  10
READ: MESH_X_DIST  22.00
READ: MESH_Y_DIST  22.00
READ: X-Axis Mesh Points at: 1.000  23.000  45.000  67.000  89.000  111.000  133.000  155.000  177.000  199.000  
READ: Y-Axis Mesh Points at: 1.000  23.000  45.000  67.000  89.000  111.000  133.000  155.000  177.000  199.000  
READ: Kill pin on :41  state:1
READ: 
READ: Unified Bed Leveling sanity checks passed.
READ: ok

when is doing the autolevel the probes will do in a diagonal order, is there a way to do all X the move next line to Y and do all X and so on?

Roxy-3D, 2018-03-04:
Your Z_PROBE_OFFSET_FROM_EXTRUDER number is way off...    Where ever you are probing...  (and I recommend the center of the bed!!!!), those mesh points should be very close to +/- 0.000mm

Let's get that fixed first.     And if you get that fixed before I get back to a computer today...  you can get rid of the row and column of dot's by doing a G29 P3 to 'Smart Fill' them in.    

Then...  We can start checking things out.    (But if you want to jump ahead...   Get a G26 print done and post a picture of it.   That is what we will use to fine tune the mesh.)

vMeph, 2018-03-04:
i have tryed use bed center like you sugested #define BED_CENTER_AT_0_0
when i did the G29 P1 it only does half of bed, not sure why only does half size?

another thing that is happening is when i tryed applied the Z offset when i did the M502, M500,M501 to store values the lcd went blank, i had to clean eeprom using arduino sketch and the uploaded marlin again.
there must be something that isnt right....

also im still trying to understand this part
#define MESH_INSET 1              // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 10      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  #define UBL_PROBE_PT_1_X 10       // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 10
  #define UBL_PROBE_PT_2_X 50
  #define UBL_PROBE_PT_2_Y 50
  #define UBL_PROBE_PT_3_X 100
  #define UBL_PROBE_PT_3_Y 100

what you sugest for PT1 XY,PT_2XY,PT3XY value

vMeph, 2018-03-04:
I used this PT´but dont understand what they actualy do :( 
  #define UBL_PROBE_PT_1_X 39 // Probing points for 3-Point leveling of the mesh
  #define UBL_PROBE_PT_1_Y 90
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 20
  #define UBL_PROBE_PT_3_X 90
  #define UBL_PROBE_PT_3_Y 20

did the z off set and got this values

SENT: G29 T
READ: 
READ: Bed Topography Report:
READ: 
READ: (0,9)                                                                   (9,9)
READ: (1,199)                                                                (199,199)
READ:      .          .          .          .          .          .          .          .          .     [    .   ] 
READ:  
READ:   0.008    0.092    0.086    0.106    0.108    0.089    0.118    0.127    0.078       .     
READ:  
READ:   0.058    0.152    0.172    0.173    0.144    0.121    0.155    0.164    0.201       .     
READ:  
READ:   0.130    0.220    0.276    0.204    0.219    0.205    0.233    0.273    0.238       .     
READ:  
READ:   0.197    0.276    0.324    0.322    0.302    0.287    0.288    0.363    0.335       .     
READ:  
READ:   0.190    0.279    0.347    0.326    0.357    0.361    0.385    0.417    0.419       .     
READ:  
READ:   0.209    0.293    0.354    0.374    0.337    0.372    0.431    0.470    0.521       .     
READ:  
READ:   0.186    0.290    0.362    0.401    0.369    0.333    0.423    0.471    0.527       .     
READ:  
READ:   0.197    0.283    0.364    0.384    0.363    0.393    0.429    0.466    0.477       .     
READ:  
READ:   0.154    0.234    0.336    0.357    0.332    0.350    0.418    0.456    0.489       .     
READ: (1,1)                                                                    (199,1)
READ: (0,0)                                                                     (9,0)
READ: ok
SENT: M105
READ: ok T:0.00 /0.00 B:22.42 /0.00 @:0 B@:0
SENT: M105
READ: ok T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
SENT: M105
READ: ok T:0.00 /0.00 B:22.30 /0.00 @:0 B@:0
SENT: M105
READ: ok T:0.00 /0.00 B:22.23 /0.00 @:0 B@:0
SENT: M105
READ: ok T:0.00 /0.00 B:22.15 /0.00 @:0 B@:0
SENT: M105
READ: ok T:0.00 /0.00 B:22.30 /0.00 @:0 B@:0
SENT: G29 W
READ: Unified Bed Leveling System v1.01 inactive.
READ: Mesh 1 Loaded.
READ: UBL object count: 1
READ: planner.z_fade_height : 10.0000
READ: # of samples: 81
READ: Mean Mesh Height: 0.285543
READ: Standard Deviation: 0.121757
READ: zprobe_zoffset: -1.7999999
READ: MESH_MIN_X  1=1
READ: MESH_MIN_Y  1=1
READ: MESH_MAX_X  200 - (1)=199
READ: MESH_MAX_Y  200 - (1)=199
READ: GRID_MAX_POINTS_X  10
READ: GRID_MAX_POINTS_Y  10
READ: MESH_X_DIST  22.00
READ: MESH_Y_DIST  22.00
READ: X-Axis Mesh Points at: 1.000  23.000  45.000  67.000  89.000  111.000  133.000  155.000  177.000  199.000  
READ: Y-Axis Mesh Points at: 1.000  23.000  45.000  67.000  89.000  111.000  133.000  155.000  177.000  199.000  
READ: Kill pin on :41  state:1
READ: 
READ: Unified Bed Leveling sanity checks passed.

Roxy-3D, 2018-03-04:
> i have tryed use bed center like you sugested #define BED_CENTER_AT_0_0 when i did the G29 P1 it only does half of bed, not sure why only does half size?

No...    Try turning off   #define BED_CENTER_AT_0_0
and set these values:
```
#define Z_SAFE_HOMING
#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif
```

vMeph, 2018-03-04:
for some reason the lcd stay blank again and i need to clean eeprom using arduino sketch, it all seems to happen after i save the z offset with M500

Roxy-3D, 2018-03-04:
Doing a M502 followed by M500 should 'clean' the EEPROM to a normal state.

vMeph, 2018-03-05:
LCD always blank no ide why :(

Free Memory: 2860  PlannerBufferBytes: 1232
TEPROM version mismatch (EEPROM= Marlin=V52)
Hardcoded Default Settings Loaded
Unknown command: "TEPM105"
SD init fail

Roxy-3D, 2018-03-05:
What type of LCD do you have?   In your Configuration.h file you don't have either of the common LCD Panels declared:
```
/**
 * LCD TYPE
 *
 * Enable ULTRA_LCD for a 16x2, 16x4, 20x2, or 20x4 character-based LCD.
 * Enable DOGLCD for a 128x64 (ST7565R) Full Graphical Display.
 * (These options will be enabled automatically for most displays.)
 ...
 */
//#define ULTRA_LCD   // Character based
//#define DOGLCD      // Full graphics display
```

vMeph, 2018-03-05:
I got a #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

vMeph, 2018-03-05:
i have try with second arduino and result is same, every time i do M502 M500 M501 when i turn off and turn on, no image on lcd
my configs:
[Config.txt](https://github.com/MarlinFirmware/Marlin/files/1779163/Config.txt)

Roxy-3D, 2018-03-05:
My nice printer (gMax v1.5+) has a RepRap Discount Full Graphics Smart Controller...  And that is the primary development machine for UBL.     I'm out of town and can't check the current release.   It is possible some changes have been made to the LCD code that is causing trouble.    But in general...   That LCD Panel is the most tested one with UBL.

thinkyhead, 2018-03-05:
@vMeph Since you're just using a `BOARD_RAMPS_14_EFB` you may as well switch to the `bugfix-1.1.x` branch and work with that. If the LCD issue also exists there, that will tell us whether it's a typo in 2.0.x or something more general.

Roxy-3D, 2018-03-05:
Agreed!   That is a good idea!   @vMeph please switch to the bugfix-1.1.x branch.  Your mesh data should still be saved.   You may need to do a M502, G28, G29 L1, G29 T, G29 A, M502 to roll the EEPROM settings over.   (Not sure on that...   But doing that sequence will insure the migration goes smoothly.)

vMeph, 2018-03-05:
@thinkyhead @Roxy-3D  i can confirm that with bugfix-1.1.x branch  everything works fine, lcd dont goes blank, i used same configs

vMeph, 2018-03-05:
@Roxy-3D can you tell me when UBL is doing the autolevel is there any way for the probe be done first line of Y and do all the X´s, then move to next line of Y and do all X´s and so on?
has i see now is doing diagonal and random probe points

Roxy-3D, 2018-03-06:
Actually...   Bob did work to make it probe similarly to the grid based leveling.   He did that because he thought it was taking too long.   But when the work was done, doing a zig zag grid didn't save enough time to warrant the extra code.

> has i see now is doing diagonal and random probe points

The current probing (and moving) code tries to expand outward from the starting location.   This was done on purpose even though it causes some back and forth movements.   The reason is, if you are probing (or editing) the mesh, you are in a state where you don't fully trust the mesh.   By expanding outward from the starting location, if the bed has too much tilt, or too much warp, you should be able to see that the nozzle is getting too close to the bed well before there is a problem.  And of course, in that situation, the user has the required time to abort the operation before anything bad happens.

This does cause the nozzle to go back and forth across already probed (or edited) areas of the bed.  But those areas are 'safe' and won't cause damage to the printer as the nozzle moves across them in its quest to expand outward.

vMeph, 2018-03-06:
@Roxy-3D  thanks for the explanation!!!
Im adjusting the bed and will provide later on results of the mesh values...

How can i  center bed?, has it is now is a litle offset from where is probing

Roxy-3D, 2018-03-06:
> How can i center bed?, has it is now is a litle offset from where is probing

The simple answer is do a G29 P1 T and the final map for the last point probed will give you information about where each corner (and the center of the bed) are.   You should be able to use that information to iterate to a very physically leveled bed.

But...   That takes a lot of time to probe the bed just to have the acquired mesh invalidated when you shift the height of a given corner(s).    So...   You want to do a G29 P1 U T and you can stop it after 10 or 12 points.   U will spread the points out as far as possible for any previously acquired point.   Mostly, for physical bed leveling, just seeing the corners and points scattered around the bed is enough.   And that is what the U option will cause to happen.

Roxy-3D, 2018-03-06:
> How can i center bed?, has it is now is a litle offset from where is probing

Or do you mean "How do I get the Z Offset correct?"

Ideally, 1/2 of your mesh points should be positive and half negative.   And where you Z-Probe when you home should be 0.00 mm.     If that is what you are asking, you adjust your 
#define Z_PROBE_OFFSET_FROM_EXTRUDER number.    Rebuild, load the firmware, and do a M502 followed by M500.

vMeph, 2018-03-06:
I have set the Z_PROBE_OFFSET_FROM_EXTRUDER  i inicial was using M 851 Z—1.85 in my case, but then i have set in the firmware,

What i was meaning is  the center of probe is not in center i need to make it go to the left like 30 mm and Y also like 20mm to the front, cause when i do the auto level edge of points in one side is 10mm to the inside of bed witch is ok, but on the other side is like 30mm away from bed, so the mesh points are off set from the bed, so was wondering how to match bed with probe mesh

vMeph, 2018-03-06:
sense i have a bed with 214 by 214mm and the probe to be at center of bed needs to be 10mm to left and 7 mm to the front, what would be the best way to center de probe on the bed center?
using
define X_MAX_POS  214
#define Y_MAX_POS  214
#define Z_MAX_POS 200

and add also

#define MANUAL_X_HOME_POS 224
#define MANUAL_Y_HOME_POS 221

or use the M206 x? y?

or is there another proper way to do it?

ManuelMcLure, 2018-03-06:
Assuming you have `X_PROBE_OFFSET_FROM_EXTRUDER`, `Y_PROBE_OFFSET_FROM_EXTRUDER`, `X_MIN_POS`, `X_MAX_POS, X_BED_SIZE` and `Y_BED_SIZE` set correctly, and you leave `X_MAX_POS`, `Y_MAX_POS`, `MANUAL_X_HOME_POS` and `MANUAL_Y_HOME_POS` set to their defaults this should happen automagically - with `Z_SAFE_HOMING` set, of course.

Note that if your X and/or Y home positions are off the bed then `X_MIN_POS` and/or `Y_MIN_POS` might be negative - for example on my Anet A8 `X_MIN_POS` is -37 and `Y_MIN_POS` is -8.

vMeph, 2018-03-07:
@ManuelMcLure thanks i got all adjust now 

@Roxy-3D  i did a quick bed level
G29 P1 and i got this values, i probably can adjust a litle more

READ: Bed Topography Report:
READ: 
READ: (0,9)                                                                   (9,9)
READ: (10,204)                                                                (204,204)
READ:  -0.444   -0.401   -0.336   -0.324   -0.227   -0.197   -0.172   -0.179   -0.192   -0.223  
READ:  
READ:  -0.368   -0.334   -0.295   -0.244   -0.199   -0.155   -0.130   -0.135   -0.147   -0.194  
READ:  
READ:  -0.386   -0.333   -0.271   -0.201   -0.196   -0.154   -0.136   -0.118   -0.118   -0.153  
READ:  
READ:  -0.373   -0.280   -0.216   -0.195   -0.163   -0.100   -0.087   -0.098   -0.103   -0.057  
READ:  
READ:  -0.305   -0.228   -0.133   -0.112   -0.092   -0.030  [-0.018]  -0.021   -0.036   -0.061  
READ:  
READ:  -0.292   -0.153   -0.075   -0.043   -0.028    0.022    0.030    0.026    0.014    0.022  
READ:  
READ:  -0.209   -0.104   -0.051    0.051    0.048    0.081    0.100    0.081    0.092    0.039  
READ:  
READ:  -0.196   -0.101   -0.033    0.035    0.067    0.088    0.091    0.094    0.054    0.025  
READ:  
SENT: M105
READ:  -0.179   -0.101   -0.044    0.008    0.012    0.060    0.049    0.038    0.040    0.029  
READ:  
READ:  -0.203   -0.146   -0.120   -0.107   -0.031   -0.025   -0.019    0.006    0.031   -0.052  
READ: (10,10)                                                                    (204,10)
READ: (0,0)                                                                     (9,0)
READ: ok
READ: ok T:0.00 /0.00 B:16.29 /0.00 @:0 B@:0
SENT: M105
READ: ok T:0.00 /0.00 B:16.21 /0.00 @:0 B@:0
SENT: M105
READ: ok T:0.00 /0.00 B:16.37 /0.00 @:0 B@:0
SENT: M105
READ: ok T:0.00 /0.00 B:16.29 /0.00 @:0 B@:0
SENT: G29 W
READ: Unified Bed Leveling System v1.01 inactive.
SENT: M105
READ: Mesh 2 Loaded.
READ: UBL object count: 1
READ: planner.z_fade_height : 10.0000
READ: # of samples: 100
READ: Mean Mesh Height: -0.106470
READ: Standard Deviation: 0.131495
READ: zprobe_zoffset: -1.8500000
READ: MESH_MIN_X  10=10
READ: MESH_MIN_Y  10=10
READ: MESH_MAX_X  214 - (10)=204
READ: MESH_MAX_Y  214 - (10)=204
READ: GRID_MAX_POINTS_X  10
READ: GRID_MAX_POINTS_Y  10
READ: MESH_X_DIST  21.56
READ: MESH_Y_DIST  21.56
READ: X-Axis Mesh Points at: 10.000  31.556  53.111  74.667  96.222  117.778  139.333  160.889  182.444  204.000  
READ: Y-Axis Mesh Points at: 10.000  31.556  53.111  74.667  96.222  117.778  139.333  160.889  182.444  204.000  
READ: Kill pin on :41  state:1
READ: 
READ: Unified Bed Leveling sanity checks passed.

viper93458, 2018-03-08:
Hello, I don’t want to hijack, but just want to report the same exact lcd panel blanking issue. If I change from UBL to bi-linear, it works fine. I noticed that if I connect from my S3D host while the panel is blank, I will see it stuck in some sort of loop where it keeps loading the UBL mesh even though it’s all zeros and inactive. It doesn’t finish the normal booting process it seems.

-William

Roxy-3D, 2018-03-09:
@viper93458    Are you using bugfix-2.0.0 ?   Because VMeph side stepped the problem by switching to bugfix-1.1.x     You won't loose any features by doing that switch.

@thinkyhead    We must have some difference in the ultralcd.cpp code.

viper93458, 2018-03-09:
@Roxy-3D affirmative. I opened a dedicated bug report. :)

thinkyhead, 2018-03-09:
> We must have some difference in the ultralcd.cpp code.

I'll paste one into the other and see what Github Desktop reveals….

thinkyhead, 2018-03-09:
Hmm. No important differences in `ultralcd.*`. Must be elsewhere.

thinkyhead, 2018-03-09:
The bug is solved. It was caused by the watchdog was being enabled just before `settings.load`, which was being called before the temperature ISR (which resets the watchdog) was enabled. Because the UBL map output has a long delay (75ms) on each line of output, this was causing the watchdog to get triggered. We patched the `setup()` function so it enables the watchdog dead last.

closed 2018-03-09 by thinkyhead

Roxy-3D, 2018-03-09:
Aaaarrrrgggggghhhhh!    Thanks @thinkyead !        But why did it only affect the bugfix-2.0.0 branch?

thinkyhead, 2018-03-10:
> why did it only affect the bugfix-2.0.0 branch?

The watchdog initialization in 2.0.x had been moved to an earlier point in the `setup()` routine, probably to be up close to the section that checks and reports the "reason for reset" status, which includes "reset by watchdog." Later is definitely better for the watchdog, which is mainly there to deal with heaters potentially being on and un-monitored.

vMeph, 2018-03-20:
@Roxy-3D im having a hard time to make bl touch with UBL mesh to work the problem im experience is i do the folowing steps
G28 home
G1 Z0 raise bed to z position
M851 Z-1.6 in my case to adjust my Z close to bed
M500 to save folow by a M501
G29 P1 to create a mesh
G29 L1 to load
G29 A to activate
M500, M501
when i do a G26 the Z is always more high, its like is not assuming the Z offset value

this is my  setting output:
READ: Unified Bed Leveling System v1.01 active.
READ: 
READ: Unified Bed Leveling initialized.
READ: 
READ: Mesh loaded from slot 1
READ: Mesh 1 loaded from storage.
READ: echo:  G21    ; Units in mm
G21    ; Units in mm
READ: echo:  M149 C ; Units in Celsius
M149 C ; Units in Celsius
READ: 
READ: echo:Filament settings: Disabled
Filament settings: Disabled
READ: echo:  M200 D1.75
M200 D1.75
READ: echo:  M200 D0
M200 D0
READ: echo:Steps per unit:
Steps per unit:
READ: echo:  M92 X160.00 Y160.00 Z800.00 E224.10
M92 X160.00 Y160.00 Z800.00 E224.10
READ: echo:Maximum feedrates (units/s):
Maximum feedrates (units/s):
READ: echo:  M203 X200.00 Y200.00 Z5.00 E5.00
M203 X200.00 Y200.00 Z5.00 E5.00
READ: echo:Maximum Acceleration (units/s2):
Maximum Acceleration (units/s2):
READ: echo:  M201 X1000 Y500 Z10 E1000
M201 X1000 Y500 Z10 E1000
READ: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
READ: echo:  M204 P1000.00 R1000.00 T1000.00
M204 P1000.00 R1000.00 T1000.00
READ: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
READ: echo:  M205 S0.00 T0.00 B20000 X15.00 Y15.00 Z0.40 E5.00
M205 S0.00 T0.00 B20000 X15.00 Y15.00 Z0.40 E5.00
READ: echo:Home offset:
Home offset:
READ: echo:  M206 X0.00 Y0.00 Z0.00
M206 X0.00 Y0.00 Z0.00
READ: echo:Unified Bed Leveling:
Unified Bed Leveling:
READ: echo:  M420 S1 Z10.00
M420 S1 Z10.00
READ: 
READ: Unified Bed Leveling System v1.01 active.
READ: 
READ: Active Mesh Slot: 1
READ: EEPROM can hold 8 meshes.
READ: 
READ: echo:Material heatup parameters:
Material heatup parameters:
READ: echo:  M145 S0 H180 B70 F0
M145 S0 H180 B70 F0
READ: echo:  M145 S1 H240 B110 F0
M145 S1 H240 B110 F0
READ: echo:PID settings:
PID settings:
READ: echo:  M301 P22.72 I2.29 D56.30
M301 P22.72 I2.29 D56.30
READ: echo:  M304 P272.82 I53.72 D346.42
M304 P272.82 I53.72 D346.42
READ: echo:Z-Probe Offset (mm):
Z-Probe Offset (mm):
READ: echo:  M851 Z-1.20
M851 Z-1.20
READ: ok
SENT: M105

not sure what im doing wrong or what im missing

Roxy-3D, 2018-03-20:
```
G28 home
G1 Z0 raise bed to z position
M851 Z-1.6 in my case to adjust my Z close to bed
M500 to save folow by a M501
G29 P1 to create a mesh
G29 L1 to load
G29 A to activate
M500, M501
when i do a G26 the Z is always more high, its like is not assuming the Z offset value
```
This sequence doesn't really make sense to me.    Why do you create a mesh...  Just to discard it with a G29 L1 ???   It would make more sense to do a G29 S1 to save the mesh.

vMeph, 2018-03-20:
@Roxy-3D ohh sorry yes i do a G29 S1 before the G29 L1
just dont know why when doing a print or a G26 it always above the place where i have set the z offset

Roxy-3D, 2018-03-20:
It is good to prefix that entire sequence with M502 to pick up the default settings.   I don't use M851.   I pretty much just change the Configuration.h settings, do a M502, and then a M500.

It maybe your Z_PROBE_OFFSET_FROM_EXTRUDER is incorrect.   Doing a G29 W T will tell us a lot about what is happening.

vMeph, 2018-03-20:
Marlin 1.1.8
Last Updated: 2017-12-25 12:00 | Author: (test)
Compiled: Mar 20 2018
Free Memory: 2646  PlannerBufferBytes: 1232
V47 stored settings retrieved (614 bytes; crc 11244)
G21    ; Units in mm
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75
M200 D0
Steps per unit:
M92 X160.00 Y160.00 Z800.00 E224.10
Maximum feedrates (units/s):
M203 X200.00 Y200.00 Z5.00 E5.00
Maximum Acceleration (units/s2):
M201 X1000 Y500 Z10 E1000
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P1000.00 R1000.00 T1000.00
Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 S0.00 T0.00 B20000 X15.00 Y15.00 Z0.40 E5.00
Home offset:
M206 X0.00 Y0.00 Z-1.20
Unified Bed Leveling:
M420 S1 Z10.00
Material heatup parameters:
M145 S0 H180 B70 F0
M145 S1 H240 B110 F0
PID settings:
M301 P22.72 I2.29 D56.30
M304 P272.82 I53.72 D346.42
Z-Probe Offset (mm):
M851 Z-1.20
Active Extruder: 0
SD init fail
SENT: M105
READ: ok T:17.46 /0.00 B:16.91 /0.00 @:0 B@:0
SENT: M105
READ: ok T:17.46 /0.00 B:16.91 /0.00 @:0 B@:0
SENT: M105
READ: ok T:17.37 /0.00 B:16.99 /0.00 @:0 B@:0
SENT: M105
READ: ok T:17.50 /0.00 B:17.03 /0.00 @:0 B@:0
SENT: G29 W T
READ: echo:Home XYZ first
Home XYZ first
READ: Unified Bed Leveling System v1.01 active.
READ: Mesh 1 Loaded.
READ: UBL object count: 1
READ: planner.z_fade_height : 10.0000
READ: # of samples: 100
READ: Mean Mesh Height: -0.029592
READ: Standard Deviation: 0.263945
READ: zprobe_zoffset: -1.1999999
READ: MESH_MIN_X  10=10
READ: MESH_MIN_Y  10=10
READ: MESH_MAX_X  214 - (10)=204
READ: MESH_MAX_Y  214 - (10)=204
READ: GRID_MAX_POINTS_X  10
READ: GRID_MAX_POINTS_Y  10
READ: MESH_X_DIST  21.56
READ: MESH_Y_DIST  21.56
READ: X-Axis Mesh Points at: 10.000  31.556  53.111  74.667  96.222  117.778  139.333  160.889  182.444  204.000  
READ: Y-Axis Mesh Points at: 10.000  31.556  53.111  74.667  96.222  117.778  139.333  160.889  182.444  204.000  
READ: Kill pin on :41  state:1
READ: 
READ: Unified Bed Leveling sanity checks passed.
READ: 
READ: Bed Topography Report:
READ: 
READ: (0,9)                                                                   (9,9)
READ: (10,204)                                                                (204,204)
READ:   0.088    0.128    0.029    0.098    0.320    0.330    0.293    0.216    0.235    0.367  
READ:  
READ:   0.075   -0.014    0.095    0.208    0.321    0.413    0.407    0.368    0.279    0.216  
READ:  
SENT: M105
READ:   0.044   -0.084    0.056    0.204    0.421    0.161    0.223    0.414    0.291    0.170  
READ:  
READ:   0.104   -0.015    0.084    0.295    0.081    0.188    0.185    0.125    0.318    0.122  
READ:  
READ:   0.072   -0.014   -0.024    0.241    0.024    0.084    0.036    0.052    0.261    0.074  
READ:  
READ:  -0.067   -0.218   -0.084    0.123   -0.019   -0.023   -0.016    0.012    0.166   -0.069  
READ:  
READ:  -0.159   -0.310   -0.168   -0.046    0.117   -0.136    0.010    0.082   -0.071   -0.081  
READ:  
READ:  -0.308   -0.179echo:busy: processing
READ:    -0.226   -0.221   -0.124   -0.151   -0.157   -0.207   -0.194   -0.348  
READ:  
READ:  -0.420   -0.432   -0.216   -0.411   -0.288   -0.305   -0.343   -0.408   -0.429   -0.239  
READ:  
READ: [-0.602]  -0.564   -0.578   -0.430   -0.305   -0.622   -0.529   -0.491   -0.345   -0.585  
READ: (10,10)                                                                    (204,10)
READ: (0,0)                                                                     (9,0)
READ: ok

Roxy-3D, 2018-03-20:
Those G29 W and G29 T values look 'reasonable'.  What happens on the G26 ?   Can you take (and post) a video of it?

vMeph, 2018-03-20:
https://drive.google.com/file/d/1aUEy1syw9uM6B1Q__37IhiDF-hxFe-Fw/view?usp=drivesdk
srated with Z_PROBE_OFFSET_FROM_EXTRUDER -1.3, but still far from bed i adjust the Z_PROBE_OFFSET_FROM_EXTRUDER -2.3 didnt se no changes, on lcd i see machine nozle down z2, nozle up z2.5, in my machine the table moves up and down so the Z_PROBE_OFFSET_FROM_EXTRUDER should be -, correct?

Roxy-3D, 2018-03-20:
Do you still have that G26 pattern on the bed?    If so...   Let's edit those mesh points.  Let's lower all of those mesh points by .3 mm.    Let's save that mesh.    And then lets re-start the G26.    (and maybe restart the printer prior to doing that.)

I don't think that is the 'correct' way to handle this problem.   But it will give us a lot more information about what is wrong so we can fix it the right way.

vMeph, 2018-03-20:
gonna have to redo been messing with it but no results, so best to try your aproach,
gonna have to guide me on that, redoing mesh has we speak and will post results again

Roxy-3D, 2018-03-20:
I'll be here for the next 90 minutes...   Post the mesh when you get it.   And I presume you are homing the Z-Axis in the center of the bed?

vMeph, 2018-03-20:
Yes z axis on center...

vMeph, 2018-03-20:
SENT: G29 W T
READ: Unified Bed Leveling System v1.01 active.
READ: Mesh 1 Loaded.
READ: UBL object count: 1
READ: planner.z_fade_height : 10.0000
READ: # of samples: 100
READ: Mean Mesh Height: 0.026575
READ: Standard Deviation: 0.168715
READ: zprobe_zoffset: -1.2999999
READ: MESH_MIN_X  10=10
READ: MESH_MIN_Y  10=10
READ: MESH_MAX_X  214 - (10)=204
READ: MESH_MAX_Y  214 - (10)=204
READ: GRID_MAX_POINTS_X  10
READ: GRID_MAX_POINTS_Y  10
READ: MESH_X_DIST  21.56
READ: MESH_Y_DIST  21.56
READ: X-Axis Mesh Points at: 10.000  31.556  53.111  74.667  96.222  117.778  139.333  160.889  182.444  204.000  
READ: Y-Axis Mesh Points at: 10.000  31.556  53.111  74.667  96.222  117.778  139.333  160.889  182.444  204.000  
READ: Kill pin on :41  state:1
READ: 
READ: Unified Bed Leveling sanity checks passed.
READ: 
READ: Bed Topography Report:
READ: 
SENT: M105
READ: (0,9)                                                                   (9,9)
READ: (10,204)                                                                (204,204)
READ:  -0.192   -0.081    0.119    0.237   -0.015    0.074    0.134    0.339    0.340    0.108  
READ:  
READ:  -0.130    0.023    0.196   -0.036    0.005    0.175    0.204    0.087    0.106    0.352  
READ:  
READ:  -0.117    0.155   -0.112   -0.020    0.205    0.220    0.158    0.236    0.150    0.099  
READ:  
READ:  -0.029    0.191   -0.057    0.126    0.229    0.024    0.061    0.194    0.191    0.107  
READ:  
READ:  -0.027    0.205   -0.048    0.138    0.173   -0.010  [ 0.122]   0.117    0.214    0.034  
READ:  
READ:  -0.101    0.177   -0.096    0.060    0.292    0.082   -0.008    0.192    0.110    0.012  
READ:  
READ:  -0.067   -0.004   -0.135   -0.041    0.135    0.273    0.235echo:busy: processing
READ:     0.190    0.005   -0.019  
READ:  
READ:  -0.200   -0.139    0.121   -0.109   -0.069   -0.056   -0.069   -0.155   -0.032    0.147  
READ:  
READ:  -0.291   -0.225   -0.099    0.127    0.092   -0.033   -0.091   -0.024    0.093   -0.037  
READ:  
READ:  -0.596   -0.369   -0.278   -0.209   -0.121   -0.123   -0.107   -0.115   -0.130   -0.500  
READ: (10,10)                                                                    (204,10)
READ: (0,0)                                                                     (9,0)

Roxy-3D, 2018-03-20:
It looks like you already filled in the unprobed column and row?  Is that correct?
And you saved the mesh in slot 1 ?
Assuming that is true...   Let's see what G26 does.

vMeph, 2018-03-20:
doing the G26 the noozle is always at 2 to 3 mm above, thats the part i dont get why isnt respecting the z off set
seems the g26 is worst now, only thing i realy chaged was the #define XY_PROBE_SPEED  from 8000 to 1500 not sure if that plays a roll on it

Roxy-3D, 2018-03-20:
XY_PROBE_SPEED won't affect this issue...      Can you do:
- G28    (to home the machine)
- G1 Z1.00    (to get a know distance above the bed)
- M114   (to tell us where the machine thinks the nozzle is)

vMeph, 2018-03-20:
READ: X:129.00 Y:119.00 Z:1.00 E:0.00 Count X:20640 Y:19040 Z:842

Roxy-3D, 2018-03-20:
And...  when you look at the nozzle...   Is it really 1.00 mm above the bed?    If it appears to be, Lets do the same thing except set the Z height to 0.00 mm.    Does that just touch the bed?
- G28
- G1 Z0.00
- M114

vMeph, 2018-03-20:
READ: X:129.00 Y:119.00 Z:0.00 E:0.00 Count X:20640 Y:19040 Z:46
ya is toutching bed

Roxy-3D, 2018-03-20:
Hmmmm....    I think I better load up the current code base and duplicate what you are doing...
Something strange is going on....

vMeph, 2018-03-20:
im using stable version 1.1.8

Roxy-3D, 2018-03-20:
Aaaaarrrrggghhhhhh!    You should be using bugfix-1.1.x

vMeph, 2018-03-20:
ohhhh ok
 i changed for the stable, if you remember i was using the 2.x but then there was that problem with the bug of the lcd going off, and was sugested to use the stable :)

Roxy-3D, 2018-03-20:
Right.   I remember.   Incidentally, that LCD Problem in bugfix-2.0.0 is fixed now.   But lets use bugfix-1.1.x to get things working.

[10 more replies]