MarlinFirmware/Marlin #9921: G28 option with no Z lift for printers which idle at max Z

sprior, 2018-03-04:
### Description

I have a MakerGear M2 which has been modified with a BLTouch probe which is used for Z homing and bed leveling.  When the motors are idle the bed on the M2 falls to the bottom of its travel, so with the current behavior the G28 sequence tries to lower the bed 5-10mm more which grinds the stepper since Z is already at its max travel.

### Steps to Reproduce

MakerGear  M2 with BLTouch modification that has idle motors due to just being powered on or after an idle timeout.

1. G28

**Expected behavior:** [What you expect to happen]
The G28 sequence would home X and Y and then start lifting the bed until the BLTouch probe is triggered.

**Actual behavior:** [What actually happens]
The G28 sequence starts, home X and Y and then the Z stepper tries to lower the bed when it is already at max travel (so it grinds for a second) and then starts lifting the bed until the BLTouch probe triggers.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1778109/configuration.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
[https://www.youtube.com/watch?v=4huztdNq0OY](https://www.youtube.com/watch?v=4huztdNq0OY)

* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

In this case I believe the correct behavior should be a configuration that sets the assumption that if the motors are idle for any reason that the bed is at its max travel and no Z lift should be performed.  If the motors are not idle and the current Z position is known to be less than some set value then this would be the only situation when a Z lift should be performed as part of G28.

thinkyhead, 2018-03-04:
Hi @sprior — Thanks for your follow-up on this issue.

I'll come up with a patch in the next day or two to get your machine to do what you need, and if it seems like a good general solution, we may adopt it as an optional behavior.

zafff, 2018-03-05:
I would like to add that I have the same issue with Manual Z Probe levelling. #9896 
https://github.com/MarlinFirmware/Marlin/issues/9896

Studiodyne, 2018-03-05:
@thinkyhead 
Be careful .... because not possible in.all config .
- bed moves on z axis .... you can up the bed ... because bl touch is deployed.
- static bed .... impossible to down the carriage  because bl touch need a distance for deployment.  All bl touch users have 10mm minimum value. If not ... bltouch error and carriage crashes on the glass because bl touch not deployed and no trigger.

Make a config or a comment of the homing of bltouvh .  May be "if negative = for z moving bed ... if positive =for static bed "

thinkyhead, 2018-03-10:
I'll add a simple behavioral option called `NO_RAISE_FOR_UNKNOWN_Z`. When this is set, the bed will not be lowered (i.e., no Z raise for homing, clearance, or deployment) unless `axis_known_position[Z_AXIS] == true`. That should cover it.

thinkyhead, 2018-03-10:
@sprior — Please try out the new option and see if it works for your setup.

Studiodyne, 2018-03-10:
👍

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

sprior, 2019-06-23:
Sorry, I was never able to test this - the configuration I had been given to start with had too many changes I couldn't account for to apply those changes to a new build.

closed 2019-06-23 by sprior