MarlinFirmware/Marlin #9929: M605 S2 - DXC_DUPLICATION_MODE - Second extruder not extruding

emphasize, 2018-03-04:
barebone version

### **_The situation:_** 
Dual X Carriage custom Sigma-style printer. Movement - in every mode - ok (T1 start scenario in duplication mode aside). But when it comes to **duplicate Extruder(E0)-motor movements in M605 S2 mode it fails** to do that.

-------------------------------------------

### _Setup/Config:_ 
latest bugfix-1.1.x (8)
DUAL EXTRUDER / EXTRUDERS = 2
DUAL_X_CARRIAGE
LIN_ADVANCE
_Maybe good to mention:_ 
DISTINCT_E_FACTORS
No LCD defined (using "independent" one; MKS TFT)

--------------

Motors (Extr.) on...  "Set extruder_duplication_enabled / Clear active_extruder_parked" as the print starts... There seems to be a problem [where the signals _should be_ split off](https://github.com/MarlinFirmware/Marlin/blob/63b13588a8a73074c44f82f1eafda8a6b06dc99d/Marlin/stepper_indirection.h#L476). I dont pretend i know whats exactly going on regarding #_APPLY_STEP / #_WRITE_STEP / #_STEP_WRITE (stepper.h/.cpp) but over time there have been numerous recoding on this. Broken there?

**As I initiated the procedure with T1 active** (T1 -> M605 S2 -> G28 -> ...) **it was the other way around** (besides obviously no movements x-wise ) **T1 extruded, T0 not.**  Something's fishy down there.
On the other side and noobish as I am, I question myself why the duplication of the extruders is not coded like the X/X2 in stepper.cpp (if I understand the code correctly)?

emphasize, 2018-03-04:
Thats what I linked above the question ;)

thinkyhead, 2018-03-04:
Oh yeah! So, you can see, the logic is as basic as it gets.

thinkyhead, 2018-03-04:
So, it appears that you have to do `M605 S2` before you un-park the extruder. Otherwise `extruder_duplication_enabled` is never set to `true`.

emphasize, 2018-03-04:
Sequence:
> T0
G21
G90
M605 S2
G28
M140 S{temperature_heatbed}
M104 T0 S{temperature_extruder1}
M109 T0 S{temperature_extruder1}
M190 S{temperature_heatbed}
                                                                    <------- here the unparking(=**extruder_duplication_enabled=true**) of T1 occurs / moving to offset 
G92 E0
M82
M107
G1 F{travel_xy_speed}
M117 Printing..

thinkyhead, 2018-03-04:
Hmm, I'm not very familiar with Dual X Carriage, so thanks for pointing out how it behaves w/r/t parking and un-parking the extruder. I see from the code that `G28` parks the active extruder, and it gets un-parked when the next G-move command is received.

It also appears that `G28` resets the active extruder to 0.

emphasize, 2018-03-04:
> It also appears that G28 resets the active extruder to 0.

i can't agree on that. Just tested it - again

thinkyhead, 2018-03-04:
Right. I see now, afterward there's a call to `tool_change`.

thinkyhead, 2018-03-04:
Try adding logging wherever you find these (where relevant to dual-x):
```
dual_x_carriage_unpark
extruder_duplication_enabled = ...
active_extruder = ...
active_extruder_parked = ...
```
In other words, the regex:
```
(dual_x_carriage_unpark|extruder_duplication_enabled = |active_extruder(_parked)? = )
```
I'm not sure the existing logging provided by `DEBUG_LEVELING_FEATURE` + `M111 S247` will give enough detail.

The issue sounds like `extruder_duplication_enabled` is getting cleared someplace.

thinkyhead, 2018-03-04:
Just for kicks, also set `MINIMUM_STEPPER_PULSE` to 1 or 2 and re-flash.

emphasize, 2018-03-04:
> Just for kicks, also set MINIMUM_STEPPER_PULSE to 1 or 2 and re-flash.

ok, i'll try that first. The logging method may take a while

thinkyhead, 2018-03-04:
Also, you might as well ZIP up your `Configuration.h` and `Configuration_adv.h` files and drop them into your next reply. We'll check them over (in the morning - yawn!) and see if anything is amiss.

emphasize, 2018-03-04:
Some weird compiling error occured as I set MINIMUM_STEPPER_PULSE to 1

with exit status 1
'i' was not declared in this scope

>#if EXTRA_CYCLES_E > 20
        if (i) while (EXTRA_CYCLES_E > (uint32_t)(TCNT0 - pulse_start) * (INT0_PRESCALER)) { /* nada */ }
      #elif EXTRA_CYCLES_E > 0
        if (i) DELAY_NOPS(EXTRA_CYCLES_E);
      #endif

thinkyhead, 2018-03-04:
Re-download `bugfix-1.1.x` or just change `i` to `e_steps`.

emphasize, 2018-03-04:
Not kicked. Changing MINIMUM_STEPPER_PULSE to 1 doesn't do the trick

thinkyhead, 2018-03-04:
Well, it was good to eliminate the possibility of too short stepper pulses. So, next step is to get detailed logging and see what's really going on.

emphasize, 2018-03-05:
In the meantime i've flashed a fresh bufix version, so that we are on the same page. I did a couple of extended logging runs, but ultimately no big surprises here.

[Extended M111 Serial Communication (Pastebin)](https://pastebin.com/7139VB6v) [Just duplication flag logging (Pastebin)](https://pastebin.com/mJnsqKyv)
_first: logging relevant funtions/loops // second: logging in `process_next_command()`_

But (out of my perspective) with those two logs its save to say that the duplication flag is set properly for further processing

And the Config files you asked for [Conifgs.zip](https://github.com/MarlinFirmware/Marlin/files/1780654/Conifgs.zip)

Would some M503 help?

emphasize, 2018-03-05:
Altough i see nothing too special the [host serial communication at startup (pastebin)](https://pastebin.com/QKQNFjBn)

The only thing what tickles me is the combination of VOLUMETRIC:1, Filament settings: Disabled, M200 D0 and being somewhat triggered by #1378 / #2490

emphasize, 2018-03-06:
I know i'm grasping straws on this:

I'm with a fairly uncommon (i think) component setup. MKS Base (Drivers onboard), but X and Y is driven by stepsticks (TMC2130) hooked over a extension board (Stepper Expander) to EXP1 (like AUX-2 Ramps) and using the MISO/MOSI/SCK on EXP 2 (l. AUX-3). (fairly "common" wiring)
X2 using the standard X onboard driver socket. The Y onboard driver socket is therefor uncrowded. E0/E1 plugged in as usual.

Though i've tested this with the second Extruder wired to the Y-socket while tweaking `pins_RAMPS.h` in that regard with no positive outcome, it may somewhat meaningful to mention. 
(I've kept something in mind like "The next extruder is using the next open socket" -which would be the Y in my case - or something like that)

Maybe some folks who read this and are using Dual_X + Duplication mode and got it working can comment if they using this "out of the box" or what they've done to accomplish this. (wiring experiments, as i said, are no viable option to me)

emphasize, 2018-03-06:
After trying freaking everything i got it running. 

It appears to not working with LIN_ADVANCE defined. (if i recall it right it wasn't running either with prior LIN_ADVANCE _off_). So i set this issue to 

**M605 S2 - DXC_DUPLICATION_MODE - Second extruder not extruding in LIN_ADVANCE mode**

Sebastianv650, 2018-03-06:
Most likely because LIN_ADVANCE uses at least partialy own macros. This Macro-wrapped-in-marcro style for doing all the step things is like enigma for me, maybe @thinkyhead is faster, else I will see what I can do in the next days.

thinkyhead, 2018-03-07:
I'll take a look at `LIN_ADVANCE` now and make any needed changes so it's ok with duplication.

thinkyhead, 2018-03-07:
Ok, I've patched up `LIN_ADVANCE` so it handles duplication mode. Will merge the changes shortly.

closed 2018-03-07 by thinkyhead

emphasize, 2018-03-07:
@thinkyhead Had some compliling error after replacing `stepper.cpp` `stepper.h` with the new ones and enabling LIN_ADVANCE

In file included from C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.h:48:0,

from C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:48:

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp: In static member function 'static void Stepper::advance_isr()':

stepper_indirection.h:480: error: expected primary-expression before '{' token

     #define REV_E_DIR() { if (extruder_duplication_enabled) { E0_DIR_WRITE(INVERT_E0_DIR); E1_DIR_WRITE(INVERT_E1_DIR); } else if (current_block->active_extruder == 0) { E0_DIR_WRITE(INVERT_E0_DIR); } else { E1_DIR_WRITE(INVERT_E1_DIR); } }

                         ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:790:70: note: in expansion of macro 'REV_E_DIR'

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { e_steps < 0 ? REV_E_DIR() : NORM_E_DIR(); } }while(0)

                                                                      ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

       case 0: SET_E_STEP_DIR(0); break;

               ^

stepper_indirection.h:480: error: expected ':' before '{' token

     #define REV_E_DIR() { if (extruder_duplication_enabled) { E0_DIR_WRITE(INVERT_E0_DIR); E1_DIR_WRITE(INVERT_E1_DIR); } else if (current_block->active_extruder == 0) { E0_DIR_WRITE(INVERT_E0_DIR); } else { E1_DIR_WRITE(INVERT_E1_DIR); } }

                         ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:790:70: note: in expansion of macro 'REV_E_DIR'

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { e_steps < 0 ? REV_E_DIR() : NORM_E_DIR(); } }while(0)

                                                                      ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

       case 0: SET_E_STEP_DIR(0); break;

               ^

stepper_indirection.h:480: error: expected primary-expression before '{' token

     #define REV_E_DIR() { if (extruder_duplication_enabled) { E0_DIR_WRITE(INVERT_E0_DIR); E1_DIR_WRITE(INVERT_E1_DIR); } else if (current_block->active_extruder == 0) { E0_DIR_WRITE(INVERT_E0_DIR); } else { E1_DIR_WRITE(INVERT_E1_DIR); } }

                         ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:790:70: note: in expansion of macro 'REV_E_DIR'

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { e_steps < 0 ? REV_E_DIR() : NORM_E_DIR(); } }while(0)

                                                                      ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

       case 0: SET_E_STEP_DIR(0); break;

               ^

stepper.cpp:790: error: expected primary-expression before ':' token

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { e_steps < 0 ? REV_E_DIR() : NORM_E_DIR(); } }while(0)

                                                                                  ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

[... 53 lines cut ...]

                                                                      ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:826:17: note: in expansion of macro 'SET_E_STEP_DIR'

         case 1: SET_E_STEP_DIR(1); break;

                 ^

stepper.cpp:790: error: expected primary-expression before ':' token

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { e_steps < 0 ? REV_E_DIR() : NORM_E_DIR(); } }while(0)

                                                                                  ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_498070\sketch\stepper.cpp:826:17: note: in expansion of macro 'SET_E_STEP_DIR'

         case 1: SET_E_STEP_DIR(1); break;

                 ^

Bibliothek TMC2130Stepper in Version 2.1.5 im Ordner: C:\Users\Swen.WorkGameLaptop\Documents\Arduino\libraries\TMC2130Stepper  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\Swen.WorkGameLaptop\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI  wird verwendet
exit status 1
expected primary-expression before '{' token

thinkyhead, 2018-03-07:
Patched… hopefully! Please test the latest code.

emphasize, 2018-03-07:
hate to write it but compiling problems remain

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_769710\sketch\stepper.cpp: In static member function 'static void Stepper::advance_isr()':

stepper.cpp:790: error: expected '}' before 'else'

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)

                                                                                      ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_769710\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

       case 0: SET_E_STEP_DIR(0); break;

               ^

stepper.cpp:790: error: expected 'while' before '}' token

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)

                                                                                                           ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_769710\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

       case 0: SET_E_STEP_DIR(0); break;

               ^

stepper.cpp:790: error: expected '(' before '}' token

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)

                                                                                                           ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_769710\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

       case 0: SET_E_STEP_DIR(0); break;

               ^

stepper.cpp:790: error: expected primary-expression before '}' token

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)

                                                                                                           ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_769710\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

       case 0: SET_E_STEP_DIR(0); break;

               ^

stepper.cpp:790: error: expected ')' before '}' token

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)

                                                                                                           ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_769710\sketch\stepper.cpp:824:15: note: in expansion of macro 'SET_E_STEP_DIR'

       case 0: SET_E_STEP_DIR(0); break;

               ^

stepper.cpp:790: error: expected ';' before '}' token

       #define SET_E_STEP_DIR(INDEX) do{ if (e_steps) { if (e_steps < 0) REV_E_DIR(); else NORM_E_DIR(); } }while(0)

                                                                                                           ^

[... 95 lines cut ...]

                                                                                                            ^

C:\Users\SWEN~1.WOR\AppData\Local\Temp\arduino_build_769710\sketch\stepper.cpp:826:17: note: in expansion of macro 'SET_E_STEP_DIR'

         case 1: SET_E_STEP_DIR(1); break;

                 ^

stepper.cpp:826: error: expected unqualified-id before 'break'

         case 1: SET_E_STEP_DIR(1); break;

                                    ^

stepper.cpp:837: error: expected declaration before '}' token

     }

     ^

Bibliothek TMC2130Stepper in Version 2.1.5 im Ordner: C:\Users\Swen.WorkGameLaptop\Documents\Arduino\libraries\TMC2130Stepper  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\Swen.WorkGameLaptop\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\libraries\SPI  wird verwendet
exit status 1
expected '}' before 'else'

thinkyhead, 2018-03-08:
Ah, thanks for the follow-up. This is now patched in the bugfix branches. Some of the `stepper_indirection` macros — such as `REV_E_DIR` needed to be wrapped in `do{ ... }while(0)` so they can be used in `if` statements.

PR #9970 (2018-03-07): [1.1.x] Support duplication mode in LIN_ADVANCE
Fix #9929 — Add support for duplication mode to `LIN_ADVANCE`.

Counterpart to #9971

[Concise Diff](9970/files?w=1)
PR #9971 (2018-03-07): [2.0.x] Support duplication mode in LIN_ADVANCE
Fix #9929 — Add support for duplication mode to `LIN_ADVANCE`.

Counterpart to #9970
[Concise Diff](9971/files?w=1)