MarlinFirmware/Marlin #9931: [2.0.x] First things first: Sort out SPI header spaghetti.

thinkyhead, 2018-03-04:
@MarlinFirmware/32bit-maintainers !

At the moment the inter-dependency, most especially around the SPI-oriented headers in the HAL is distressing (`BitOrder` not defined, etc.). Let's figure out how to disentangle these –and any similar circular dependencies– so that we can move forward.

teemuatlut, 2018-03-04:
I think the clearest way is `Marlin -> Possible library -> External SPI library -> Platform core`.
Not everything can be included in Marlin since outside resources will not have access to the Marlin defined implementations.

BitOrder can be problematic since some define MSB/LSBFIRST as `#define MSBFIRST` and some use an enum of the two.

thinkyhead, 2018-03-04:
I mean, we can keep adding workarounds, like defining BitOrder in `HAL_SPI.h` for each platform. But ultimately, it just keeps digging us deeper.

boelle, 2019-10-24:
@thinkyhead still working on this one?

boelle, 2019-11-24:
this one has gone stale and discussion could as well be held on discord

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-24 by boelle