MarlinFirmware/Marlin #9933: Using Marlin firmware to make assorted Gcode controlled equipment

estechnical, 2018-03-04:
### Description

Hi there :)

I'm using Marlin as the basis for a few weird machines which require Gcode for control. Obviously Marlin provides a great base for this sort of project. Most of the time I have needed the steppers in some way, or it hasn't mattered in the cases where I have not, eg I've been using a RAMPS kit and there's been no need to think about it.

I have now come to a situation where I would like to run the Marlin firmware I have been adapting on an ATmega 328p and as such I'm looking to not compile any of the motion code with strategically placed #ifdef type statements.

I have questions and also wondered whether this ability may be of interest to others who adapt marlin to "strange machines"...

- Is it a desirable feature to be able to compile Marlin without specific axes (eg an XY table with no Z)?
- Is it desirable to have a new configuration file example for the bare minimum?
- Would you accept a PR that adds eg HASX, HASY, HASZ and HASMOTION flags?

Since Gcode is such a widely applicable control language and Marlin does such a good job of low-cost and stand-alone gcode control, it seems ideal to have these options available to machine builders.

Does anyone have any quick suggestions for how to implement this for maximum program memory saving? Where to start? etc.

Thanks,
Ed

thinkyhead, 2018-03-09:
> Would you accept a PR that adds eg HASX, HASY, HASZ and HASMOTION flags?

I think we would, but be warned, it's daunting to even make a version of Marlin with as simple a change as allowing `EXTRUDERS` to be set to 0. We've had several contributions attempting this, all of them too incomplete to use. These PRs tend to get stale very quickly, and when/if this change is introduced, it will put an extra burden on contributors to ensure their code can compile with `EXTRUDERS` set to 0.

Not that we don't already ensure a few burdens…

Currently it should be theoretically possible to compile Marlin with some or all of the X, Y, or Z step pins commented out, and this just skips any code that uses those pins. We have `HAS_X_STEP` (etc.) macros in `Conditionals_post.h` that can be used to conditionally skip certain axes.

I would say that if the options you propose were introduced as complete and working PRs for the `bugfix-2.0.x` then I wouldn't resist their inclusion at some point. But it is a very tricky proposition.

Where to begin is a very big question also. Arbitrary removal of X, Y, and/or Z axes goes to the very root of Marlin which establishes defines like `NUM_AXES`, macros like `LOOP_XYZ`, and so on to make it ostensibly easier to adapt to these kinds of changes. But these macros are not used in every possible instance, and there are many arrays and settings that expect to have at least X, Y, and Z axes.

I would say, try it and see how far you can get. We're migrating to the 2.0.x layout soon so you should only worry about that version. Push your commits to your Github fork and Travis CI will check to see that it compiles with a variety of options. The status will appear on your Commits page.

boelle, 2019-02-20:
@estechnical 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

estechnical, 2019-03-18:
@boelle I am looking to add features to Marlin myself - issues are the ideal place to keep track of this before I submit a PR. Your comment is unhelpful to say the least.

thinkyhead, 2019-05-16:
Closing since this is not actionable. Please submit a PR when you have something ready. (Marlin 2.0.)

closed 2019-05-16 by thinkyhead