MarlinFirmware/Marlin #9936: Z high is short by 10%-ish

Classictinker, 2018-03-04:
Before I forget my conf files are here [Marlin-Conf.zip](https://github.com/MarlinFirmware/Marlin/files/1778824/Marlin-Conf.zip)
So Hello again :)

I finally have the printer printing rather than blocking the extruder throat so I'm make some head way. My 1st few attempts were terrible due to over extrusion but lowering the extrusion multiplier to 0.9 seems to have sorted that issue out.

My problem now is that my test prints are about 10-15% short...well the 10mm cube I printed out certainly is.

I'm printing in 1.75mm PLA, bed temp 55, extruder 210,  0.35mm 1st layer all others are 0.25 with rectilinear infill.

When printing I can see the nozzle digging in to the previous layer as it does the infill. I measured the bed high then raised it 100mm using the touch screen, remeasured and it had raised exactly 100mm so I think my z calibration is correct in configuration.h although it's much smaller than the values I see in examples????

I don't want to start blindly adjusting things as I'll probably make things worse. Could somebody suggest a sensible approach to figuring out the problem? It doesn't help that I can't find any specs on my z Stepper motors other than the number 42SHDC4047Z-23B. Other than that all I know is that the Z lead screws are 8mm and to raise the bed 10mm takes approximately 1.25 turns of the lead screws. Are there any gcode commands I can use to determine the specs of the Z stepper motors?

I've attached a couple of pics of the 10mm cube below.

Many thanks

Rich
[image]
[image]

thinkyhead, 2018-03-04:
We are asking users to please in the future, ask for help in the [Marlin Forum](http://forums.reprap.org/list.php?415) and the [Marlin Facebook Group](https://www.facebook.com/groups/1049718498464482/) before interrupting the developers to ask these kinds of questions.

To figure out your discrepancy:
- Put a flag on your Z screw.
- Measure the height of the carriage with a caliper or ruler, and use `M114` to see the current Z.
- Move Z up with your host until it has done 10 full 360 rotations.
- Measure the height and subtract the starting height.
- Compare the measured movement to the movement reported by `M114`.

Correct your Z steps with `M92 Znnn`:
- CURRENT_Z_STEPS_PER_MM * (DESIRED_MM / ACTUAL_MM)

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead