MarlinFirmware/Marlin #9939: Y endstop does not respond till X endstop

ruggb, 2018-03-04:
I am in the middle of an upgrade. I ran into this new problem with the 3/1 commit 63b1358.
Since then I went back to my previous commit with its config and the problem is still there. That would say it is a hardware issue. But it may be something that got changed in the config - though when I went back to a0d452a with its config, it still did it. I can find no issue with the hardware or connections. But I am lost for an idea.
If I G28 with an initial position of X < Y everything look fine.
If I G28 with an initial position of X > Y it takes out my Y endstop.
As long as the X endstop is triggered first the Y endstop works.
It is a corexy and enabled as such.
the endstops work per m119
So what might be causing the Y endstop to be masked until the X triggers????????

ruggb, 2018-03-05:
I'm closing in on it. I went back to Marlin-bugfix-1.1.x-9cf5b96 and used that config. It works.
I then transferred only those settings to 63B1358 configs. That appears to work now EXCEPT for one small quirk.
If I trip Y (before it gets there) it stops then continues to X and stops then continues homing Y.
It does not do that if I trip X first. When I then trip Y everything stops. X does not continue to home.
Is this "normal"?
I don't remember if it did that before I moved the config to the new build.
It is like it goes back to look at Y after X and if it doesn't see it active it homes again.
But if  X is tripped first it doesn't go back to look at X after Y completes.
That isn't symmetrical, or consistent.
Now to see if I can find the setting that caused the original issue.

zafff, 2018-03-05:
I do not think that this is a bug per se. In the Configuration_adv.h in Marlin, there is an option for that (Line 359):
//#define HOME_Y_BEFORE_X
If you uncomment, it will home Y first and the X. When homing Y, it will ignore the X endstop totally. If you leave it commented, then Marlin will do its regular thing, that is home X first (while ignoring Y) and then home Y.

ruggb, 2018-03-05:
thanks @zafff but that isn't the problem. Since going way back in builds.......and redoing the new config file with the old info, I haven't reproduced it yet. It was homing both XY with quick_home. While it was doing that and b4 X was hit, it was totally ignoring Y and ripping straight thru the hardware.
I am changing one thing at a time so I can pinpoint it. HOME_Y_BEFORE_X was definitely never enabled.

ruggb, 2018-03-07:
The problem went away when I backed up. I am back on 63b1358 - I know it is a few days old.
When I home Y and hit the Y endstop, it stops.
When I home all and hit the Y endstop it does not stop.
After the X endstop is hit the Y endstop will function again.
It is repeatable.

The only dif between a config that works and one that does not is the definition of the bed size.
#define Y_BED_SIZE 305 //** max size
#define ABL_PROBE_PT_1_Y 300 //** use right rear for 1st

Previously it was 295 and 290. The files were compared with by the computer with my editor program. No eyes involved except to read the results.
I also have added a MY_REV variable to keep track of the files and it changed from 1J to 1K.

I checked it 3 times. All I changed between loads it the configuration.h file that has the above dif.

I am also getting a volume.init failed error the first time I test it after a load. I don't know what that is.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1790816/Configuration.zip)

ruggb, 2018-03-07:
The problem went away when I backed up. I am back on 63b1358 - I know it is a few days old.
When I home Y and hit the Y endstop, it stops.
When I home all and hit the Y endstop it does not stop.
After the X endstop is hit the Y endstop will function again.
It is repeatable.

The only dif between a config that works and one that does not is the definition of the bed size.
#define Y_BED_SIZE 305 //** max size
#define ABL_PROBE_PT_1_Y 300 //** use right rear for 1st

Previously it was 295 and 290. The files were compared with by the computer with my editor program. No eyes involved except to read the results.
I also have added a MY_REV variable to keep track of the files and it changed from 1J to 1K.

I checked it 3 times. All I changed between loads it the configuration.h file that has the above dif.

I am also getting a volume.init failed error the first time I test it after a load. I don't know what that is.

Configuration.zip

ruggb, 2018-03-07:
IF X_BED_SIZE = Y_BED_SIZE is crashes.
If they are not = either way, it is OK.

AnHardt, 2018-03-07:
@ruggb 
Are you using a coreXV machine?
IF X_BED_SIZE = Y_BED_SIZE only one stepper is moving. I'd guess X.

ruggb, 2018-03-08:
As long as the X_BED_SIZE is > than the Y_BED_SIZE it works. I tried X=305 Y=304 and it works.
305,305 does not.
304,305 works

thinkyhead, 2018-03-09:
Hmm… I thought we had fixed this one. The issue is that the stepper functions that poll the endstops didn't (or don't) account for the quirks of CoreXY motion. We'll have to re-check this section of the code and make sure it's compliant.

thinkyhead, 2018-03-09:
For CoreXY here's the current logic to determine if X is moving and if Y is moving:
```cpp
  // Test whether the X axis is moving in a CoreXY
  if (stepper.current_block->steps[A_AXIS] != stepper.current_block->steps[B_AXIS]
    || (stepper.current_block->steps[A_AXIS] > 0
        && stepper.motor_direction(A_AXIS) == stepper.motor_direction(B_AXIS) )
  ) . . .

  // Test whether the Y axis is moving in a CoreXY
  if (stepper.current_block->steps[A_AXIS] != stepper.current_block->steps[B_AXIS]
    || (stepper.current_block->steps[A_AXIS] > 0
        && stepper.motor_direction(A_AXIS) != stepper.motor_direction(B_AXIS) )
  ) . . .
```
I'm not sure what the exact correction is, but hopefully I'll be able to work it out by going back to the kinematic equations, where A is the sum X+Y and B is the difference X-Y…
```cpp
    block->steps[A_AXIS] = labs(da + db);
    block->steps[B_AXIS] = labs(da - db);
    block->steps[Z_AXIS] = labs(dc);
```

thinkyhead, 2018-03-09:
@ruggb — Do you have `ENDSTOP_INTERRUPTS_FEATURE` enabled?

ruggb, 2018-03-09:
I had them enabled before my upgrade started. But the BED_SIZE has never been =. I think interrupts were enabled the first time I ran into this. I don't have it enabled now, so I think it is an issue either way.

Not that I should, but I just don't understand what the bed size has to do with monitoring the endstops.
Both need to be monitored until one is triggered. Then I guess you can stop monitoring the one that is triggered.
Then there is the other part of this; 
When the bed sizes are unequal if Y is triggered it stops until X is triggered then Y starts homing again. It doesn't do that if I trigger X first. IE, X stops and stays stopped even after Y is triggered.
I'll recheck that in the AM as I am not certain when that happened. These are mechanical endstops so the trigger is momentary. So apparently it isn't remembering that Y was triggered. Maybe that helps.

thinkyhead, 2018-03-09:
> I just don't understand what the bed size has to do with monitoring the endstops.

When a perfect diagonal move is made (e.g., 0,0 to -300,-300) this results in only one Core stepper moving (A). If the move is off even by a little bit (e.g., 0,0 to -300,-301) then there will be a very slight difference in steps, and both A and B will be queued for movement.

ruggb, 2018-03-09:
right, but that happens only if it is at 0,0 going to 300,300. For 99% of the cases the carriage is someplace else. So regardless of the bed size both endstops need to be monitored because you have no idea where it is starting from, only where it is going. Well, sometimes you may know where it is, but I would think a home all would not care where it is.

thinkyhead, 2018-03-09:
G28 sets the position to 0,0 before going to -300,-300.

ruggb, 2018-03-09:
OK but that 0,0 is fictitious. If I am in the middle of the bed it is not 0,0. Well, maybe for some printers it is.
So it is using the bed size to limit the time it spends banging into endstops if something is wrong. But it should be monitoring both endstops.

thinkyhead, 2018-03-09:
I've explained why I think it's not. It depends on the stepper ISR and its logic to decide which endstops to look at. As I explained above, if it doesn't think Y is moving, it won't check the Y endstop.

ruggb, 2018-03-09:
and it doesn't think Y is moving because.........

thinkyhead, 2018-03-09:
That's what I hope my notes above will assist a savvy reader of this thread to help figure out.

ruggb, 2018-03-09:
when it sees both = it is assuming it is moving at 45 and only a or b is moving. But if you are not moving on a corner to corner diagi=onal the logic doesn't work.

ruggb, 2018-03-09:
except it is A & B and not X & Y  so it is driving toward both endstops not just X or Y

thinkyhead, 2018-03-09:
The array `current_block->steps` is directly related to the A and B steppers. Comparing `current_block->steps[A] to current_block->steps[B]` should clearly indicate that one is stationary and the other is moving. So, I cannot conclude that the "axis is moving" logic is the ultimate cause of the Y endstop being ignored at this time.

ruggb, 2018-03-09:
So if it thinks it is on a perfect diagonal, it figures just one motor is moving and ignores the Y endstop. Which would work in Cartesian space. But one motor moving is driving toward both endstops.
If both endstops were monitored for CoreXY regardless the problem would not exist.
Then why does Y start homing again after X is triggered when they are not =.

ruggb, 2018-03-09:
Sounds like Cartesian and CoreXY logic are getting mixed up.

thinkyhead, 2018-03-09:
Please try this in `planner.cpp` (at your convenience) so we can see whether these values are all as we expect they should be during `G28`. Contrast the values when the `X_BED_SIZE` and `Y_BED_SIZE` is different versus when they are the same…
```diff
  #if CORE_IS_XY
    block->steps[A_AXIS] = labs(da + db);
    block->steps[B_AXIS] = labs(da - db);
    block->steps[Z_AXIS] = labs(dc);
+   SERIAL_ECHOPAIR("Head X", TEST(dm, X_HEAD) ? '-' : '+');
+   SERIAL_ECHOPAIR(" Y", TEST(dm, Y_HEAD) ? '-' : '+');
+   SERIAL_ECHOPAIR(" ... Axis A", TEST(dm, A_AXIS) ? '-' : '+');
+   SERIAL_ECHOLNPAIR(" B", TEST(dm, B_AXIS) ? '-' : '+');
+   SERIAL_ECHOPAIR("Steps A", block->steps[A_AXIS]);
+   SERIAL_ECHOLNPAIR(" B", block->steps[B_AXIS]);
```

AnHardt, 2018-03-09:
About the second part of the problem:
The 'diagonal move' done with `QUICK_HOME` is just a additional diagonal move from a unknown position to a unknown position, to save some fractions of a second for the followed by the normal homing sequence. It does not really home anything.

ruggb, 2018-03-09:
X=300, Y=299
home all
07:26:38.226 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:38.228 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:38.232 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:38.675 : Head X- Y- ... Axis A- B- ... Steps A89850 B150
07:26:38.676 : Head X- Y- ... Axis A- B- ... Steps A89850 B150
Y endstop tripped manually
07:26:39.877 : Head X- Y+ ... Axis A- B- ... Steps A45000 B45000
07:26:39.880 : Head X- Y+ ... Axis A- B- ... Steps A45000 B45000
X endstop hit
continues homing Y
07:26:42.736 : Head X+ Y+ ... Axis A+ B+ ... Steps A500 B500
07:26:42.739 : Head X+ Y+ ... Axis A+ B+ ... Steps A500 B500
07:26:42.847 : Head X- Y+ ... Axis A- B- ... Steps A1000 B1000
07:26:42.848 : Head X- Y+ ... Axis A- B- ... Steps A1000 B1000
07:26:43.055 : Head X+ Y- ... Axis A- B+ ... Steps A44850 B44850
07:26:43.057 : Head X+ Y- ... Axis A- B+ ... Steps A44850 B44850
Y endstop hit
07:26:45.919 : Head X+ Y+ ... Axis A+ B- ... Steps A500 B500
07:26:45.921 : Head X+ Y+ ... Axis A+ B- ... Steps A500 B500
07:26:46.029 : Head X+ Y- ... Axis A- B+ ... Steps A1000 B1000
07:26:46.031 : Head X+ Y- ... Axis A- B+ ... Steps A1000 B1000
07:26:46.742 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:46.744 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:46.747 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:47.357 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:47.359 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:47.564 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:47.566 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:48.668 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:48.670 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:49.415 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:49.417 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:49.894 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:49.895 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:26:49.898 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0
;
X=300, Y=300
Axis placed so X > Y distance to home
home all
07:40:59.988 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:40:59.990 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:40:59.994 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:41:00.436 : Head X- Y- ... Axis A- B+ ... Steps A90000 B0
07:41:00.439 : Head X- Y- ... Axis A- B+ ... Steps A90000 B0
Y endstop hit manually
no response
reset activated
;
X=300, Y=300
Axis placed so Y > X distance to home
home all
07:44:01.458 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:01.460 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:01.464 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:01.917 : Head X- Y- ... Axis A- B+ ... Steps A90000 B0
07:44:01.919 : Head X- Y- ... Axis A- B+ ... Steps A90000 B0
X endstop hit
continues homing Y
07:44:03.139 : Head X- Y+ ... Axis A- B- ... Steps A45000 B45000
07:44:03.142 : Head X- Y+ ... Axis A- B- ... Steps A45000 B45000
07:44:03.146 : Head X+ Y+ ... Axis A+ B+ ... Steps A500 B500
07:44:03.148 : Head X+ Y+ ... Axis A+ B+ ... Steps A500 B500
07:44:03.256 : Head X- Y+ ... Axis A- B- ... Steps A1000 B1000
07:44:03.258 : Head X- Y+ ... Axis A- B- ... Steps A1000 B1000
07:44:03.463 : Head X+ Y- ... Axis A- B+ ... Steps A45000 B45000
07:44:03.467 : Head X+ Y- ... Axis A- B+ ... Steps A45000 B45000
Y endstop hit
homes Z
07:44:06.853 : Head X+ Y+ ... Axis A+ B- ... Steps A500 B500
07:44:06.856 : Head X+ Y+ ... Axis A+ B- ... Steps A500 B500
07:44:06.964 : Head X+ Y- ... Axis A- B+ ... Steps A1000 B1000
07:44:06.965 : Head X+ Y- ... Axis A- B+ ... Steps A1000 B1000
07:44:07.677 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:07.679 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:07.682 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:08.292 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:08.294 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:08.499 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:08.501 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:09.939 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:09.941 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:10.674 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:10.676 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:11.154 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:11.155 : Head X+ Y+ ... Axis A+ B+ ... Steps A0 B0
07:44:11.158 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count A:0 B:0 Z:0

Given that I have VERY limited idea of what we are looking at:
Since B is 0 Marlin is not looking at the Y endstop
Not a good plan for core machines???

It works if it hits X first, if not it breaks my Y endstop (2 down).
So the movement routine appears to work fine, what you do with the info is the problem.
A and B are not X and Y. The Y endstop (or X) cannot be ignored unless it has already been hit.
I know this is obvious, but someone got confused. I know the feeling...........

AnHardt, 2018-03-09:
@thinkyhead 
As expected.
Thanks for expanding the `?_MOVE_TEST` macros, but i guess the problem is two steps deeper in the `UPDATE_ENDSTOP(Y, MIN)` macro. 
```
  #define UPDATE_ENDSTOP(AXIS,MINMAX) do { \
      UPDATE_ENDSTOP_BIT(AXIS, MINMAX); \
      if (TEST_ENDSTOP(_ENDSTOP(AXIS, MINMAX)) && stepper.current_block->steps[_AXIS(AXIS)] > 0) { \
        _ENDSTOP_HIT(AXIS, MINMAX); \
        stepper.endstop_triggered(_AXIS(AXIS)); \
      } \
    }while(0)
```
Because of
```
enum AxisEnum {
  NO_AXIS   = -1,
  X_AXIS    = 0,
  A_AXIS    = 0,
  Y_AXIS    = 1,
  B_AXIS    = 1,
  Z_AXIS    = 2,
  C_AXIS    = 2,
  E_AXIS    = 3,
  X_HEAD    = 4,
  Y_HEAD    = 5,
  Z_HEAD    = 6,
  ALL_AXES  = 100
};
```
`stepper.current_block->steps[_AXIS(AXIS)]` is 0 here.

thinkyhead, 2018-03-09:
Ah, yes, quite right, quite right. Some fool (surely not me!) forgot to change this macro when he was fixing up the rest. It looks like this might fix it then:
```diff
  #define UPDATE_ENDSTOP(AXIS,MINMAX) do { \
      UPDATE_ENDSTOP_BIT(AXIS, MINMAX); \
-     if (TEST_ENDSTOP(_ENDSTOP(AXIS, MINMAX)) && stepper.current_block->steps[_AXIS(AXIS)] > 0) { \
+     if (TEST_ENDSTOP(_ENDSTOP(AXIS, MINMAX))) { \
        _ENDSTOP_HIT(AXIS, MINMAX); \
        stepper.endstop_triggered(_AXIS(AXIS)); \
      } \
    }while(0)
```

AnHardt, 2018-03-09:
@thinkyhead 
I guess ` && stepper.current_block->steps[_AXIS(AXIS)] > 0` is 
a) wrong in this special case (CORE_XY)
b) and redundant in all cases - because of the previous check by the ´?_MOVE_TEST´.
so can be omitted. But i'm unsure about b). Has to be tested.

thinkyhead, 2018-03-09:
It's definitely redundant and/or incompatible in all cases. That macro used to be the whole of the endstop test logic, but now `*_MOVE_TEST` handles all of that, tailored to the core or vanilla cartesian setup as the case may be. It is the little bone in a big fish.

ruggb, 2018-03-09:
OK, I verified the change works - bed_sizes set to 300,300. No Y crash
About the 2nd part of the problem where Y homes twice.
Is this the effect the b) issue is causing?

The action is slightly different that I previously detailed.
If I set it so the Y endstop is hit first, then home all, it hits the Y endstop ONCE then waits for the X endstop, which hits twice, then backs Y off and hits it again.

I would expect it to hit Y twice (complete home), then proceed with homing X.
That IS the way it works when it hits X first.

It is not an operational problem (yet), but I think still a bug. But then it may be as designed.

AnHardt, 2018-03-09:
After the 'diagonal move Y is already at the endstop or very close by. So if there is a first probe at all it is very smale. -> No error.

ruggb, 2018-03-09:
I don't think you got it.

AnHardt, 2018-03-09:
;-) Me to. :-(

thinkyhead, 2018-03-10:
The `QUICK_HOME` option makes the behavior of `G28`…
- Diagonal move until some endstop is struck
- Home X axis
- Home Y axis

ruggb, 2018-03-10:
I am not getting the feeling y'all are understanding what this is.
If I home all with the carriage set so that X > Y distance, Y homes with 1 approach, then X homes with 2 approaches - Then Y completes its home with the final approach. NOT NORMAL.
If I home all with the carriage set so that Y > X distance X homes with 2 approaches, then Y homes with 2 approaches - NORMAL.
Is that any clearer????

AnHardt, 2018-03-11:
One last time i try to explain what is going on.
When the diagonal move hits y-min-endstop first it is stopped.
Now x is homed normal.
Than we try the first approach for y-homing is tried. Since we already have the y-stop trigerd, this move will be max 2 steps long until this is realised. You will mot realize it at all. Now y backs up a bit and the real homing move follows.

thinkyhead, 2018-03-11:
> Is that any clearer????

Yes, it very clearly reiterates what I have already explained. You'll notice that if you set `HOME_Y_BEFORE_X` you'll get the opposite effect. There's nothing wrong. That is how we expect it to behave.

closed 2018-03-11 by thinkyhead

ruggb, 2018-03-11:
OK, it is just strange that one does one thing and the other does something else.
It is just real obvious when I trigger Y manually and it stops in the middle of the travel then starts again after X is triggered. I did that to stop it from ramming my Y endstop when I was trouble shooting the other issue.
Thanks for the quick fix to the endstop annihilator issue.

ruggb, 2018-03-11:
we may have an artifact of this fix. Babystepping no longer works.

thinkyhead, 2018-03-12:
Too soon to say. Please post a New Issue for all to see.

commit 12b697c5 (2018-03-09): Fix UPDATE_ENDSTOP for CorePQ compatibility

Fixes #9939
commit 161ff9ff (2018-03-09): Fix UPDATE_ENDSTOP for CorePQ compatibility

Fixes #9939