MarlinFirmware/Marlin #9942: Design for Piezo endstop/probe

ruggb, 2018-03-05:
I am in the process of installing a piezo probe/endstop attached to the hotend.
There are a few issues with this, but one I just noticed and realized is that a piezo only produces a pulse out when the bed it hit. Therefore operating as an endstop, the pulse goes out and it stops. The problem is if I issue another home Z command, because the endstop isn't active as in a mechanical endstop, the bed immediately drives into the head seeking the home, then backs off and rehomes.
If the home routine backed off first, then advanced to seek the first pulse and continue with the home this would accommodate this device. That should not affect current operation with a mechanical stop.
I may have to restore my mechanical endstop til I figure out the mod needed. That may not work either.
Please let me know if this is possible in the short term.

ruggb, 2018-03-05:
I put a mod in Marlin_main.cpp around line 4016 and added homeZ.
Works fine for my home Z command.
What will that break, if anything?

AnHardt, 2018-03-05:
What branch?
4016 is pure white space here (bugfix-1.1.x).

ruggb, 2018-03-05:
sorry, thought I put it in
3/1/18 1.1.x. build 63b1358
Marlin_main.cpp
4016     if (home_all || homeX || homeY || homeZ) {

AnHardt, 2018-03-05:
The general problem here is, the endstops/probe are debounced. A 'pulse' needs to be seen in two consecutive steps to be detected, better longer. There exists a workaround for the touch probes. Maybe you should have a look into that.

Your workaround, to lift Z even before homing Z to z-min in line 4016 should not harm, but causes a far z-travel downwards when homed to z-max a few lines above.

AnHardt, 2018-03-05:
Maybe unintuitive, but homing/probing **fast** may help to detect the pulse as a valid signal. (If the pulse is long enough at all.)

ruggb, 2018-03-05:
I have already increased the homing speed. That helps make the piezo op more reliable. Adding homeZ seems to work fine so far. I'll look at the touch code to see if that is a better solution. It does produce and detect a pulse as configured. The problem is it is already at Z0 but doesn't know it because the endstop is not active, so to see it it must drive into the bed. Moving away first, allows it to produce the pulse without driving into the bed.
I haven't looked at the output pulse width, but I believe the circuit stretches out the actual piezo pulse.

I also created a home all script in Repetier to add a 0.5 sec G4 delay between XY home and Z home. That also helps. You might consider an option for that in firmware for G28 since I think these piezo sensors are going to become very common. The issue is with triggering the piezo from other movement because they are so sensitive. Mine is mounted on the hotend carriage and a pulse is produced with a very minuet movement. With CoreXY, 99% of the movement is XY. XY movement does trigger it so the delay is to allow settling out. I may have to implement something in the probing sequence also, but I haven't gotten to that yet. Since I have a MIC6 plate I am only doing 3 point.

boelle, 2019-02-20:
@ruggb 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

ruggb, 2019-02-23:
another forgotten issue. bye.

closed 2019-02-23 by ruggb