MarlinFirmware/Marlin #9945: RAMPS1.4/Re-Arm Wiring For TMC2130

556duckvader, 2018-03-05:
I have searched and I am trying to locate a good diagram for wiring the TMC2130 drivers up to a RAMPS1.4/Re-Arm.  I have seen the wiring diagrams for the RAMPs and the Mega's but I can't seem to find one for the Re-Arms as it looks like different pins are being used.  I also have and LCD connected to my RAMPS.

teemuatlut, 2018-03-05:
Use SW SPI.

556duckvader, 2018-03-05:
I understand that but do you still wire it the same way that I have seen all the other or do I just pick pins that are free and map then.  I see in the pins file that the data pins appear to be mapped to the eth ports so I can only assume I use those ports on the re-arm and not use the ones on the RAMPS.

teemuatlut, 2018-03-05:
Try with the default pins at first.

556duckvader, 2018-03-05:
I know I am asking s lot of questions but is there a link to a diagram of the wiring?  I have an LCD too.

LichtiMC, 2018-03-05:
Is it mandatory to use SW SPI if I have a Display connected? Shouldn't they work both via SPI (and different CS-Pins)?

teemuatlut, 2018-03-06:
There's no diagram. Feel free to contribute if you like.

I've only tested the SW SPI with the Re-Arm+TMC2130 combo, but I also remember one user reporting that HW SPI works as well. Start with SW but you can try both.

Negaduck, 2018-03-06:
Hello Guys, 
i hope somebody can help me to get my 2130`s spinning. I´ve have posted my problem somewhere other but it seems that nobody really knowws how to work this out with Marlin 2.0 and a Re-Arm board. My current setup is a Re-Arm/Ramps combo with the TMCs on X and Y-axis and Marlin 2.0. Basically I`ve followed Thomas Sanladerer`s video on YouTube to set everything up, just instead of Marlin 1.1.8 ,Marlin 2.0. Just to test if everything works like expected I only connected the 2 stepper drivers, one stepper on y and also an optical endstop(sensorless homing not activated). Connection via pronterface, tried to home y but nothing happened. I also enabled debug command m122 but there are also no valid registers shown.  Just to be sure that my motor is working i`ve switched everything back to MEGA2560/Ramps/Marlin1.1.6/A4988 and TA-DA, it works.
I`ve got these little guys for over two weeks but just cant get them to let my motors spin. 
[MY_CONFIG.zip](https://github.com/MarlinFirmware/Marlin/files/1786434/MY_CONFIG.zip)

teemuatlut, 2018-03-06:
Use SW SPI with the default pins.

Negaduck, 2018-03-06:
I have enabled 
`#define TMC_USE_SW_SPI`
and connected all wires to the corresponding pins but now the LCD shows  directly after boot 
Driver Error

Negaduck, 2018-03-06:
[image]
[image]
[image]
i have everything connected like this

teemuatlut, 2018-03-06:
I've gone over your wiring and can't immediately see anything wrong but I'll test it myself tomorrow.

gunnarfreyr, 2018-03-06:
https://github.com/gunnarfreyr/3DPrinting/blob/master/Re_Arm_2130_wires.png

Negaduck, 2018-03-07:
U got a successful m122 with my settings? I cant even Connect it via pronterface to Check it...

teemuatlut, 2018-03-07:
With bugfix-2.0.x and TMC2130 on X axis enabled.

gunnarfreyr, 2018-03-11:
So what is u guys status on the TMC2130 ? I've got a Re-Arm + Ramps 1.4 and now using TMC2130 on all axis... However there are few things im wondering about and here is my status. 

My problem is that i feel that Stealthchop is a bit underpowered and if i print faster than 35mm sec and dont lower accel. to less than 3600 ~ i get layershifts on the Y axis.. I was thinking Stealthchop should be able to handle at least 50mm/sec or something .. 
Well should not be a problem , there is also Spreadcycle, but for me that mode is where i run into some problems,, it prints without any issues but the sounds coming from the printer are very annoying high pitch whines.. 
In both modes, i feel like the steppers get quite hot, like u cant touch them for 3 sec without burning your finger a bit, but i've added a fan to cool them down so that is  "under control" at the moment.. 

I was looking though the datasheet for the 2130's and then some googling  and there is that TMC2130_ADV, and it has some options like : 

   stepperX.external_ref(1); 
   stepperY.external_ref(1); 
   stepperX.blank_time(24);
   stepperY.blank_time(24);
   stepperX.off_time(2);
   stepperY.off_time(2);
   stepperX.hysterisis_start(0);
   stepperY.hysterisis_start(0);
   stepperX.hysterisis_low(13);
   stepperY.hysterisis_low(13);
   stepperX.run_current(31);
   stepperY.run_current(31);
   stepperX.hold_current(12);
   stepperY.hold_current(12);
   stepperX.power_down_delay(2);
   stepperY.power_down_delay(2);
   stepperX.coolstep_min_speed(300);
   stepperY.coolstep_min_speed(300);
   stepperX.sg_min(4);
   stepperY.sg_min(4);
   stepperX.sg_max(10);
   stepperY.sg_max(10);
   stepperX.smart_min_current(1);
   stepperY.smart_min_current(1);
   stepperX.sg_step_width(8);
   stepperY.sg_step_width(8);
   stepperX.sg_current_decrease(32);
   stepperY.sg_current_decrease(32);

Does anybody here know if some of these options could help with the annoying sound in the spreadcycle mode ?

Also, when i try to enable the TMC_Debug thing so i can do M122, i can boot cause i get a driver error .. I've got SW_SPI enabled.. What is the diffrence between the SW SPI and "normal" SPI is there a reason to use one over the other ?

Negaduck, 2018-03-11:
I wish i could get these guys to work. i simply dont get any valid registers with an m122 and if i enable 
`#define MONITOR_DRIVER_STATUS` the printer Halted directly after showing the bootimage... im lost with tmc2130... really sad

thinkyhead, 2018-03-11:
They are definitely a challenge to get working. The "Einsy" Rambo board takes all the work out of it and we have seen that they "just work" – unlike the self-wired stepstick style. Likely this is for two main reasons: better SPI wiring and better heat-sinking.

teemuatlut, 2018-03-11:
@gunnarfreyr
stealthChop definitely can handle 50mm/s on a typical setup but it's unlikely to be able to print at 3600mm/s^2 acceleration values. Altough perhaps with a bowden setup and a lightweight carriage.
The highest force is required for accelerating a carriage so tune down that setting but you can increase your max speed settings.

Not quite sure where you got that settings list, but to address your noise related issue you can try decreasing the `off_time` parameter. By default Marlin uses `off_time(5)`.

The difference between HW and SW communication is that HW is (much) faster but uses pins that are taken by the LCD (or rather the SD) header. Software SPI option was added so users would have an easier time installing the drivers while keeping their screen.

gunnarfreyr, 2018-03-12:
@teemuatlut Yeah i had noticed that the printer was skipping steps on the x and y doing fast travel moves cause travel was set at 90mm/sec.. 

So where i seem to have failed was too just increse the current (up to 1050 max at one time) to try to add more torque, but since i was not using a cooling fan on the steppers and printing from SD i didnt notice the heat warnings, now I atleast have a successful Benchy with no layer shifting at 40mm/sec which had failed before, even failed at 35mm/sec with 45mm/sec accel. 
I dont know if my drivers are heating up unsually much, but since cooling them with a fan seems to enable me to get a successful print with the motors around 830ma without any layershifting or issues , im just gonna rewire the ramps and steppers so the can have more room to breath and add a sidefan blowing on them 

That list i found at http://forums.reprap.org/read.php?415,780775

closed 2019-07-06 by shitcreek