MarlinFirmware/Marlin #9949: Bed bad level

PabloSica, 2018-03-05:
Hello. 
My Graber i3 with Marlin VERSION 010107 has a problem with self leveling of the bed. 
I'm using an SN04-N sensor, octorpint with RPI 3+.
T
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1780994/Configuration.h.zip)
he level system is bilinear.
When leveling begins, the system measures the first and second position without problems. But in the third, the Z axis does not stop coming down, even the sensor accusing the table, and I have to press the e-stop not to break the table and the structure.
I have already updated and changed the Marlin version with my sets.
The behavior began after several times using, without presenting problems.
My configuration.h is attached
I'm already grateful for the help. 8)

PabloSica, 2018-03-08:
You have Z_CLEARANCE_DEPLOY_PROBE 10 and Z_CLEARANCE_BETWEEN_PROBES 5
Test by raising the Z axis to the maximum, and I turned on the G29. With a piece of metal I approached the SN04-N to make a fake table, and test how far it would go.
The first points of measurements were ok, but the third one went down more than 120mm and did not stop.

thinkyhead, 2018-03-09:
The only reason that it should continue to drop is if the probe isn't being seen as triggered by the firmware. It's a pretty low level function.

Try this:
- Enable the `PINS_DEBUGGING` feature and re-flash.
- Deploy and enable the probe with `M401`.
- Send `M43 E` to start monitoring probes and endstops.
- Test with a piece of metal to see how the probe responds.
- Press RESET when done testing.

PabloSica, 2018-03-19:
I changed the sensor for RN04 N for a LJ12A3 4Z and I will do the tests now.
Sobre o Google TradutorComunidadeCelularSobre o GooglePrivacidade e TermosAjudaEnviar feedback

PabloSica, 2018-03-19:
Dude, I do not know you, but I love you! hehehe It worked. Thanks a lot for the help.

closed 2018-03-21 by thinkyhead