MarlinFirmware/Marlin #9950: Compiling with UBL enabled fails for BOARD_RAMPS_14_RE_ARM_EFB for Delta

r1cht3r, 2018-03-05:
### Description

Using any delta example config with `#define MOTHERBOARD BOARD_RAMPS_14_RE_ARM_EFB` and `#define AUTO_BED_LEVELING_UBL` fails to compile using `LPC1768`.

### Steps to Reproduce
1. Copy `src/config/examples/delta/kossel_xl/Configuration.h` & `src/config/examples/delta/kossel_xl/Configuration.adv.h` to `Marlin/Marlin`
2. Configure `#define MOTHERBOARD BOARD_RAMPS_14_RE_ARM_EFB`
3. Configure `#define AUTO_BED_LEVELING_UBL` & `#define RESTORE_LEVELING_AFTER_G28`
4. Run `platformio run --environment LPC1768`

**Expected behavior:** [What you expect to happen]
Successful compilation

**Actual behavior:** [What actually happens]
Build failure attached in [Marlin_UBL_error.txt](https://github.com/MarlinFirmware/Marlin/files/1781561/Marlin_UBL_error.txt)

#### Additional Information
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1781566/configuration.zip)

Roxy-3D, 2018-03-06:
It looks like an include file needs to be included by the HAL stuff.

It also maybe we should make the 3-point leveling locations fit better onto a Delta bed.

thinkyhead, 2018-03-06:
Try defining your probe points without using `_PX`. Calculate them and put in constants, and see if that helps. The pre-compiler doesn't seem to have access to a cosine function.

r1cht3r, 2018-03-07:
Yes, works with constants. Will try to work out why the cosine function isn't available. 

`  #define _PX(R,A) (R) * cos(RADIANS(A))
  #define _PY(R,A) (R) * sin(RADIANS(A))
  //#define UBL_PROBE_PT_1_X _PX(DELTA_PROBEABLE_RADIUS, 0)   // Probing points for 3-Point leveling of the mesh
  //#define UBL_PROBE_PT_1_Y _PY(DELTA_PROBEABLE_RADIUS, 0)
  //#define UBL_PROBE_PT_2_X _PX(DELTA_PROBEABLE_RADIUS, 120)
  //#define UBL_PROBE_PT_2_Y _PY(DELTA_PROBEABLE_RADIUS, 120)
  //#define UBL_PROBE_PT_3_X _PX(DELTA_PROBEABLE_RADIUS, 240)
  //#define UBL_PROBE_PT_3_Y _PY(DELTA_PROBEABLE_RADIUS, 240)
  #define UBL_PROBE_PT_1_X 39
  #define UBL_PROBE_PT_1_Y 100
  #define UBL_PROBE_PT_2_X 39
  #define UBL_PROBE_PT_2_Y 20
  #define UBL_PROBE_PT_3_X 100
  #define UBL_PROBE_PT_3_Y 20
`

p3p, 2018-03-07:
The cos, sin ect maths functions in the standard library are not constexpr in c++11, it's just the way it is. there are discussions on changing that but there are issues with the function not being pure (they set global error values) and floating point results not being deterministic depending on architecture. It may take a while we don't even get constexpr min/max until we move to c++14.

closed 2018-03-11 by thinkyhead