MarlinFirmware/Marlin #9953: Different extruded distance G1 x LCD

Luis-Lange, 2018-03-05:
Description:

Greetings !
I found this issue (at least for me) when extruding a fillament. I previoulsy calibrated extruder steps/mm using G1 command and it is sharp enough for me. But when using the LCD to extrude some filament I noticed that extruded distance is not equal. With G1 E10 F1000 I got exactly 10mm, but with LCD when setting 10mm I got only 4,7mm. Attached are both Configuration.h and Configuration_adv.h. Extruding backwards in both situations I got back to same initial point.

Thank you !

Expected behaviour: ---
WIth both G1 and LCD command extruded distance should be the same

Actual behaviour: ---
For example:
with G1 E10 F1000 got 10mm
with LCD 10mm distance gives real 4,7mm

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1782658/Configurations.zip)

thinkyhead, 2018-03-08:
See if it helps to lower your `MANUAL_FEEDRATE` for E from 120 down to 60.

Luis-Lange, 2018-03-08:
Hello ! No, it didn't work, same result, about half of distance. Just the extrusion speed that took 50sec to extrude 50mm.

Sebastianv650, 2018-03-08:
Just tried it with bugfix from today, no issue here. Which version are you using?

Luis-Lange, 2018-03-09:
I'm using 1.1.8 Oficial release. Tomorrow morning I'll try with today's bugfix, thanks.

Luis-Lange, 2018-03-09:
No success. I flashed with bugfix, initialized EEPROM, but same behavior. One thing I noticed, if I reset my board, come with it fresh, and send a G1 E100, for example, it moves only 50mm. Following, I send G1 E-100 and it moves 100mm. After that, all lengths are ok. Except for LCD commands, that still behaves with half distance.... :) :) :).
By the way. I'm using "Marlin Configurator" from https://github.com/akaJes/marlin-config. I'm still fighting with Arduino IDE..

AnHardt, 2018-03-09:
Check `MANUAL_FEEDRATE` (C*_adv.h).

Luis-Lange, 2018-03-09:
Hi Guys ! My mistake, sorry !!! :( :( 
G1 is for absolute positioning and I'm doing G1 E100 and afterwards G1 E-100, so 200mm extruded, not 100mm. Correct should be G1 E100 and G1 E0 to get 100mm. So, LCD is correct and my steps/mm was with half value. M114 tip was great to reveal my mistake.
Just for your information, I could compile the firmware with Arduino IDE and result was the same with "Marlin Configurator" (should be...) and this tool make configuration easier than notepad.
Thank you for your efforts and sorry again...

closed 2018-03-09 by Luis-Lange

thinkyhead, 2018-03-10:
> By the way. I'm using "Marlin Configurator"

It's a great tool, and always improving. Be sure to send some love to its author.