MarlinFirmware/Marlin #9955: Z-Stepper Moves Inaccurately

DBSETIAWAN, 2018-03-06:
Hi All.
First of all, I wanna say thanks to all of readers in here that have a minutes to read my question.

A few weeks ago, I've purchased Cultivate 3D with Repetier as a firmware. But, cause by something, I should change it to Marlin as soon as possible, and it works, I just calibrate some cables and axis.

#### First,
after I change into marlin, I followed the tutorial in here:
https://github.com/MarlinFirmware/Marlin/issues/3810
The movements of every axis are smooth. But I realized that movement 10mm XY-axis is different with Z-axis. The Z 10mm movement looks like 100mm. Could you help me to fix this problem ?

#### Second
Because of Cultivate 3D Beast have 600x600x700 mm area printing, but after I applied Marlin, the extruder only moves 200mm maximum. What should I change the section of the codes ?

Thank you very much to read, and please help me to fix those problems.
And sorry for my bad english :)

ruggb, 2018-03-06:
Unfortunately, the items you need to change are spread around in configuration.h and configuration_adj.h files there may also be settings in your control program - Repetier-Host maybe.
In marlin there is bed size and steps/mm settings. let me get them..........
 #define DEFAULT_AXIS_STEPS_PER_UNIT   { 200, 200, 800, 185.2 }

// The size of the print bed
#define X_BED_SIZE 335 
#define Y_BED_SIZE 295 
// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 295 

Those settings are mine and not the default settings, but I think the default for bed size is 200 which would be why yours only moves that far.

There are a lot of other settings that will affect your printer. You will have to step thru these to files and carefully set all that apply to you.
If you use Repetier-Host, there are items there you need to set. Your slicer also has settings that need attention.
good luck

DBSETIAWAN, 2018-03-06:
Thanks a lot @ruggb !!
It definetely works for me !

closed 2018-03-06 by DBSETIAWAN