MarlinFirmware/Marlin #9958: RE-ARM stepper issues

cyris69, 2018-03-06:
### Description

I'm running the most recent release of 2.0.x, I've configured it all up with manually moving values from my 1.1.8 build and all flashed perfect. I'm able to connect over octoprint, send commands etc. Fan, bed, and hotend commands work for heating/cooling. However I can't enable //#define PIDTEMPBED, while it will build and flash the board locks up during boot.

So that is what I've done and what works. The issue I'm having with the steppers is that when either using the terminal to tell an axis to move or a general G28 to xy or z will engage the motors and lock them but nothing moves. If you attempted to send any command(s) soon after any homing or move command even if you try homing again. I like to do a home xy and while that's going hit home z which will start after the xy is finished. I have my ramps 1.4 premium sb board wired the same as when it had an atmega 2560 board attached running 1.1.8 and that wiring also worked with smoothie when messing with re-arm.
```
Send: G91
Send: G28 X0 Y0
Recv: ok
Send: G90
[...]
Recv: echo:busy: processing
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
```

Unless the jumper pins for the driver step mode (4,8,16,32,etc) are different now in 2.0 vs 1.1.8 and before then I can see that being the issue if you have 32step drivers and its thinking you want them at half step or something.

### Steps to Reproduce

1. Flash board using my FW and connect over serial in octoprint
2. Try and home/move any axis 
3. Profit?

**Expected behavior:** Expected normal homing/movement

**Actual behavior:** Motors engage/lock and causing a busy loop in terminal for about 30+ seconds

#### Additional Information
Configuration.h & Configuration_adv.h files
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1783515/Marlin.zip)
Firmware .bin compiled using above files.
[firmware.zip](https://github.com/MarlinFirmware/Marlin/files/1783527/firmware.zip)

cyris69, 2018-03-06:
I will give that a shot, I'm using 8825s on all motors but have the driver for the extruder set to 1/16 and the rest 1/32. Will this mess with the movement speeds, where normally you have 80,80,400,95 at 1/16 and double those on 1/32?

kAdonis, 2018-03-06:
of course you have to double the Steps for every axis you changed to 1/32
in your case 160,160,800,95

cyris69, 2018-03-06:
Just wanted to make sure that changing pulses didn't affect those values.

cyris69, 2018-03-06:
@kAdonis that seems to be the ticket, forgot to enable safe z homing so now just fixing that and see how she chooches.

p3p, 2018-03-06:
The `MINIMUM_STEPPER_PULSE` setting appears to trip a lot of people up when switching to a faster MCU, the correct value is hardware dependant and can be found in your stepper driver datasheet, it can only be 0 on 8bit AVR because the MCU isn't fast enough to toggle the pins too fast anyway. 

The DRV8825 datasheet says a minimum delay of 1.9us is needed so setting `MINIMUM_STEPPER_PULSE` to 2 should work well and half the wasted cycles.

cyris69, 2018-03-06:
Ill try recompiling with 2, once I figure out why after the print heats up to the desired bed temp and starts it turns off the bed.

556duckvader, 2018-03-07:
Question will this change #define MINIMUM_STEPPER_PULSE 2 affect the way the A4988's work if you have a mix of stepper drivers?

p3p, 2018-03-07:
Marlin treats all steppers the same, so you will need to choose a `MINIMUM_STEPPER_PULSE` that works for all your drivers, the good thing is that A4988 (like other common stepper drivers, except the drv8825) have a less than 1us minimum pulse requirement. This is a minimum time, longer times than necessary are not a problem for the drivers, but Marlin can do nothing else during this delay so keep it as short as possible or it will hit performance.

thinkyhead, 2018-03-09:
> it can only be 0 on 8bit AVR because the MCU isn't fast enough to toggle the pins too fast anyway.

A setting of 1 can be helpful for the Linear Advance ISR, since it only pulses one stepper. That value has the added benefit of adding no extra delay to the main stepper ISR.

boelle, 2019-02-20:
@cyris69 still having issues?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-15 by p3p