MarlinFirmware/Marlin #9962: 1.1.x bugfix: 'FILAMENT_UNLOAD_RETRACT_LENGTH' was not declared in this scope

gorischek, 2018-03-06:
Hi,

I enabled _ADVANCED_PAUSE_FEATURE_ but get now the error _'FILAMENT_UNLOAD_RETRACT_LENGTH' was not declared in this scope_ if I verify the code

### Steps to Reproduce

1. use Anet A8 configuration files 
2. enable "ADVANCED_PAUSE_FEATURE" in configuration_adv.h
3. Verify the code

**Expected behavior:** The code is verified without an error
**Actual behavior:** This error is shown: 'FILAMENT_UNLOAD_RETRACT_LENGTH' was not declared in this scope

Best regards
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1785224/Configuration_adv.zip)

GMagician, 2018-03-06:
Update your configuration.h, check from examples and compare with yours. Some define have been added/modified

thinkyhead, 2018-03-07:
https://sourceforge.net/projects/npp-compare/

closed 2018-03-07 by thinkyhead

gorischek, 2018-03-08:
sadly this doesn´t solve it. I have done following steps:

1) I copied the original configuration.h and configuration_adv.h
2) ->verified->pass
3) I enabled option #define ADVANCED_PAUSE_FEATURE in line 879 (configuration_adv.h)
4) ->verified-> #error "FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS is now FILAMENT_CHANGE_ALERT_BEEPS. Please update your configuration."
5) changed line 899 to   #define FILAMENT_CHANGE_ALERT_BEEPS 5 (configuration_adv.h)
6) ->verified-> #error "ADVANCED_PAUSE_FEATURE requires NOZZLE_PARK_FEATURE."
7) enabled option NOZZLE_PARK_FEATURE in line 1174 (configuration.h)
8) ->verified->'FILAMENT_UNLOAD_RETRACT_LENGTH' was not declared in this scope

gorischek, 2018-03-08:
@thinkyhead : please reopen this ticket

thinkyhead, 2018-03-08:
As "GMagician" points out, your configurations are simply out of date, and the solution is to update them. Start with the current configuration files and use some kind diff tool or plugin for your favorite text editor to apply your customizations.

This is what your advanced pause feature should look like now:
```cpp
/**
 * Advanced Pause
 * Experimental feature for filament change support and for parking the nozzle when paused.
 * Adds the GCode M600 for initiating filament change.
 * If PARK_HEAD_ON_PAUSE enabled, adds the GCode M125 to pause printing and park the nozzle.
 *
 * Requires an LCD display.
 * Requires NOZZLE_PARK_FEATURE.
 * This feature is required for the default FILAMENT_RUNOUT_SCRIPT.
 */
//#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE 60      // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH 2         // (mm) Initial retract.
                                              // This short retract is done immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 10  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 100   // (mm) The length of filament for a complete unload.
                                              //   For Bowden, the full length of the tube and nozzle.
                                              //   For direct drive, the full length of the nozzle.
                                              //   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 6     // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_LOAD_LENGTH 0       // (mm) Load length of filament, from extruder gear to nozzle.
                                              //   For Bowden, the full length of the tube and nozzle.
                                              //   For direct drive, the full length of the nozzle.
  #define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 3   // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 50    // (mm) Length to extrude after loading.
                                              //   Set to 0 for manual extrusion.
                                              //   Filament can be extruded repeatedly from the Filament Change menu
                                              //   until extrusion is consistent, and to purge old filament.

                                              // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_RETRACT_LENGTH 13   // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY 5000          // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH 8      // (mm) An unretract is done, then this length is purged.

  #define PAUSE_PARK_NOZZLE_TIMEOUT 45        // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS 10      // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT       // Enable for XYZ steppers to stay powered on during filament change.

  //#define PARK_HEAD_ON_PAUSE                // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE       // Ensure homing has been completed prior to parking for filament change

  //#define FILAMENT_LOAD_UNLOAD_GCODES       // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  //#define FILAMENT_UNLOAD_ALL_EXTRUDERS     // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif
```
See **`FILAMENT_UNLOAD_RETRACT_LENGTH`**??